shmuelzon/esp32-ble2mqtt #80: Q: Will this setup work with your code?

gvdhoven, 2020-11-11:
Use-case: portable barcode scanner which I want to use at home for packages I ship.
I have [a working Android studio project](https://gist.github.com/gvdhoven/2146f469aacc68f81f29ca695a023d4b) which connects and subscribes to the right characteristic. I also have a version which scans every other time but the above one was the one i ended up coding.

However, I also have the `MiJia` sensors, therefore hoping your solution works for adding my scanner.
The primary service of the advertised device is: `0xFEE7` 
The 'second' service on which the correct characteristic is available is: `0xFEEA` called `Unknown service` (as per nRF Connect)
The characteristic itself i am interested in is: `0x2AA1` called `Magnetic Flux Density - 3D` (as per nRF Connect)

Is my assumption correct that i would only need to add the service and characteristic `0xFEEA` and `0x2AA1` to the config file, if needed only white-listing my scanner mac `aa:a8:ac:18:21:3d`? The data which comes in is an array of decimal bytes.

Assuming if you don't see any issues above, naming `0xFEE7` "BarcodeService" in the config, and `0xFEEA` "Barcode" the topic would become `<prefix>/aa:a8:ac:18:21:3d/BarcodeService/Barcode` ?

shmuelzon, 2020-11-11:
Hey,

I think you're pretty spot on regarding what needs to be done.
This device apparently uses 16bit UUIDs which are reserved for official GATT services and characteristics. That's why `0x2AA1` appears as `Magnetic Flux Density - 3D`. The complete UUID would be: `00002aa1-0000-1000-8000-00805f9b34fb` and this project would also publish it as `MagneticFluxDensity-3D` (and try to parse it as such).
Likewise, the `0xFEEA` service would be published as `0000feea-0000-1000-8000-00805f9b34fb`.

Anyways, as you suggested, yes, I would add these services and characteristics to the configuration file (using the complete UUIDs) to override the default names and types. If you want the values published as decimal byte arrays use an empty `types` array (`[]`) when adding them to the characteristics' definitions to override the default types. Unless that value is some other type, i.e. string, in which case you can have this application publish it parsed for simplicity.

As for whitelisting/blacklisting, I, personally, whitelist only the devices I'm interested in as there are lot's of devices (and cellphones which use random MAC addresses) that I don't want to try and connect to. You can also whitelist only the barcode characteristic if the rest of the characteristics aren't of interest, it depends on what's exposed. Perhaps you'd want the battery level, etc.

Let me know how if works out for you.

BTW - Which barcode scanner are you using? I was planning on building something similar, but if there is something useful already available, I might just use that.

Thanks

gvdhoven, 2020-11-11:
Thank you for your reply @shmuelzon ; [i have this device](https://nl.aliexpress.com/item/4000313730039.html), and the array is actually a string indeed; I'll try it tomorrow!

gvdhoven, 2020-11-16:
@shmuelzon i am about to attempt this during lunch; you mention:
> Modify the configuration file to fit your environment, build and flash (make sure to modify the serial device your ESP32 is connected to)

Where do i set the serial port for initial flashing?

shmuelzon, 2020-11-16:
I don't have it in front of me at the moment, but I believe it's under the 'serial flasher' menu, or something like that.

gvdhoven, 2020-11-16:
Here it is; which useful to add to the readme if you ask me:
1. [Install ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#installation-step-by-step)
2. Follow at least [until step 5](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-2-get-esp-idf).

Then i was able to execute `make flash` which in the first attempt asks you for the ESP-IDF config via the menu.
p.s. you can invoke this manually with `idf.py menuconfig`.

However; build failed with this error:
`fatal error: eth_phy/phy_lan8720.h No such file or directory`

shmuelzon, 2020-11-16:
You're not using the ESP-IDF version specified in the README file. You should install v3.2.2.
Newer versions should be supported very soon.

gvdhoven, 2020-11-16:
my bad :) but the 1. and 2. would make sense to add if you ask me, i'll see if i can open a PR for it.

shmuelzon, 2020-11-18:
@gvdhoven, were you able to get this project up and running?
Please note that I've updated it for ESP-IDF v4.1 so their latest installation procedures are now relevant for this project as well.

gvdhoven, 2021-05-27:
No not yet, life and changed work/life balance due to Covid working from home interfered :) i am trying to get this done pretty soon  tough and will update you here.

On another note; can i also use your project to detect if certain phones are in range (e.g. presence detection?)

shmuelzon, 2021-05-27:
> can i also use your project to detect if certain phones are in range (e.g. presence detection?)

Yes and no. While it will technically work, i.e. if the phone will broadcast something we can pick up on it, but mobile phones randomize their MAC addresses to avoid just that. So, you'll see a device, but you won't know for sure if it's your phone or not

gvdhoven, 2021-05-28:
crap. Before i reinstall my windows PC (since the ESP IDF compiler doesn't like it when your login name is "Name1 & Name2" (i think it chokes on the "&") would compiling this project also be possible on the mac?

shmuelzon, 2021-05-28:
In my case, it didn't like the spaces in the directory name.
As for a Mac, it used to compile a while back, when I still used one. ESP-IDF does support Macs

gvdhoven, 2021-05-28:
Confirmed :)

[image]

gvdhoven, 2021-05-28:
Thank you @shmuelzon everything works as expected :)

closed 2021-05-28 by gvdhoven

shmuelzon, 2021-05-28:
Cool. Glad to hear you got it working for you!