shmuelzon/esp32-ble2mqtt #81: No such file or directory... eth_phy...

Yury777, 2020-11-12:
Hi,
I ran "make flash", but the following error occurred:

App "ble2mqtt" version: v0.12.0-5-g6ae729e
CC build/main/eth.o
In file included from /Users/1/esp/esp32-ble2mqtt/main/eth.c:4:
/Users/1esp/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
/Users/1/esp/esp32-ble2mqtt/main/eth.c:7:10: fatal error: eth_phy/phy_lan8720.h: No such file or directory
 #include <eth_phy/phy_lan8720.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [eth.o] Error 1
make: *** [component-main-build] Error 2

shmuelzon, 2020-11-12:
Hey,

I believe you are not using the correct version of ESP-IDF as stated in the README file, v3.2.2.

Please try to change the ESP-IDF version and see if that helps.

Let me know...

alxlo, 2020-11-15:
Can confirm. This does not work with the  v4.1 IDF. Had to learn it the hard way and am happy to have found this issue. It worked with IDF v3.2.2. 

@shmuelzon Maybe you could add to the README that a newer IDF version will not work to make this point more explicit.

shmuelzon, 2020-11-15:
Hey,

There is a brach dedicated for supporting v4.1 but I found it less stable than v3.2.2 so it wasn't merged to master. However, I do have some not-yet-pushed changes that might improve things and I want it to run for a few days to see how it performs.

I'll add a note to the README and I might also add a note when opening issues to avoid these kinds of problems.

Thanks for the feedback!

gvdhoven, 2020-11-16:
see #80 -> correct guide to follow (for now) is: https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started/#setup-toolchain

commit 04d3dcea (2020-11-18): Added issue templates

Fixes #81
closed 2020-11-18 by shmuelzon