shmuelzon/esp32-ble2mqtt #83: how read characteristic-notification

Yury777, 2020-11-25:
Hi!
I have Thingy:52.
I was able to install and configure esp 32-ble2mqtt using the configuration from [wiki](https://github.com/shmuelzon/esp32-ble2mqtt/wiki/Nordic-Thingy%3A52), but it only scans and transmits characteristics of the "Read/Write" type and does not scan the "Notify"type (temperature, hummidity, pressure, button...). Please tell me, How can I subscribe to Notify and get their data?

shmuelzon, 2020-11-25:
Hey,

Those should work automatically. Every characteristic that supports notifications is automatically registered on and once data is sent from the BLE peripheral, it's published on the MQTT bus.

Which version of this project are you using? Which ESP-IDF version? Do you have anything visible in the logs that might point to an issue?
BTW - v0.13.0 is new and uses the latest ESP-IDF, v4.1. As it may still have some issues, you can try v0.12.0 that used ESP-IDF v3.2.2 that was very stable for a long time.

Thanks

Yury777, 2020-11-26:
App version:      v0.13.0
ESP-IDF:          v4.3-dev-1901-g178b122c1
here is the log in the monitor: 
I (12642) BLE2MQTT: Discovered BLE device: da:a6:47:1c:6e:a2 (RSSI: -47), connecting
I (15942) BLE2MQTT: Connected to device: da:a6:47:1c:6e:a2, scanning
I (16542) BLE2MQTT: Publishing: da:a6:47:1c:6e:a2/Thingy Configuration Service/Thingy Name Characteristic = flokk1
I (16632) BLE2MQTT: Publishing: da:a6:47:1c:6e:a2/Thingy User Interface Service/Thingy LED Characteristic = 1,255,0,0,0
I (16692) BLE2MQTT: Publishing: da:a6:47:1c:6e:a2/Thingy Sound Service/Thingy Sound Configuration Characteristic = 1,1

I tried installing V. 3. 2. 2 on raspberry pi 3b+ but couldn't...

shmuelzon, 2020-11-26:
I'm not sure ESP-IDF v3.2.2 can be installed on a Raspberry Pi but, regardless, I suggest switch to v4.1 and not v4.3 as only the former is supported.
A lot of time has passed since the publications and the error you posted in the log. It would be interested to know what went on in between. Can you attach the complete log?

shmuelzon, 2020-12-02:
Hey, could you please try the latest version of this project (from 3 minutes ago). There was a fix there that might be applicable here as well.

Yury777, 2020-12-06:
I installed idf 4.1 on a raspberry pi 3b+ and downloaded the firmware. MQTT is receiving normal data from the sensors! HURRAY! but when updating the configuration via OTA, it returns an error:

Executing action: upload
Running ninja in directory /home/pi/esp/esp32-ble2mqtt/build
Executing "ninja upload"...
[2/3] cd /home/pi/esp/esp32-ble2mqtt/b...pi/esp/esp32-ble2mqtt/requirements.txt
FAILED: CMakeFiles/check-project-python-requirements 
cd /home/pi/esp/esp32-ble2mqtt/build && /home/pi/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/pi/esp/esp-idf/tools/check_python_dependencies.py -r /home/pi/esp/esp32-ble2mqtt/requirements.txt
The following Python requirements are not satisfied:
paho-mqtt
To install the missing packages, please run "/home/pi/esp/esp-idf/install.sh"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

pi@raspberrypi:~/esp/esp-idf $ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20191114 (already installed)
Installing Python environment and packages
Installing Python packages from /home/pi/esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 4)) (44.1.1)
Requirement already satisfied: click>=5.0 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 11)) (3.2.1)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 13)) (0.27)
Requirement already satisfied: ipaddress; python_version < "3" in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: enum34; python_version < "3" in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.1.10)
Requirement already satisfied: six>=1.4.1 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.14.4)
Requirement already satisfied: pycparser in /home/pi/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (2.20)
All done! You can now run:

  . ./export.sh

shmuelzon, 2020-12-06:
Yeah, I'm using a script that's a part of ESP-IDF to check for the needed Python libraries but the command to fix it isn't really relevant...
Try to run:
```bash
/home/pi/.espressif/python_env/idf4.1_py2.7_env/bin/python -m pip install paho-mqtt
```

Yury777, 2020-12-07:
Thank you!!!
everything worked!

closed 2020-12-07 by Yury777

shmuelzon, 2020-12-07:
Glad to here it works for you. Thanks for the update!