shmuelzon/esp32-ble2mqtt #84: Characteristics only update once upon connect

natwilson, 2020-11-29:
I'm trying to read the status of my Sonicare toothbrush.  I've been able to get the program to work, but all of the characteristics are only updated once when it first connects.  Characteristics like the active seconds timer should be updating every second.    I've confirmed the toothbrush is pushing updates every second using a debugger on my mobile phone.

I'm able to poll for updates using the /Get command.  My understanding of the wiki is that it should work as a notification service and polling shouldn't be required.

App version 0.13
ESP-IDF v4.1

I've attached my config and logs.  The MAC of the toothbrush is 24:e5:aa:b0:f6:6f

[BLE2MQTT.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/5612945/BLE2MQTT.log)
[config.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/5612947/config.txt)

shmuelzon, 2020-11-30:
Hey,

I don't see any relevant errors in the logs that would point to an issue with notifications.
Which debugger are you using? Is it an app like nRF connect / LightBlue or have you captured BLE traffic while using an app made for the toothbrush?
I'm asking because in case of the latter, you might need to write something to a characteristic to make the toothbrush start notifying. In general, if this application detects a characteristic that supports notifications or indications, it will register for updates but that won't be enough if there is an application level configuration needed as well.
It's also possible that you need some pairing done beforehand that your mobile phone has already completed, but that's not as common.
If you can, try to enable debug level logs and it will print every characteristic found and its properties so we can make sure we're registering for them as expected.

BTW - in your configuration file you've added types "int8" and "int16". Those do not exist and that's why those characteristics aren't parsed as they're supposed to. If those characteristics are unsigned, use "uint8" and "uint16" instead, otherwise use "sint8" and "sint16". I know it's not very intuitive but those are the names the Bluetooth SIG have came up with.

Thanks

natwilson, 2020-12-01:
Thanks for the reply.  I've tested the toothbrush using the LightBlue app and it will push notifications every second.

I've tried enabling debug level logs in menuconfig, but then the ESP32 won't boot and fails with the error esp_image: Image length 1631424 doesn't fit in partition length 1572864.  I'm too new to ESP-IDF to know what to do next.

shmuelzon, 2020-12-01:
It seems that enabling debug logs also increases the application size so it doesn't fit anymore in its designated partition.
Could you please also change the compiler optimization to "size", that reduces it enough to fit (and I hope it will run as expected, I can't test it just yet). I did give me a compiler warning but that shouldn't interfere.
Anyways, open the configuration menu with `idf.py menuconfig`. Search, using the `/` key, for the `LOG_DEFAULT_LEVEL` choice menu and select `DEBUG`. Next, search for the `COMPILER_OPTIMIZATION` choice menu and select `Size` and build again. This time the image will fit and should load.

natwilson, 2020-12-01:
That worked, thanks for the instructions.  I've attached the debug level logs.  The toothbrush ran for about one minute.  As you can see the Active_time was only updated once.
[Sonicare-log-12-1.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/5625095/Sonicare-log-12-1.txt)

shmuelzon, 2020-12-02:
Unfortunately, I don't see any problems in the logs. The service is discovered as expected. The `0x12` properties means it's readable and supports notifications. Those operations (reading its value and registering for notifications) are enqueued:
```
D (57449) BLE: Enqueue: type: 0, device: 24:e5:aa:b0:f6:6f, char: 477ea600-a260-11e4-ae37-0002a5d54090, len: 0, val: 0x0
D (57459) BLE: Enqueue: type: 3, device: 24:e5:aa:b0:f6:6f, char: 477ea600-a260-11e4-ae37-0002a5d54090, len: 2, val: 0x3ffec374
```
And they are both later dequeued and send to the BLE library:
```
D (61769) BLE: Dequeue: type: 0, device: 24:e5:aa:b0:f6:6f, char: 477ea600-a260-11e4-ae37-0002a5d54090, len: 0, val: 0x0
D (61899) BLE: Dequeue: type: 3, device: 24:e5:aa:b0:f6:6f, char: 477ea600-a260-11e4-ae37-0002a5d54090, len: 2, val: 0x3ffec374
```
And we get an acknowledgements for both:
```
D (61889) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
D (61889) BLE2MQTT: Queuing event BLE_DEVICE_CHARACTERISTIC_VALUE (24:e5:aa:b0:f6:6f, 477ea600-a260-11e4-ae37-0002a5d54090, 0x3ffefd28, 2)
D (61939) BLE: Received GATTC event 9 (ESP_GATTC_WRITE_DESCR_EVT), gattc_if 1
```
From that point on we should be notified by the BLE stack when new data is available...

There's only one thing I can think of that might be an issue but as it didn't show up until now I can't be sure. Maybe it relates to the BLE stack on the toothbrush. Anyways, can you please try to make the following modification and see if it helps?
```patch
--- a/main/ble.c
+++ b/main/ble.c
@@ -153,7 +153,7 @@ static inline void ble_operation_perform(ble_operation_t *operation)
         esp_ble_gattc_write_char_descr(g_gattc_if, operation->device->conn_id,
             operation->characteristic->client_config_handle, operation->len,
             operation->value,
-            ESP_GATT_WRITE_TYPE_NO_RSP, ESP_GATT_AUTH_REQ_NONE);
+            ESP_GATT_WRITE_TYPE_RSP, ESP_GATT_AUTH_REQ_NONE);
         break;
     }
 }
```

natwilson, 2020-12-02:
That worked, thank you!   The Active_seconds characteristic is updating every second and other characteristics like bushing mode are also updating when they change.

Also, thanks for the earlier help with the integers.  I was able to clear up those errors.

shmuelzon, 2020-12-02:
Glad to hear it worked out for you!

I'm not sure how common these toothbrushes are, but I would appreciate it if you could add your services and characteristics configuration to the Wiki.

commit 6168e2b1 (2020-12-02): Use a "regular" write command for description

Fixes #84
closed 2020-12-02 by shmuelzon