shmuelzon/esp32-ble2mqtt #87: Battery status in Mqtt and web page

nubicula, 2020-12-22:
Would it be possible to expose the battery status to the MQTT node and also in the http server.

shmuelzon, 2020-12-23:
I don't have a device that runs on battery and I don't think there's some standard way to read it.
I'm guessing most will use some sort of voltage regulator meaning they'll connect the battery voltage (through a resistor divider) to some random GPIO. Even in this case, you'll need to know which resistors were used and what type of battery it is (and its discharge voltage curve) to convert it to a percentage value.

Regardless, this project doesn't really take battery life into consideration. Both WiFi and Bluetooth are always on and it never sleeps. What's your use-case for running it on battery?

nubicula, 2021-01-05:
So I have both an m5stick and an m5stack core 2 (https://m5stack.com/collections/m5-core). These both have batteries, and having read they have libraries for their batteries, but as your project is to work with all esp32's I understand you probably won't be able to integrate. 
I was more intending to use it to report back if their battery starts going down then their power supply had been interrupted, and this would give potential time to act before losing access entirely.

shmuelzon, 2021-01-06:
From what I gather, both device have their own power management IC (AXP192) that the ESP32 communicates with over I²C to get the battery level. I fear this is too specific for this project. If it was possible to read the battery level via ADC from a GPIO then I might have been able to set up something. Sorry.

nubicula, 2021-01-06:
No Problem. Thanks for taking your time to discuss it. All the best for the project.

closed 2021-01-06 by nubicula