shmuelzon/esp32-ble2mqtt #88: FAILED: CMakeFiles/validate-config

nyqeest, 2020-12-22:
Hi, I have a problem with uploading the program to my esp32. I installed idf version 4.1.
In the config.json file I changed only the ssid, password and the ip address of the mqtt server.
After running idf.py build I get this message:

> Running ninja in directory c:\users\ny\desktop\esp-idf\projekty\ble2mqtt\build
> Executing "ninja all"...
> [1/1198] cmd.exe /C "cd /D C:\Users\ny\Desktop\esp-idf\pro..."Error: Invalid JSON in configuration file." && false ) )"
> FAILED: CMakeFiles/validate-config
> cmd.exe /C "cd /D C:\Users\ny\Desktop\esp-idf\projekty\ble2mqtt\build && ( C:\Users\ny\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe -m json.tool C:/Users/ny/Desktop/esp-idf/projekty/ble2mqtt/data/config.json 1>/dev/null || ( echo "Error: Invalid JSON in configuration file." && false ) )"
> System nie może odnaleźć określonej ścieżki.
> "Error: Invalid JSON in configuration file."
> 'false' is not recognized as an internal or external command,
> operable program or batch file.
> [6/1198] cmd.exe /C "cd /D C:\Users\ny\Desktop\esp-idf\pro...-obj-name-len=32 --meta-len=4 --use-magic --use-magic-len"
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1

Uploading projects from the example folder works. I would be very grateful for your help.

shmuelzon, 2020-12-23:
Looks like you're running this natively on Windows and not via WSL or MinGW.
I never tried this kind of setup so it will take a while for a proper fix. In the meantime, the following change should get you started:
```patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 1ccecf8..67d2790 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -18,8 +18,7 @@ add_custom_target(check-project-python-requirements
         -r ${PROJECT_DIR}/requirements.txt)

 add_custom_target(validate-config
-    COMMAND ${python} -m json.tool ${PROJECT_DIR}/data/config.json 1>/dev/null
-        || (echo "Error: Invalid JSON in configuration file." && false ))
+    COMMAND echo "Done")

 add_custom_target(upload
     COMMAND ${python} ${PROJECT_DIR}/ota.py -f ${build_dir}/${PROJECT_BIN}
```

salbader, 2020-12-29:
Hi,

I had the same issue while running this natively under windows. The quick fix worked for me.
A further issue was the missing gzip. After installation, I could compile and burn to my ESP, thanks.

shmuelzon, 2021-03-14:
Hey, a proper fix is now in place. As @salbader mentioned, you would still need Gzip installed on your machine. You can download it from http://gnuwin32.sourceforge.net/packages/gzip.htm.
Newer versions of CMake have built-in support for compression but not the version currently installed by ESP-IDF so I'll stick with `gzip` for now.

closed 2021-03-14 by shmuelzon

bulazhang, 2021-03-15:
On Windows OS,  It should changed /dev/null to NUL. Like this:
add_custom_target(validate-config
    COMMAND ${python} -m json.tool ${PROJECT_DIR}/data/config.json 1>NUL
        || (echo "Error: Invalid JSON in configuration file." && false ))

shmuelzon, 2021-03-15:
Yes, that was what I did. Also `false` doesn't exist and was changed to `exit 1`.
Do you still have issues compiling on windows?

commit cfd2bb4a (2021-03-14): Support validate-config on Windows

Fixes #88