shmuelzon/esp32-ble2mqtt #89: Send byte array to Yellight candela

lucs7, 2020-12-27:
I am trying to use the gateway to send a byte array of hex values to a command characteristic of a yeelight candela through mqtt, but I got stuck at understanding the type definition. Currently my config looks like this:

```
{
   "wifi":{
      "ssid":"xxxxxxxxx",
      "password":"xxxxxxxxxxxxxxx"
   },
   "mqtt":{
      "server":{
         "host":"192.168.180.50",
         "port":1883
      },
      "publish":{
         "retain":true
      },
      "topics":{
         "prefix":"",
         "get_suffix":"/Get",
         "set_suffix":"/Set"
      }
   },
   "ble":{
     "whitelist": ["f8:24:41:c3:a3:2d"],
     "services":{
         "definitions":{
            "8E2F0CBD-1A66-4B53-ACE6-B494E25F87BD":{"name":"Milight Candela"}
         }
      },
      "characteristics":{
         "whitelist":["aa7d3f34-2d4f-41e0-807f-52fbf8cf7443"],
         "definitions":{
            "aa7d3f34-2d4f-41e0-807f-52fbf8cf7443":{"name":"Light State", "types":[]}
         }
      },
      "passkeys":{}
   }
}
```

which successfully connects and monitors the characteristic aa7d3f34-2d4f-41e0-807f-52fbf8cf7443. When I connect through nRF Connect on android I can send 4340010101f4 to turn on or4340020101f4 to turn off as byte arrays (as described in [here](https://github.com/nklerk/nl.nielsdeklerk.XiaomiYeelightCandela/blob/master/drivers/candela/device.js)) In mqtt explorer the values are shown as  _67,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0_ which correspond to the default decimal definition.
I tried various types, but don't really understand how to define the hex bytes. 

How would I correctly define the bytes? Which type?
Thanks for any help

---
device serial log

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6944
ho 0 tail 12 room 4
load:0x40078000,len:14288
load:0x40080400,len:3672
entry 0x40080678
[0;32mI (31) boot: ESP-IDF v4.3-dev-2136-gb015061-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 16:16:25[0m
[0;32mI (32) boot: chip revision: 1[0m
[0;32mI (36) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (43) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (47) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (52) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (56) boot: Enabling RNG early entropy source...[0m
[0;32mI (62) boot: Partition Table:[0m
[... 319 lines cut ...]
[0;32mI (568916) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (571156) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (572196) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (573236) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (579236) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;33mW (590056) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0008[0m
[0;32mI (590056) BLE: Connection closed, reason = 0x8[0m
[0;32mI (590056) BLE2MQTT: Disconnected from device: f8:24:41:c3:a3:2d[0m
[0;32mI (590596) BLE2MQTT: Discovered BLE device: f8:24:41:c3:a3:2d (RSSI: -51), connecting[0m
[0;32mI (591116) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (591346) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (592146) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (592406) BLE2MQTT: Connected to device: f8:24:41:c3:a3:2d, scanning[0m
>>>> Unhandled characteristic type 0 <<<<
[0;32mI (593066) BLE2MQTT: Publishing: f8:24:41:c3:a3:2d/0000fe87-0000-1000-8000-00805f9b34fb/Light State = 67,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0[0m
[0;32mI (593196) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;32mI (599436) BLE2MQTT: Discovered iBeacon broadcaster[0m
[0;33mW (624566) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0008[0m
[0;32mI (624566) BLE: Connection closed, reason = 0x8[0m
[0;32mI (624566) BLE2MQTT: Disconnected from device: f8:24:41:c3:a3:2d[0m
[0;32mI (624706) BLE2MQTT: Discovered BLE device: f8:24:41:c3:a3:2d (RSSI: -61), connecting[0m
[0;32mI (627296) BLE2MQTT: Connected to device: f8:24:41:c3:a3:2d, scanning[0m
>>>> Unhandled characteristic type 0 <<<<
[0;32mI (627896) BLE2MQTT: Publishing: f8:24:41:c3:a3:2d/0000fe87-0000-1000-8000-00805f9b34fb/Light State = 67,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0[0m
[0;32mI (632776) BLE2MQTT: Discovered iBeacon broadcaster[0m

shmuelzon, 2020-12-27:
Hey,

This project uses decimal values and not hex values for bytes array. The types definition is only needed if you want the values to be parsed, e.g. if they are 16/32 bit, strings, etc.
So, if you want to send `4340010101f4` as HEX, you should publish `67,64,1,1,1,224` to `f8:24:41:c3:a3:2d/0000fe87-0000-1000-8000-00805f9b34fb/Light State`. `4340020101f4` would be `67,64,2,1,1,224`.

Hope that helps

shmuelzon, 2021-07-17:
Hey @lucs7,

I haven't heard back from you in a while so I'm assuming this either works for you or you gave up and I'm closing this ticket.
If you still have any question, feel free to reopen this ticket or start a new one.

Thanks!

closed 2021-07-17 by shmuelzon

redviper2100, 2022-05-24:
For those still struggling with (Xiaomi) Yeelight Candela, this is how I control it.

Configuration file:
```
{
  "network": {
    "hostname": "blegateway",
    "wifi": {
      "ssid": "MYWIFI",
      "password": "MyPass1234"
    }
  },
  "mqtt": {
    "server": {
      "host": "mymqttsrv",
      "port": 1883
    },
    "publish": {
      "retain": true
    },
    "topics": {
      "prefix": "ESP32/",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "ble": {
    "whitelist": [
      "f8:24:41:c0:de:f4"
    ],
    "services": {
      "definitions": {
        "0000fe87-0000-1000-8000-00805f9b34fb": {
          "name": "Service"
        }
      }
    },
    "characteristics": {
      "whitelist": [
        "aa7d3f34-2d4f-41e0-807f-52fbf8cf7443"
      ],
      "definitions": {
        "aa7d3f34-2d4f-41e0-807f-52fbf8cf7443": {
          "name": "State",
          "types": []
        }
      }
    }
  }
}
```

The Mqtt commands:
```
topic: ESP32/f8:24:41:c0:de:f4/Service/State/Set
Message:
67,64,2 //to turn off
67,66,x // x representing the brightness which is a number from 1 to 100; for a 50% dim should be: 67,66,50
```

Note that f8:24:41:c0:de:f4 is the address of the Candela