shmuelzon/esp32-ble2mqtt #90: Unable to connect to device, keep getting: Authentication failed, status: 0x66

rwl4, 2020-12-30:
I am unable to get my ESP32 to connect to my Opal Ice Maker. I am using esp-idf v4.1 with the latest from master as the README suggests. It connects, appears to try to scan, then returns with errors.

I have attached a snippet of the log. I just see what is in the log below printed repeatedly.

```
2020-12-30 00:24:04.060130 (BLE2MQTT-5E94.lan): I (3281336) BLE2MQTT: Discovered BLE device: d8:28:c9:10:46:34 (RSSI: -68), connecting
2020-12-30 00:24:06.563670 (BLE2MQTT-5E94.lan): I (3283916) BLE2MQTT: Connected to device: d8:28:c9:10:46:34, scanning
2020-12-30 00:24:07.439977 (BLE2MQTT-5E94.lan): W (3284526) BT_BTM: Security Manager: BTM_SetEncryption busy, enqueue request

2020-12-30 00:24:33.551553 (BLE2MQTT-5E94.lan): I (3310586) BLE2MQTT: Discovered BLE device: 7f:f1:b9:6c:99:44 (RSSI: -82), not connecting
2020-12-30 00:24:33.765834 (BLE2MQTT-5E94.lan): W (3311086) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0013
2020-12-30 00:24:33.765999 (BLE2MQTT-5E94.lan): W (3311096) BT_BTM: Security Manager: BTM_SetEncryption not connected

2020-12-30 00:24:33.766061 (BLE2MQTT-5E94.lan): E (3311096) BT_BTM: Device not found

2020-12-30 00:24:33.766097 (BLE2MQTT-5E94.lan): E (3311106) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: d8:28:c9:10:46:34
2020-12-30 00:24:33.766136 (BLE2MQTT-5E94.lan): E (3311116) BLE: Failed reading characteristic, status = 0x85
2020-12-30 00:24:33.766173 (BLE2MQTT-5E94.lan): I (3311116) BLE: Connection closed, reason = 0x13
2020-12-30 00:24:33.766342 (BLE2MQTT-5E94.lan): I (3311126) BLE2MQTT: Disconnected from device: d8:28:c9:10:46:34
2020-12-30 00:24:33.766404 (BLE2MQTT-5E94.lan): E (3311126) BLE: Authentication failed, status: 0x66
```

Here is my config file:

```
{
  "wifi": {
    "ssid": "XXXXXX",
    "password": "XXXXXXXX"
  },
  "mqtt": {
    "server": {
      "host": "192.168.0.52",
      "port": 1883,
      "username": "XXXXXXXX",
      "password": "XXXXXXXX"
    },
    "publish": {
      "retain": true
    }
  },
  "log": {
    "host": "192.168.0.132",
    "port": 5000
  },
  "ble": {
    "whitelist": [
      "d8:28:c9:10:46:34"
    ],
    "services": {
        "whitelist": [
          "3e6763c5-9429-40cc-909e-bebf8c7487be"
        ],
        "definitions": {
            "3e6763c5-9429-40cc-909e-bebf8c7487be": {
                "name": "DeviceService"
            }
        }
    },
    "characteristics" : {
        "whitelist": [
          "79994230-4b04-40cd-85c9-02dd1a8d4dd0"
        ],
        "definitions": {
           "79994230-4b04-40cd-85c9-02dd1a8d4dd0": {
                "name": "PowerStatus",
                "types": ["uint8"]
           }
        }
     }
  }
}
```

Any help would be greatly appreciated!

shmuelzon, 2020-12-30:
Hey,

The problem seems to be security related and at some point, probably because we don't do whatever security-related operation the ice maker expects, it disconnects from us.
I'm assuming you used a specific app on your phone to connect first, did you have some sort of pairing process you needed to do first?
Can you use a different app, e.g. nRF connect or LightBlue, to connect with the ice maker and see which characteristics it exposes? If it works, can you try the same from a phone that hasn't paired with the ice make in the past with the app?
I would also be happy for debug level logs, it will show a bit more information as to what was detected on the ice box.

It's a long list, sorry, but I hope it will help pinpoint the issue

rwl4, 2020-12-31:
I just used my wife's iPhone with LightBlue. Her phone has never connected to the ice maker. When I tapped on the device I received a pairing request. After I paired it was able to connect and I managed to fetch the device power status and also publish a message to it to have it power down, which it did. I then was able to turn it back on as well.

I will get you the debug logs next.

shmuelzon, 2020-12-31:
Sorry, didn't mean to close this issue.
I see you've enabled BLE debug log levels but I actually meant to enable the application logs.
Regardless, what the pairing process like? Do you have to enter a PIN code or something? Is it always the same?
In this application I initiate the pairing process when reading a characteristic fails with with insufficient authentication or encryption. Perhaps with your device it needs to be done before reading but the debug logs will help understand at what stage it fails.
If you do have a constant pairing key, you'll need to add that to the configuration file.

rwl4, 2020-12-31:
It does not ask for a PIN. In the case of my wife's iPhone which has never paired, I opened LightBlue, found the Opal item in the list, tapped on it, then it said it wanted to pair, I tapped OK and I had access to all the characteristics.

shmuelzon, 2020-12-31:
It's difficult to tell for sure without the application debug logs, but I think the issue is that we don't respond to security requests from the ice maker (we currently only initiate them when we notice it's needed).

Can you try the following patch at let me know if it helps?
```patch
diff --git a/main/ble.c b/main/ble.c
index 38768ba..6985649 100644
--- a/main/ble.c
+++ b/main/ble.c
@@ -593,6 +593,9 @@ static void gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)

         break;
     }
+    case ESP_GAP_BLE_SEC_REQ_EVT:
+        esp_ble_gap_security_rsp(param->ble_security.ble_req.bd_addr, true);
+        break;
     case ESP_GAP_BLE_PASSKEY_REQ_EVT:
         esp_ble_passkey_reply(param->ble_security.ble_req.bd_addr, true,
             on_passkey_requested_cb ?
```

rwl4, 2021-01-01:
I added the lines you mentioned. Now the MAC address topic is not being added to MQTT, where it was earlier. I am away from my house connecting from remote at the moment. But I will add that I was able to get EspruinoHub to connect without any special pairing process and was also able to make it read and write. But it didn't work very well with Home Assistant which is why I decided to buy some ESP32s and switch to your app.

I have attached the log with the results of your change.

[ble_log_2.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/5759423/ble_log_2.txt)

shmuelzon, 2021-01-01:
I'm sorry it's not working out for you but I'll do my best to get to the bottom of this.
It seems from the logs that the device resets at some point during the connection, that's probably why you don't see anything after the modifications. I guess it crashes from something but the lines I added were taken from ESP-IDF's examples so I think they're good. But, it also probably means that we do in fact get that event and, once handled correctly, things should work.

If you can, please enable the application debug logs by opening the configuration menu with `idf.py menuconfig`. Search, using the `/` key, for the `LOG_DEFAULT_LEVEL` choice menu and select `DEBUG`.
Regarding the crash, can you use the serial connection for capturing the logs? We might get a crash dump to tell us where the problem is.

rwl4, 2021-01-01:
I have attached the log from the serial port with the log level set to DEBUG as you asked.

[serial.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/5759530/serial.log)

Thank you!

shmuelzon, 2021-01-02:
Thanks for that!
The crash I saw earlier seems to a stack corruption in the BT task:
```
***ERROR*** A stack overflow in task btuT has been detected.
abort() was called at PC 0x4008d6f0 on core 0
```
According to [this](https://github.com/espressif/esp-idf/issues/2503), it's probably because of the Bluetooth debug logs you've enabled earlier, can you restore those to default, leaving only the application debug levels set last time, so we can continue to fix the original pairing issue?

rwl4, 2021-01-02:
OMG. That fixed it. I am able to get and publish and control my ice maker now. I **really** appreciate it and would like to send you a tip. Is there a place I can send you a tip?

shmuelzon, 2021-01-03:
Glad to hear it works for you now.
Thanks for the generous offer but I'm really not set up for that kind of thing.

commit c84c28ca (2021-01-03): Accept security requests from peripherals

Fixes #90
closed 2021-01-03 by shmuelzon