shmuelzon/esp32-ble2mqtt #91: Are bin files possible?

anarchking, 2021-01-04:
Ive never worked with esp idf, Is It possible to Make bin files of the releases So that way people can test and give more feedback that don't work with esp-idf?

shmuelzon, 2021-01-04:
It's on the roadmap :)
#46 and #66 were recently implemented to allow just this as without them you won't be able to do anything once you would be able to flash a binary image.
There are, however, two things (at least) that need to be done before I can start posting images along with releases.
The first is the option modify the configuration file via the web interface. Right now it's only possible to upload a file-system image but you'll need ESP-IDF to generate that...
The second is a utility to combine all of the required images (bootloader, partition table and application) into a single image that one can flash together.

After all of that's done you'll still need to install the esptool.py utility to actually flash over serial the ESP32 but that should be as simple as installing a PIP package.

shmuelzon, 2021-07-15:
Hey,

It took a (long) while, but from now on, releases will include a pre-built binary you can directly flash on your device without the need for ESP-IDF.
See https://github.com/shmuelzon/esp32-ble2mqtt/releases

Good day!

closed 2021-07-15 by shmuelzon