shmuelzon/esp32-ble2mqtt #92: How to disconnect ble device after publishing data?

Cirobar, 2021-01-13:
Hi there, first of all, congrats on the project (very well done). Well, i have been trying to disconnect from the device through function "ble_disconnect(mac)" after it has already published a characteristic or service of interest, and then esp connects to another device in line/scanned. 
Im having some issues since i don't know for sure where would be the best place to call it.  So, if i want it to publish the characteristics inside the service "Device Information" and then disconnect, for example, what would be the best way to do it?

shmuelzon, 2021-01-13:
To be honest, this project is meant for continuous connections and expects the BLE peripherals to notify when something changes and not poll devices every so often. Mostly to conserve the battery life of the BLE peripheral.

If you do want to disconnect, you can do that [here](https://github.com/shmuelzon/esp32-ble2mqtt/blob/c84c28ca24c1b5314648409bee133c85c26a21e4/main/ble2mqtt.c#L454), after the value was published. You should have all the information you need to decide if you should disconnect (you have the MAC address, service and characteristic UUIDs).
The problem is that after you disconnect, this application will try to connect to it again when it re-discovers it and I'm assuming that's not what you want. But then you're left with defining when you should connect to it again to get an updated value.
The other "problem" is that the BLE wrapper code also caches the devices it finds so it won't bug you about them every time they're discovered. So if you modify [ble_on_device_discovered()](https://github.com/shmuelzon/esp32-ble2mqtt/blob/c84c28ca24c1b5314648409bee133c85c26a21e4/main/ble2mqtt.c#L300) to not connect after you already got a value at one time, it won't ask if it should connect again*.

*It actually will... See, saving a cache of all the devices ever seen will take up memory. And most smart phone will randomize their MAC address (to avoid them being tracked). So, even if you don't get a lot of BLE traffic near you, one phone could fill up the ESP's memory over time. To mitigate this, (every hour)[https://github.com/shmuelzon/esp32-ble2mqtt/blob/c84c28ca24c1b5314648409bee133c85c26a21e4/main/ble.c#L835] I purge the cache and remove all non-connected devices to free up memory. If your device is discovered after the cache was cleared, it will call `ble_on_device_discovered()` again.

If you can provide a bit more information as to what you're trying to achieve, perhaps I can pinpoint you further, or come up with a method that could be generic enough and added to this project.

Cirobar, 2021-01-13:
Thank you very much for answering. I'm trying to use it as a presence detector device. In order to explain my purposes better, let's suppose that i have 3 Bluetooth devices around my esp, i need it to connect to one of them and publish some characteristics to the MQTT server, then it disconnects from device 1, connects to the next device and so on. There's actually no problem in connecting to the same device again, since the most important standard to define if it will connect again, in that case, would be the RSSI. But for now i would be satisfied if i could just order the stack so it would connect and disconnect from device 1, connect and disconnect from device 2 and then connect/disconnect from device 3. 
At my first attempt, i called ble_disconnect_all() inside uptime_publish(), seemed good enough looking at the mqtt pubs when i was testing with 2 devices, but i can't say for sure if it was ok to do it as i did. Also, by doing that i could not choose what characteristics would be published.

shmuelzon, 2021-01-14:
I'm not sure which BLE devices you have, but from a presence perspective, I think you would be better off with a broadcaster of some sort, e.g. an iBeacon or Eddystone. With these, you'll see publications on the MQTT bus whenever they publish something along with their RSSI. My plan was to use this information gathered from multiple ESP's to "triangulate" a location a detect the room the device is in, but, lack of time... :)
Now, even if your device isn't a typical broadcaster, so long as you can identify it from its advertisement packet, you can still implement your own broadcaster type in `main/broadcasters.c` so it will publish some data and RSSI. In this case, you can set an empty whitelist in your configuration file and the application won't try to connect to any device at all.

Anyways, if you prefer to continue with the current path, the `uptime_publish()` function is simply called every minute so you can use it to just disconnect periodically from every device. You can't enforce the order in which they connect again as it depends which device is detected first. As for which characteristics to publish, in the configuration file you can also whitelist or blacklist services and characteristics so you can use those set what you want published on the MQTT bus.

Cirobar, 2021-01-14:
I was about to do something similar to your original goals on finding the room the device is in, considering that there is mutiple devices and multiple esps too. But i thought it would be necessary to order the stack on ble connection, because the devices can be taken to another room, so it needs to be "triangulated" again. Also the device might be out of range from some esps and the closer esp to the device must connect to it. There is a way to enforce the order or do you think there is better way to do all this?

shmuelzon, 2021-01-14:
I bought some cheap iBeacon devices for this and was planning to use them only as broadcasters and not connect to them at all. This has the added value of having multiple ESP devices in range receiving the same broadcasted message.
Then, some service listening on the MQTT bus could log which ESP saw which device and what was the calculated distance from the ESP, combining this with some prior knowledge of where the ESPs were physically located could allow calculating an approximate location.
With broadcasters you won't have the need to connect in a specific order and you won't have the limitation that if one ESP is connected to the BLE device, no other ESP can connect at the same time.

Anyways, if you do decide to go ahead with a project like this, please let me know and I'll try to contribute to it.

Cirobar, 2021-01-15:
The devices being used here are those Itag cheap devices, however, they have some peculiarities that improves the project. How do i contact you privately so i can explain to you?

closed 2021-01-18 by Cirobar