shmuelzon/esp32-ble2mqtt #95: rebooting esp32 after bug

Yury777, 2021-01-26:
I (31363) BLE2MQTT: Connected to device: e8:6a:98:80:d5:3b, scanning
I (31993) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/CS/Name = flokk6
I (32083) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/ES/Temperature = 26,21
I (32193) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/ES/Pressure = 981,39
I (32283) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/ES/Humidity = 39
I (32483) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/ES/ConfigW = 60000,60000,60000,60000,3,103,78,29
I (32593) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/ES/Light = 1474,2626,1517,342
I (32603) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/UIS/LED = 1,0,200,0,0
I (32693) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/UIS/EXT = 0,0,0,0
I (32803) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/SS/Sound = 1,1
I (32893) BLE2MQTT: Publishing: e8:6a:98:80:d5:3b/BS/Battery = 81
I (33983) BLE2MQTT: Discovered Eddystone broadcaster
I (34063) BLE2MQTT: Discovered BLE device: a4:c1:38:96:8e:a5 (RSSI: 229376), not connecting
I (35943) BLE2MQTT: Discovered BLE device: 52:8e:e0:04:83:87 (RSSI: 0), not connecting
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060530  A0      : 0x801b1ed8  A1      : 0x3ffeefc0
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffeef70
A10     : 0x00000000  A11     : 0x3f40362e  A12     : 0x3ffef204  A13     : 0x3ffb65fc
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: fc404808e77364af

Backtrace: 0x400014fa:0x3ffeefc0 0x401b1ed5:0x3ffeefd0 0x401b3525:0x3ffef2e0 0x401bf1ae:0x3ffef310 0x400911f5:0x3ffef340 0x400d7865:0x3ffef390
0x401b1ed5: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528

0x401b3525: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34

0x401bf1ae: esp_log_writev at /home/yury/esp/esp-idf/components/log/log.c:189

0x400911f5: esp_log_write at /home/yury/esp/esp-idf/components/log/log.c:199

0x400d7865: ble_on_broadcaster_discovered at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:289
 (inlined by) ble2mqtt_handle_event at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:545
 (inlined by) ble2mqtt_task at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:589

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (29) boot: ESP-IDF v4.1 2nd stage bootloader
I (29) boot: compile time 22:02:46
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (99) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (106) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (114) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (121) boot: End of partition table
I (126) boot_comm: chip revision: 1, min. application chip revision: 0
I (133) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x41694 (267924) map
I (244) esp_image: segment 1: paddr=0x000516bc vaddr=0x3ffbdb60 size=0x064a4 ( 25764) load
I (255) esp_image: segment 2: paddr=0x00057b68 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (255) esp_image: segment 3: paddr=0x00057f74 vaddr=0x40080404 size=0x080a4 ( 32932) load
I (276) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xef240 (979520) map
0x400d0020: _stext at ??:?

I (649) esp_image: segment 5: paddr=0x0014f268 vaddr=0x400884a8 size=0x13368 ( 78696) load
0x400884a8: vhci_recv at ??:?

I (699) boot: Loaded app from partition at offset 0x10000
I (699) boot: Disabling RNG early entropy source...
I (699) cpu_start: Pro cpu up.
I (703) cpu_start: Application information:
I (708) cpu_start: Project name:     ble2mqtt
I (713) cpu_start: App version:      v0.13.0-5-gc84c28c
I (719) cpu_start: Compile time:     Jan 20 2021 22:02:35
I (725) cpu_start: ELF file SHA256:  fc404808e77364af...
I (731) cpu_start: ESP-IDF:          v4.1
I (735) cpu_start: Starting app cpu, entry point is 0x400812cc
0x400812cc: call_start_cpu1 at /home/yury/esp/esp-idf/components/esp32/cpu_start.c:271

I (728) cpu_start: App cpu up.
I (746) heap_init: Initializing. RAM available for dynamic allocation:
I (753) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (759) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (765) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (771) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (777) heap_init: At 3FFD3690 len 0000C970 (50 KiB): DRAM
I (783) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (790) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (796) heap_init: At 4009B810 len 000047F0 (17 KiB): IRAM
I (802) cpu_start: Pro cpu start user code
I (821) spi_flash: detected chip: generic
I (821) spi_flash: flash io: dio
I (821) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (894) BLE2MQTT: Version: v0.13.0-5-gc84c28c
I (904) Config: Initializing configuration
I (1014) Config: version: a3e3ec6a439e8daaea9be157f81627c732af69a36923c7505b7589a9b18a0ade
I (1014) Log: Initializing remote logging
I (1014) OTA: Initializing OTA
I (1034) wifi:wifi driver task: 3ffd50a4, prio:23, stack:6656, core=0
I (1034) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1034) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1064) wifi:wifi firmware version: 3ea4c76
I (1064) wifi:config NVS flash: enabled
I (1064) wifi:config nano formating: disabled
I (1064) wifi:Init dynamic tx buffer num: 32
I (1064) wifi:Init data frame dynamic rx buffer num: 32
I (1074) wifi:Init management frame dynamic rx buffer num: 32
I (1074) wifi:Init management short buffer num: 32
I (1084) wifi:Init static rx buffer size: 1600
I (1084) wifi:Init static rx buffer num: 10
I (1094) wifi:Init dynamic rx buffer num: 32
I (1094) Resolve: Initializing resolver
I (1104) BTDM_INIT: BT controller compile version [219866f]
I (1104) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1204) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1594) HTTPD: Initializing HTTP server
I (1604) WiFi: Connecting to SSID ap1_2
I (1614) wifi:mode : sta (24:6f:28:55:5a:dc)
I (2344) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (3234) wifi:state: init -> auth (b0)
I (3284) wifi:state: auth -> assoc (0)
I (3324) wifi:state: assoc -> run (10)
I (3344) wifi:connected with ap1_2, aid = 6, channel 6, BW20, bssid = 28:d1:27:80:4a:f9
I (3344) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (3354) wifi:pm start, type: 1

I (3354) WiFi: Connected
I (3424) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4014) esp_netif_handlers: sta ip: 10.10.4.35, mask: 255.255.255.0, gw: 10.10.4.1
I (4014) BLE2MQTT: Connected to the network, connecting to MQTT
I (4014) MQTT: Connecting MQTT client
I (4064) MQTT: MQTT client connected
I (4064) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (5054) BLE2MQTT: Discovered BLE device: 34:69:76:39:55:e1 (RSSI: -52), not connecting
I (5134) BLE2MQTT: Discovered BLE device: eb:b1:58:4f:fb:15 (RSSI: -84), not connecting
I (5454) BLE2MQTT: Discovered BLE device: 80:e6:50:ed:e9:c5 (RSSI: -83), not connecting
I (5854) BLE2MQTT: Discovered BLE device: 28:ff:3c:98:8f:94 (RSSI: -84), not connecting
I (8014) BLE2MQTT: Discovered BLE device: 17:d6:51:19:66:10 (RSSI: -88), not connecting

shmuelzon, 2021-01-26:
Thanks for the info. Indeed something is wrong here but, unfortunately, it doesn't seem straight forward.
The line that crashes should never do that as it's supposed to reference a static structure with a static string.
My hunch is that there's some memory corruption going on, specially with the RSSI values printed here:
```
I (34063) BLE2MQTT: Discovered BLE device: a4:c1:38:96:8e:a5 (RSSI: 229376), not connecting
I (35943) BLE2MQTT: Discovered BLE device: 52:8e:e0:04:83:87 (RSSI: 0), not connecting
```
Now, these come after a device you've connected to and after an Eddystone beacon.
Can you reproduce this bug easily? Does it always happen shortly after you connect to `e8:6a:98:80:d5:3b`? Or shortly after the Eddystone beacon is detected?
Regardless, If you can provide me with a debug-level log, it will help me understand what flows happened before the crash and I can try to pinpoint where the problematic code is.

Yury777, 2021-02-07:
Thank you for this wonderful program!
It was found out that the reboot occurs after sending data from mqtt to a ble device with an incorrect structure.
Thingy52 connects with its own mac address and simultaneously transmits values from the Eddystone beacon with another mac address that I blacklisted.

shmuelzon, 2021-02-07:
> It was found out that the reboot occurs after sending data from mqtt to a ble device with an incorrect structure.

Can you please elaborate more on this? Do you mean that you try to write to a characteristic with a wrong format and that causes the ESP to crash? Which characteristic? How is it defined in the configuration file? What payload are you sending on the MQTT?
In any case, it shouldn't crash so if there's a scenario I can reproduce, I would like to make sure it doesn't crash the application.

> Thingy52 connects with its own mac address and simultaneously transmits values from the Eddystone beacon with another mac address that I blacklisted.

Broadcasters, such as the Eddystone beacon, don't follow the white/black list. It was intentional at the time as it doesn't "cost" much to handle them and some broadcasters would change their MAC address and one needs to rely on the payload it self.

Yury777, 2021-02-11:
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (28) boot: ESP-IDF v4.1 2nd stage bootloader
I (28) boot: compile time 22:02:46
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (98) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (105) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (113) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (120) boot: End of partition table
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x41694 (267924) map
I (243) esp_image: segment 1: paddr=0x000516bc vaddr=0x3ffbdb60 size=0x064a4 ( 25764) load
I (253) esp_image: segment 2: paddr=0x00057b68 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (254) esp_image: segment 3: paddr=0x00057f74 vaddr=0x40080404 size=0x080a4 ( 32932) load
I (275) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xef240 (979520) map
0x400d0020: _stext at ??:?

I (648) esp_image: segment 5: paddr=0x0014f268 vaddr=0x400884a8 size=0x13368 ( 78696) load
0x400884a8: vhci_recv at ??:?

I (698) boot: Loaded app from partition at offset 0x10000
I (698) boot: Disabling RNG early entropy source...
I (698) cpu_start: Pro cpu up.
I (702) cpu_start: Application information:
I (707) cpu_start: Project name:     ble2mqtt
I (712) cpu_start: App version:      v0.13.0-5-gc84c28c
I (718) cpu_start: Compile time:     Jan 20 2021 22:02:35
I (724) cpu_start: ELF file SHA256:  fc404808e77364af...
I (730) cpu_start: ESP-IDF:          v4.1
I (734) cpu_start: Starting app cpu, entry point is 0x400812cc
0x400812cc: call_start_cpu1 at /home/yury/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (745) heap_init: Initializing. RAM available for dynamic allocation:
I (752) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (758) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (764) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (770) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (776) heap_init: At 3FFD3690 len 0000C970 (50 KiB): DRAM
I (782) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (789) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (795) heap_init: At 4009B810 len 000047F0 (17 KiB): IRAM
I (801) cpu_start: Pro cpu start user code
I (819) spi_flash: detected chip: gd
[... 217 lines cut ...]
I (4993) BLE2MQTT: Discovered BLE device: 53:a7:da:30:1b:7d (RSSI: -75), not connecting
I (5393) BLE2MQTT: Discovered BLE device: 5b:31:5d:53:17:95 (RSSI: -80), not connecting
I (6023) BLE2MQTT: Discovered BLE device: 0a:5b:8e:cf:65:27 (RSSI: -81), not connecting
I (6023) BLE2MQTT: Discovered BLE device: 43:fb:b8:60:17:b1 (RSSI: -51), not connecting
I (6103) BLE2MQTT: Discovered BLE device: 5d:ad:3e:64:fc:2d (RSSI: -36), not connecting
I (6103) BLE2MQTT: Discovered BLE device: 28:ff:3c:98:8f:94 (RSSI: -79), not connecting
I (7233) BLE2MQTT: Discovered BLE device: eb:b1:58:4f:fb:15 (RSSI: -83), not connecting
I (8273) BLE2MQTT: Discovered BLE device: 5b:06:46:29:ca:53 (RSSI: -77), not connecting
I (8353) BLE2MQTT: Discovered BLE device: 1f:e7:03:97:d6:84 (RSSI: -57), not connecting
I (8353) BLE2MQTT: Discovered BLE device: 75:4a:08:2e:60:ee (RSSI: -86), not connecting
I (8913) BLE2MQTT: Discovered BLE device: 80:e6:50:ed:e9:c5 (RSSI: -85), not connecting
I (9153) BLE2MQTT: Discovered BLE device: a4:c1:38:96:8e:a5 (RSSI: -74), not connecting
I (10193) BLE2MQTT: Discovered BLE device: a4:c1:38:51:77:01 (RSSI: -59), not connecting
I (10433) BLE2MQTT: Discovered BLE device: e8:d2:21:2f:61:58 (RSSI: -66), connecting
I (14013) BLE2MQTT: Connected to device: e8:d2:21:2f:61:58, scanning
I (14703) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/CS/Name = flokk4
I (14733) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Temperature = 26,38
I (14823) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Pressure = 994,50
I (14913) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Humidity = 46
I (15093) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/ConfigW = 60000,60000,60000,60000,3,103,78,29
I (15163) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/UIS/LED = 0,0,88,2,0
I (15193) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/UIS/EXT = 0,0,0,0
I (15213) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Light = 1770,2777,2509,379
I (15223) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/SS/Sound = 1,1
I (15273) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/BS/Battery = 84

Yury777, 2021-02-11:
I sent to mqtt e8:d2:21:2f:61:58/US/LED/Set the value of 1,0,72,0,0

shmuelzon, 2021-02-11:
Thanks! Can you please also post your configuration file, at least the part that includes the BLE services and characteristics definitions? I guess I'm not passing something correctly there.

Yury777, 2021-02-11:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (28) boot: ESP-IDF v4.1 2nd stage bootloader
I (28) boot: compile time 22:02:46
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (98) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (105) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (113) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (120) boot: End of partition table
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x41694 (267924) map
I (243) esp_image: segment 1: paddr=0x000516bc vaddr=0x3ffbdb60 size=0x064a4 ( 25764) load
I (253) esp_image: segment 2: paddr=0x00057b68 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (254) esp_image: segment 3: paddr=0x00057f74 vaddr=0x40080404 size=0x080a4 ( 32932) load
I (275) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xef240 (979520) map
0x400d0020: _stext at ??:?

I (648) esp_image: segment 5: paddr=0x0014f268 vaddr=0x400884a8 size=0x13368 ( 78696) load
0x400884a8: vhci_recv at ??:?

I (698) boot: Loaded app from partition at offset 0x10000
I (698) boot: Disabling RNG early entropy source...
I (698) cpu_start: Pro cpu up.
I (702) cpu_start: Application information:
I (707) cpu_start: Project name:     ble2mqtt
I (712) cpu_start: App version:      v0.13.0-5-gc84c28c
I (718) cpu_start: Compile time:     Jan 20 2021 22:02:35
I (724) cpu_start: ELF file SHA256:  fc404808e77364af...
I (730) cpu_start: ESP-IDF:          v4.1
I (734) cpu_start: Starting app cpu, entry point is 0x400812cc
0x400812cc: call_start_cpu1 at /home/yury/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (745) heap_init: Initializing. RAM available for dynamic allocation:
I (752) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (758) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (764) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (770) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (776) heap_init: At 3FFD3690 len 0000C970 (50 KiB): DRAM
I (782) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (789) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (795) heap_init: At 4009B810 len 000047F0 (17 KiB): IRAM
I (801) cpu_start: Pro cpu start user code
I (819) spi_flash: detected chip: gd
I (820) spi_flash: flash io: dio
I (820) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (891) BLE2MQTT: Version: v0.13.0-5-gc84c28c
[... 1513 lines cut ...]
I (5213) BLE2MQTT: Discovered Eddystone broadcaster
I (5223) BLE2MQTT: Discovered BLE device: e8:d2:21:2f:61:5a (RSSI: -69), not connecting
I (5613) BLE2MQTT: Discovered BLE device: 80:e6:50:ed:e9:c5 (RSSI: -82), not connecting
I (6013) BLE2MQTT: Discovered BLE device: 55:29:0c:c5:d8:45 (RSSI: -47), not connecting
I (6333) BLE2MQTT: Discovered BLE device: 21:d3:16:4f:f3:d9 (RSSI: -55), not connecting
I (7693) BLE2MQTT: Discovered BLE device: e8:d2:21:2f:61:58 (RSSI: -69), connecting
I (8163) BLE2MQTT: Discovered BLE device: 7b:bc:58:da:db:6d (RSSI: -42), not connecting
I (8253) BLE2MQTT: Discovered BLE device: d0:03:4b:e7:b7:3a (RSSI: -86), not connecting
I (9213) BLE2MQTT: Discovered BLE device: a4:c1:38:51:77:01 (RSSI: -62), not connecting
I (9283) BLE2MQTT: Discovered BLE device: 49:c6:58:70:6f:e2 (RSSI: -68), not connecting
I (9683) BLE2MQTT: Discovered BLE device: 7b:7c:d2:99:b4:30 (RSSI: -88), not connecting
I (9843) BLE2MQTT: Discovered BLE device: eb:b1:58:4f:fb:15 (RSSI: -88), not connecting
I (10323) BLE2MQTT: Discovered BLE device: 73:f2:5c:ea:c6:07 (RSSI: -78), not connecting
I (11053) BLE2MQTT: Discovered BLE device: 28:ff:3c:98:8f:94 (RSSI: -75), not connecting
I (11373) BLE2MQTT: Connected to device: e8:d2:21:2f:61:58, scanning
I (11973) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/CS/Name = flokk4
I (12063) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Temperature = 26,58
I (12153) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Pressure = 994,13
I (12183) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Humidity = 48
I (12423) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/ConfigW = 60000,60000,60000,60000,3,103,78,29
I (12553) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/ES/Light = 1765,2769,2507,380
I (12573) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/UIS/LED = 0,0,88,2,0
I (12663) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/UIS/EXT = 0,0,0,0
I (12693) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/SS/Sound = 1,1
I (12783) BLE2MQTT: Publishing: e8:d2:21:2f:61:58/BS/Battery = 84

Yury777, 2021-02-11:
{
  "wifi": {
    "ssid": "mywifi",
    "password": "password"
  },
  "mqtt": {
    "server": {
      "host": "192.168.21.2",
      "port": 1883,
      "username": "user",
      "password": "password",
      "client_id": "esp2"
    },
    "publish": {
      "retain": true
    },
    "topics" :{
      "prefix": "",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "ble": {
    "blacklist": [
	"e8:6a:98:80:d5:3d"
    ],

    "whitelist": [
	"da:a6:47:1c:6e:a2",
	"C5:4A:D2:94:7D:11",
	"db:ec:fb:ab:e3:ce",
	"E8:D2:21:2F:61:58",
	"f6:94:bc:3c:5e:39",
	"e8:6a:98:80:d5:3b"
    ],
    "services": {
      "whitelist": [
        "ef680100-9b35-4933-9b10-52ffa9740042",
        "ef680200-9b35-4933-9b10-52ffa9740042",
        "ef680300-9b35-4933-9b10-52ffa9740042",
        "ef680500-9b35-4933-9b10-52ffa9740042",
        "0000180F-0000-1000-8000-00805f9b34fb"
      ],
      "definitions": {
        "ef680100-9b35-4933-9b10-52ffa9740042": {
          "name": "CS"
        },
        "ef680200-9b35-4933-9b10-52ffa9740042": {
          "name": "ES"
        },
        "ef680300-9b35-4933-9b10-52ffa9740042": {
          "name": "UIS"
        },
        "ef680400-9b35-4933-9b10-52ffa9740042": {
          "name": "MS"
        },
        "ef680500-9b35-4933-9b10-52ffa9740042": {
          "name": "SS"
        },
        "0000180F-0000-1000-8000-00805f9b34fb": {
          "name": "BS"
        }
      }
    },
    "characteristics": {
      "whitelist": [
	"00002a19-0000-1000-8000-00805f9b34fb",
        "ef680101-9b35-4933-9b10-52ffa9740042",
        "ef680201-9b35-4933-9b10-52ffa9740042",
        "ef680202-9b35-4933-9b10-52ffa9740042",
        "ef680203-9b35-4933-9b10-52ffa9740042",
        "ef680204-9b35-4933-9b10-52ffa9740042",
        "ef680205-9b35-4933-9b10-52ffa9740042",
	"ef680206-9b35-4933-9b10-52ffa9740042",        
	"ef680301-9b35-4933-9b10-52ffa9740042",
        "ef680302-9b35-4933-9b10-52ffa9740042",
	"ef680303-9b35-4933-9b10-52ffa9740042",
        "ef680501-9b35-4933-9b10-52ffa9740042",
        "ef680502-9b35-4933-9b10-52ffa9740042"
      ],
      "definitions": {
        "00002a19-0000-1000-8000-00805f9b34fb": {
          "name": "Battery",
          "types": [
            "uint8"
          ]
        },
        "ef680206-9b35-4933-9b10-52ffa9740042": {
          "name": "ConfigW",
          "types": [
            "uint16", "uint16", "uint16", "uint16",  "uint8",  "uint8", "uint8", "uint8"
          ]
        },
        "ef680303-9b35-4933-9b10-52ffa9740042": {
          "name": "EXT",
          "types": [
            "uint8", "uint8", "uint8", "uint8"
          ]
        },

        "ef680101-9b35-4933-9b10-52ffa9740042": {
          "name": "Name",
          "types": [
            "utf8s"
          ]
        },
        "ef680201-9b35-4933-9b10-52ffa9740042": {
          "name": "Temperature",
          "types": [
            "sint8",
            "uint8"
          ]
        },
        "ef680202-9b35-4933-9b10-52ffa9740042": {
          "name": "Pressure",
          "types": [
            "sint32",
            "uint8"
          ]
        },
        "ef680203-9b35-4933-9b10-52ffa9740042": {
          "name": "Humidity",
          "types": [
            "uint8"
          ]
        },
        "ef680204-9b35-4933-9b10-52ffa9740042": {
          "name": "Air",
          "types": [
            "uint16",
            "uint16"
          ]
        },
        "ef680205-9b35-4933-9b10-52ffa9740042": {
          "name": "Light",
          "types": [
            "uint16",
            "uint16",
            "uint16",
            "uint16"
          ]
        },
        "ef680301-9b35-4933-9b10-52ffa9740042": {
          "name": "LED",
          "types": [
            "uint8",
            "uint8",
            "uint8",
            "uint8"
          ]
        },
        "ef680302-9b35-4933-9b10-52ffa9740042": {
          "name": "Button",
          "types": [
            "boolean"
          ]
        },
        "ef680501-9b35-4933-9b10-52ffa9740042": {
          "name": "Sound",
          "types": [
            "uint8",
            "uint8"
          ]
        },
        "ef680502-9b35-4933-9b10-52ffa9740042": {
          "name": "Speaker",
          "types": [
            "uint8"
          ]
        }
      }
    }
  }
}

Yury777, 2021-02-11:
I deleted the mqtt database, I do not send anything... esp32 is rebooted regularly :-(

shmuelzon, 2021-02-12:
Hey,

I think I found the problem which could, indeed, cause a memory overrun.
Can you please make the following changes and run it again? If it will fix your crashes, I'll commit it:
```patch
diff --git a/main/broadcasters.c b/main/broadcasters.c
index 0de7085..f7caa77 100644
--- a/main/broadcasters.c
+++ b/main/broadcasters.c
@@ -142,7 +142,7 @@ typedef struct {
         eddystone_url_t url;
         eddystone_tlm_t tlm;
     } u;
-} eddystone_t;
+} __attribute__((packed)) eddystone_t;

 static eddystone_t *eddystone_data_get(uint8_t *adv_data, uint8_t adv_data_len,
     uint8_t *eddystone_len)
```

Thanks!

Yury777, 2021-02-14:
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (28) boot: ESP-IDF v4.1 2nd stage bootloader
I (28) boot: compile time 22:02:46
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (98) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (105) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (113) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (120) boot: End of partition table
I (125) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x41694 (267924) map
I (243) esp_image: segment 1: paddr=0x000516bc vaddr=0x3ffbdb60 size=0x064a4 ( 25764) load
I (254) esp_image: segment 2: paddr=0x00057b68 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (255) esp_image: segment 3: paddr=0x00057f74 vaddr=0x40080404 size=0x080a4 ( 32932) load
I (275) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xef27c (979580) map
0x400d0020: _stext at ??:?

I (648) esp_image: segment 5: paddr=0x0014f2a4 vaddr=0x400884a8 size=0x13368 ( 78696) load
0x400884a8: vhci_recv at ??:?

I (698) boot: Loaded app from partition at offset 0x10000
I (698) boot: Disabling RNG early entropy source...
I (699) cpu_start: Pro cpu up.
I (702) cpu_start: Application information:
I (707) cpu_start: Project name:     ble2mqtt
I (712) cpu_start: App version:      v0.13.0-5-gc84c28c
I (718) cpu_start: Compile time:     Jan 20 2021 22:02:35
I (724) cpu_start: ELF file SHA256:  83e7d30d870c43ce...
I (730) cpu_start: ESP-IDF:          v4.1
I (735) cpu_start: Starting app cpu, entry point is 0x400812cc
0x400812cc: call_start_cpu1 at /home/yury/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (745) heap_init: Initializing. RAM available for dynamic allocation:
I (752) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (758) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (764) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (770) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (776) heap_init: At 3FFD3690 len 0000C970 (50 KiB): DRAM
I (783) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (789) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (795) heap_init: At 4009B810 len 000047F0 (17 KiB): IRAM
I (802) cpu_start: Pro cpu start user code
I (820) spi_flash: detected chip: gd
[... 201 lines cut ...]
I (1623) wifi:mode : sta (24:0a:c4:0c:b9:60)
I (1753) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2633) wifi:state: init -> auth (b0)
I (2683) wifi:state: auth -> assoc (0)
I (2723) wifi:state: assoc -> run (10)
I (2753) wifi:connected with mywifi, aid = 7, channel 1, BW20, bssid = 28:d1:27:80:4a:f9
I (2753) wifi:security: WPA2-PSK, phy: bgn, rssi: -41
I (2763) wifi:pm start, type: 1

I (2763) WiFi: Connected
I (2853) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4513) esp_netif_handlers: sta ip: 192.168.1.51, mask: 255.255.255.0, gw: 192.168.1.1
I (4513) BLE2MQTT: Connected to the network, connecting to MQTT
I (4513) MQTT: Connecting MQTT client
I (4643) MQTT: MQTT client connected
I (4643) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (4673) BLE2MQTT: Discovered BLE device: 28:ff:3c:98:8f:94 (RSSI: -87), not connecting
I (5313) BLE2MQTT: Discovered BLE device: 4a:9a:f3:84:12:1f (RSSI: -64), not connecting
I (5553) BLE2MQTT: Discovered BLE device: 5c:30:82:16:6a:74 (RSSI: -90), not connecting
I (6513) BLE2MQTT: Discovered BLE device: 58:1a:3c:e5:bd:48 (RSSI: -47), not connecting
I (7313) BLE2MQTT: Discovered BLE device: 4c:eb:ce:de:d2:35 (RSSI: -58), not connecting
I (7793) BLE2MQTT: Discovered BLE device: 1e:b0:34:29:30:c5 (RSSI: -48), not connecting
I (8193) BLE2MQTT: Discovered BLE device: 80:e6:50:ed:e9:c5 (RSSI: -80), not connecting
I (10193) BLE2MQTT: Discovered BLE device: 0a:5b:8e:cf:65:27 (RSSI: -87), not connecting
I (11953) BLE2MQTT: Discovered BLE device: c5:4a:d2:94:7d:11 (RSSI: -60), connecting

shmuelzon, 2021-02-15:
So, I take it that didn't help? :)
Can you please checkout the `memory_override_debug` branch and run that? It will dump out the broadcaster's raw advertisement data and I would use that to reproduce the problem here and hope to find the issue.

Yury777, 2021-02-16:
How do I do this? :-)

shmuelzon, 2021-02-16:
If you've used the git command line to clone this project, you can run `git fetch && git checkout memory_override_debug`.
If you downloaded a tar/zip file, you can get it from: https://github.com/shmuelzon/esp32-ble2mqtt/tree/memory_override_debug

I can send you just the changes I've done, but I thought it would be simpler like this.

Yury777, 2021-02-25:
error: Your local changes to the following files would be overwritten by checkout:
        main/broadcasters.c
Please commit your changes or stash them before you switch branches.
Aborting

shmuelzon, 2021-02-25:
That's probably because of the changes from earlier, run first:
```bash
git reset --hard origin/master
```

Yury777, 2021-02-25:
I (214822) BLE2MQTT: Discovered Eddystone broadcaster
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014e8  PS      : 0x00060f30  A0      : 0x801b1f48  A1      : 0x3ffeeed0
A2      : 0x90d4000a  A3      : 0x90d40008  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x801ade76  A9      : 0x3ffeee80
A10     : 0x00000000  A11     : 0x3f40362e  A12     : 0x3ffef114  A13     : 0x3ffb65fc
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x90d40008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

ELF file SHA256: 63110af333a6ad24

Backtrace: 0x400014e5:0x3ffeeed0 0x401b1f45:0x3ffeeee0 0x401b3595:0x3ffef1f0 0x401bf21e:0x3ffef220 0x400911f5:0x3ffef250 0x400d783d:0x3ffef2a0
0x401b1f45: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528

0x401b3595: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34

0x401bf21e: esp_log_writev at /home/yury/esp/esp-idf/components/log/log.c:189

0x400911f5: esp_log_write at /home/yury/esp/esp-idf/components/log/log.c:199

0x400d783d: ble_on_broadcaster_discovered at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:289
 (inlined by) ble2mqtt_handle_event at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:545
 (inlined by) ble2mqtt_task at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:589

Rebooting...

Yury777, 2021-02-25:
I (702) cpu_start: Application information:
I (707) cpu_start: Project name:     ble2mqtt
I (712) cpu_start: App version:      v0.13.0-5-gc84c28c
I (718) cpu_start: Compile time:     Jan 20 2021 22:02:35
I (724) cpu_start: ELF file SHA256:  63110af333a6ad24...
I (730) cpu_start: ESP-IDF:          v4.1
I (734) cpu_start: Starting app cpu, entry point is 0x400812cc
0x400812cc: call_start_cpu1 at /home/yury/esp/esp-idf/components/esp32/cpu_start.c:271

Yury777, 2021-02-25:
I deleted the ble2mqtt directory. 
I deleted the ble2mqtt directory. then I executed the following commands:
#git clone --recursive https://github.com/shmuelzon/esp32-ble2mqtt
fixed config.json on your own
#idf.py flash -p /dev/ttyS1
#idf.py monitor -p /dev/ttyUSB0

here's what happened:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (28) boot: ESP-IDF v4.1 2nd stage bootloader
I (28) boot: compile time 16:47:27
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (97) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (105) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (113) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (120) boot: End of partition table
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x417dc (268252) map
I (243) esp_image: segment 1: paddr=0x00051804 vaddr=0x3ffbdb60 size=0x064a4 ( 25764) load
I (254) esp_image: segment 2: paddr=0x00057cb0 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (254) esp_image: segment 3: paddr=0x000580bc vaddr=0x40080404 size=0x07f5c ( 32604) load
I (275) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xef460 (980064) map
0x400d0020: _stext at ??:?

I (648) esp_image: segment 5: paddr=0x0014f488 vaddr=0x40088360 size=0x134b0 ( 79024) load
0x40088360: btdm_task_post_from_isr at ??:?

I (698) boot: Loaded app from partition at offset 0x10000
I (698) boot: Disabling RNG early entropy source...
I (699) cpu_start: Pro cpu up.
I (702) cpu_start: Application information:
I (707) cpu_start: Project name:     ble2mqtt
I (712) cpu_start: App version:      v0.13.0-6-gcb2ae40
I (718) cpu_start: Compile time:     Feb 25 2021 16:47:16
I (724) cpu_start: ELF file SHA256:  24c67699ce6bd883...
I (730) cpu_start: ESP-IDF:          v4.1
I (735) cpu_start: Starting app cpu, entry point is 0x400812cc
0x400812cc: call_start_cpu1 at /home/yury/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (745) heap_init: Initializing. RAM available for dynamic allocation:
I (752) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (758) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (764) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (770) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
[... 762 lines cut ...]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dad2a  PS      : 0x00060d30  A0      : 0x800d790c  A1      : 0x3ffef220
0x400dad2a: _uuidtoa at /home/yury/esp/esp32-ble2mqtt/build/../main/broadcasters.c:30
 (inlined by) ibeacon_metadata_get at /home/yury/esp/esp32-ble2mqtt/build/../main/broadcasters.c:95

A2      : 0x3ffbabf8  A3      : 0x0000001d  A4      : 0xffffffc6  A5      : 0x400d746c
0x400d746c: ble_on_broadcaster_metadata at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:276

A6      : 0x3ffbab10  A7      : 0x3ffc1604  A8      : 0x800dad2a  A9      : 0x3ffef200
A10     : 0x00000000  A11     : 0x000000ff  A12     : 0x3ffef250  A13     : 0x00000000
A14     : 0x00000001  A15     : 0xff000000  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000013  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 24c67699ce6bd883

Backtrace: 0x400dad27:0x3ffef220 0x400d7909:0x3ffef280
0x400dad27: ibeacon_data_get at /home/yury/esp/esp32-ble2mqtt/build/../main/broadcasters.c:66
 (inlined by) ibeacon_metadata_get at /home/yury/esp/esp32-ble2mqtt/build/../main/broadcasters.c:93

0x400d7909: ble_on_broadcaster_discovered at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:297
 (inlined by) ble2mqtt_handle_event at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:548
 (inlined by) ble2mqtt_task at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:592

Rebooting...

shmuelzon, 2021-02-26:
I'm sorry but I made a stupid mistake.
Can you please grab the latest updates (`git checkout memory_override_debug && git pull`) and run it again?

Yury777, 2021-02-26:
I (74864) BLE2MQTT: Discovered Eddystone broadcaster
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x2c343032  PS      : 0x00060031  A0      : 0x40082848  A1      : 0x3ffbe470
0x40082848: _xt_lowint1 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1153

A2      : 0x00000000  A3      : 0x2c343032  A4      : 0x3ffc1a2c  A5      : 0xefffffff
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80089c86  A9      : 0x3ffbe450
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x800d40db  A13     : 0x3ffbbce0
A14     : 0x3ffbbe00  A15     : 0x3ffbbd00  SAR     : 0x00000000  EXCCAUSE: 0x00000014
EXCVADDR: 0x2c343030  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x2c343032  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

ELF file SHA256: 64548ecb5c5687b9

Backtrace: 0x2c34302f:0x3ffbe470 |<-CORRUPTED

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (29) boot: ESP-IDF v4.1 2nd stage bootloader
I (29) boot: compile time 20:36:17
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (99) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (106) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (114) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (121) boot: End of partition table
I (126) boot_comm: chip revision: 1, min. application chip revision: 0
I (133) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x417dc (268252) map
I (244) esp_image: segment 1: paddr=0x00051804 vaddr=0x3ffbdb60 size=0x064a4 ( 25764) load
I (255) esp_image: segment 2: paddr=0x00057cb0 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (255) esp_image: segment 3: paddr=0x000580bc vaddr=0x40080404 size=0x07f5c ( 32604) load
I (276) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xef464 (980068) map
0x400d0020: _stext at ??:?

I (649) esp_image: segment 5: paddr=0x0014f48c vaddr=0x40088360 size=0x134b0 ( 79024) load
0x40088360: btdm_task_post_from_isr at ??:?

I (699) boot: Loaded app from partition at offset 0x10000
I (699) boot: Disabling RNG early entropy source...
I (700) cpu_start: Pro cpu up.
I (703) cpu_start: Application information:
I (708) cpu_start: Project name:     ble2mqtt
I (713) cpu_start: App version:      v0.13.0-7-gb993ead
I (719) cpu_start: Compile time:     Feb 26 2021 20:36:06
[... 1820 lines cut ...]
I (341054) Broadcaster: 02 01 06 1a 16 95 fe 58 58 5b 05 62 01 77 51 38
I (341054) Broadcaster: c1 a4 0d 0d 9e 9d 8b 0b 00 00 3e 98 86 c3
I (341064) BLE2MQTT: Queuing event BLE_BROADCASTER_DISCOVERED (a4:c1:38:51:77:01, 0x3fff45d0, 30, -54, 0x3ffc15b4)
I (341074) BLE2MQTT: 02 01 06 1a 16 95 fe 58 58 5b 05 62 01 77 51 38
I (341084) BLE2MQTT: c1 a4 0d 0d 9e 9d 8b 0b 00 00 3e 98 86 c3
I (341084) BLE2MQTT: Handling broadcaster(a4:c1:38:51:77:01, 0xc000e67e, 3221284478, -1073682818, 0xc000e67e)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c29b  PS      : 0x00060d30  A0      : 0x8015bfc1  A1      : 0x3ffef1f0
A2      : 0x3ffef231  A3      : 0xc000667e  A4      : 0x00000010  A5      : 0x3ffef231
A6      : 0x3ffcc2b8  A7      : 0x00000005  A8      : 0x801bf3d5  A9      : 0x3ffef1d0
A10     : 0x00000079  A11     : 0x3ffef3c0  A12     : 0x3ffef240  A13     : 0x0000000c
A14     : 0x00000031  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0xc000667e  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

ELF file SHA256: 64548ecb5c5687b9

Backtrace: 0x4000c298:0x3ffef1f0 0x4015bfbe:0x3ffef200 0x400d78bf:0x3ffef280
0x4015bfbe: esp_log_buffer_hex_internal at /home/yury/esp/esp-idf/components/log/log_buffers.c:44

0x400d78bf: ble_on_broadcaster_discovered at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:291
 (inlined by) ble2mqtt_handle_event at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:548
 (inlined by) ble2mqtt_task at /home/yury/esp/esp32-ble2mqtt/build/../main/ble2mqtt.c:592

Rebooting...

shmuelzon, 2021-02-27:
Unfortunately, only bad news...
There does indeed seem to be a memory overrun, as can be seen from the log.
We detect the a broadcaster:
```
I (194474) Broadcaster: Matched a broadcaster Mijia Temp+Hum (0x3ffc1580) to:
I (194474) Broadcaster: 02 01 06 1a 16 95 fe 58 58 5b 05 5c 01 77 51 38
I (194484) Broadcaster: c1 a4 03 62 f7 6e 74 0b 00 00 b7 51 19 05
```
Send that to the main application (from the BLE thread):
```
I (194484) BLE2MQTT: Queuing event BLE_BROADCASTER_DISCOVERED (a4:c1:38:51:77:01, 0x3fff59d0, 30, -65, 0x3ffc15b4)
I (194494) BLE2MQTT: 02 01 06 1a 16 95 fe 58 58 5b 05 5c 01 77 51 38
I (194504) BLE2MQTT: c1 a4 03 62 f7 6e 74 0b 00 00 b7 51 19 05
```
So far so good. This information is saved in a queue that the main application thread reads and acts upon (this is so the application won't try to do things from a thread that doesn't belong to it and block. for example, the ESP's BLE thread).
However, when the main application wakes up and gets the information from the queue, this is what it gets:
```
I (194514) BLE2MQTT: Handling broadcaster(a4:c1:38:51:77:01, 0x17ff0000, 402587648, 402587648, 0x17ff0000)
```
The MAC address is correct but the pointer to the broadcasted data, it's length, RSSI value and pointer to the detected broadcaster, have all changed to `0x17ff0000`. Some other part of the code wrote over the queue's memory and these kinds of things are hard to debug. It doesn't seem related to the broadcasters code as I initially though but rather just a symptom.
From what I read, ESP-IDF does have some tools to try and find these issues but it will take some time until I get acquainted with
them and have something you can run.

In the meantime, as it seems to reproduce rather easily for you, I would be happy if you can enable debug logs. Run `idf.py menuconfig` and choose `Component config` -> `Log output` and change `Default log verbosity` to `Debug`. This might show what's going on between adding the broadcaster to the queue and reading it afterwards.
In addition, I want to try to rule out other things so, if you agree, I want to try a run without any services/characteristic definitions in your configuration file and another with an empty whitelist so the ESP won't connect to any devices.
By disabling these two parts, there would be less things running in the background and that might hint at where the problem is (or isn't). Another option is, if you have another ESP-32 around, is to try and run the application on it, again, just to rule out some things.

I really appreciate you trying to help me with this and will understand if you can't invest the time for all these runs. If we can't find the root cause, I might need to invest in a Thingy 52 myself so I can reproduce it here.

Thanks again!

Yury777, 2021-02-28:
Thank you for this wonderful program! I'm happy to help you! I still have 10 pieces of ESP32.
I'll send the log in five minutes.

Yury777, 2021-02-28:
Executing action: monitor
Running idf_monitor in directory /home/yury/esp/esp32-ble2mqtt
Executing "/home/yury/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/yury/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- /home/yury/esp/esp32-ble2mqtt/build/ble2mqtt.elf -m '/home/yury/.espressif/python_env/idf4.1_py2.7_env/bin/python' '/home/yury/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (28) boot: ESP-IDF v4.1 2nd stage bootloader
I (28) boot: compile time 20:46:17
I (28) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (97) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (105) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (112) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (120) boot: End of partition table
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x59494 (365716) map
I (280) esp_image: segment 1: paddr=0x000694bc vaddr=0x3ffbdb60 size=0x06504 ( 25860) load
I (291) esp_image: segment 2: paddr=0x0006f9c8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (291) esp_image: segment 3: paddr=0x0006fdd4 vaddr=0x40080404 size=0x00244 (   580) load
I (298) esp_image: segment 4: paddr=0x00070020 vaddr=0x400d0020 size=0xfff94 (1048468) map
0x400d0020: _stext at ??:?

I (706) esp_image: segment 5: paddr=0x0016ffbc vaddr=0x40080648 size=0x1b234 (111156) load
I (770) boot: Loaded app from partition at offset 0x10000
I (770) boot: Disabling RNG early entropy source...
I (770) cpu_start: Pro cpu up.
I (774) cpu_start: Application information:
I (779) cpu_start: Project name:     ble2mqtt
I (784) cpu_start: App version:      v0.13.0-7-gb993ead
I (789) cpu_start: Compile time:     Feb 28 2021 20:46:10
I (796) cpu_start: ELF file SHA256:  6250bc22862aa041...
I (802) cpu_start: ESP-IDF:          v4.1
I (806) cpu_start: Starting app cpu, entry point is 0x400812e0
0x400812e0: call_start_cpu1 at /home/yury/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
D (817) memory_layout: Checking 11 reserved memory ranges:
D (822) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (828) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (835) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (841) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (848) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (854) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffd3718
D (860) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (867) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (873) memory_layout: Reserved memory range 0x40070000 - 0x40078000
[... 1979 lines cut ...]
D (2003) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (2003) efuse: coding scheme 0
D (2013) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (2013) efuse: coding scheme 0
D (2023) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (2023) efuse: coding scheme 0
D (2033) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (2033) efuse: coding scheme 0
D (2043) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (2043) nvs: nvs_get_str_or_blob cal_data
D (2063) nvs: nvs_close 4
D (2063) efuse: coding scheme 0
D (2063) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (2063) efuse: coding scheme 0
D (2063) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (2073) efuse: coding scheme 0
D (2073) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (2083) efuse: coding scheme 0
D (2083) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (2093) efuse: coding scheme 0
D (2093) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (2103) efuse: coding scheme 0
D (2103) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (2113) efuse: coding scheme 0
D (2123) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit

Yury777, 2021-02-28:
data/config.json:
{
  "wifi": {
    "ssid": "mywifi",
    "password": "password"
  },
  "mqtt": {
    "server": {
      "host": "192.168.1.2",
      "port": 1883,
      "username": "esp2",
      "password": "esp2",
      "client_id": "esp2"
    },
    "publish": {
      "retain": true
    },
    "topics" :{
      "prefix": "",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  }
}

shmuelzon, 2021-03-04:
Hey, I think the two logs are the same, no?
In any case, the good news are that I identified a problem from your latest logs. However, I don't see yet how this relates to the original issue you've had. I'll try to fix what I saw here and update you when it's ready so we can continue from there.

shmuelzon, 2021-03-05:
Hey, I pushed a fix to the bug from the previous log, can you please pull the latest updates and try again?
Thanks!

Yury777, 2021-03-06:
Hi!
I (6103) MQTT: MQTT client connected
D (6113) BLE2MQTT: Queuing event MQTT_CONNECTED
I (6113) BLE2MQTT: Connected to MQTT, scanning for BLE devices
D (6123) MQTT: Subscribing to BLE2MQTT-B960/OTA/Firmware
D (6123) MQTT_CLIENT: mqtt_enqueue id: 24370, type=8 successful
D (6133) OUTBOX: ENQUEUE msgid=24370, msg_type=8, len=33, size=33
D (6143) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT-B960/OTA/Firmware, id: 24370, type=8 successful
D (6143) MQTT: Subscribing to BLE2MQTT/OTA/Firmware
D (6153) MQTT_CLIENT: mqtt_enqueue id: 60723, type=8 successful
D (6153) OUTBOX: ENQUEUE msgid=60723, msg_type=8, len=28, size=61
D (6163) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT/OTA/Firmware, id: 60723, type=8 successful
D (6173) MQTT: Subscribing to BLE2MQTT-B960/OTA/Config
D (6173) MQTT_CLIENT: mqtt_message_receive: first byte: 0x90
D (6183) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (6193) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
D (6203) MQTT_CLIENT: mqtt_message_receive: read_len=3
D (6203) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (6213) MQTT_CLIENT: msg_type=9, msg_id=24370
D (6213) MQTT_CLIENT: pending_id=60723, pending_msg_count = 2
D (6223) OUTBOX: DELETED msgid=24370, msg_type=8, remain size=28
D (6223) MQTT_CLIENT: Subscribe successful
D (6233) MQTT_CLIENT: mqtt_enqueue id: 27732, type=8 successful
D (6233) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd69a0
D (6233) OUTBOX: ENQUEUE msgid=27732, msg_type=8, len=31, size=59
D (6253) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT-B960/OTA/Config, id: 27732, type=8 successful
D (6263) MQTT: Subscribing to BLE2MQTT/OTA/Config
D (6263) MQTT_CLIENT: mqtt_message_receive: first byte: 0x90
D (6273) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (6273) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
D (6283) MQTT_CLIENT: mqtt_message_receive: read_len=3
D (6293) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (6293) MQTT_CLIENT: msg_type=9, msg_id=60723
D (6303) MQTT_CLIENT: pending_id=27732, pending_msg_count = 2
D (6303) OUTBOX: DELETED msgid=60723, msg_type=8, remain size=31
D (6313) MQTT_CLIENT: Subscribe successful
D (6313) MQTT_CLIENT: mqtt_enqueue id: 37917, type=8 successful
D (6323) OUTBOX: ENQUEUE msgid=37917, msg_type=8, len=26, size=57
D (6333) MQTT_CLIENT: Sent subscribe topic=BLE2MQTT/OTA/Config, id: 37917, type=8 successful
D (6333) BLE: Starting BLE scan
D (6333) MQTT_CLIENT: mqtt_message_receive: first byte: 0x90
D (6343) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (6343) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)
D (6353) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
D (6373) MQTT_CLIENT: mqtt_message_receive: read_len=3
D (6373) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (6383) MQTT_CLIENT: msg_type=9, msg_id=27732
D (6383) MQTT_CLIENT: pending_id=37917, pending_msg_count = 2
D (6393) OUTBOX: DELETED msgid=27732, msg_type=8, remain size=26
D (6393) MQTT_CLIENT: Subscribe successful
D (6403) MQTT_CLIENT: mqtt_message_receive: first byte: 0x90
D (6403) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (6413) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
D (6423) MQTT_CLIENT: mqtt_message_receive: read_len=3
D (6423) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (6433) MQTT_CLIENT: msg_type=9, msg_id=37917
D (6433) MQTT_CLIENT: pending_id=37917, pending_msg_count = 1
D (6443) OUTBOX: DELETED msgid=37917, msg_type=8, remain size=0
D (6443) MQTT_CLIENT: Subscribe successful
D (6513) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd69a0
D (6683) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (6813) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd69a0
D (7113) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd69a0
D (8213) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd69a0
D (9313) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd69a0

Yury777, 2021-03-06:
and it froze, no messages

Yury777, 2021-03-06:
and then once a minute:
D (63073) BLE2MQTT: Queuing event HEARTBEAT_TIMER
D (123073) BLE2MQTT: Queuing event HEARTBEAT_TIMER

shmuelzon, 2021-03-06:
Sorry, I made a mistake while testing this otherwise I would have seen it myself...
Anyways, pushed a(nother) fix

Yury777, 2021-03-06:
D (67353) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
E (67363) BT_APPL: Max Notification Reached, registration failed.
D (67363) MQTT_CLIENT: msg_type=9, msg_id=4285
D (67373) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
D (67373) MQTT_CLIENT: pending_id=16316, pending_msg_count = 2
E (67383) BLE: Failed registering for notification, status = 0x80
D (67393) OUTBOX: DELETED msgid=4285, msg_type=8, remain size=102
E (67403) BT_APPL: Max Notification Reached, registration failed.
D (67403) MQTT_CLIENT: Subscribe successful
D (67413) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
I (67323) BLE2MQTT: Discovered BLE device: 28:ff:3c:98:8f:94 (RSSI: -90), connecting
E (67423) BLE: Failed registering for notification, status = 0x80
D (67423) MQTT_CLIENT: mqtt_message_receive: first byte: 0x90
E (67443) BT_APPL: Max Notification Reached, registration failed.
D (67443) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (67453) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
D (67453) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
E (67463) BLE: Failed registering for notification, status = 0x80
D (67473) MQTT_CLIENT: mqtt_message_receive: read_len=3
E (67483) BT_APPL: Max Notification Reached, registration failed.
D (67483) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (67493) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
D (67503) MQTT_CLIENT: msg_type=9, msg_id=16316
E (67503) BLE: Failed registering for notification, status = 0x80
D (67513) MQTT_CLIENT: pending_id=16316, pending_msg_count = 1
E (67523) BT_APPL: Max Notification Reached, registration failed.
D (67523) OUTBOX: DELETED msgid=16316, msg_type=8, remain size=0
D (67533) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
D (67533) MQTT_CLIENT: Subscribe successful
E (67543) BLE: Failed registering for notification, status = 0x80
E (67553) BT_APPL: Max Notification Reached, registration failed.
D (67563) BLE: Received GATTC event 38 (ESP_GATTC_REG_FOR_NOTIFY_EVT), gattc_if 1
E (67573) BLE: Failed registering for notification, status = 0x80
D (67573) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (67583) BLE2MQTT: Queuing event BLE_DEVICE_DISCOVERED (67:e6:83:e6:96:8b, -50)
I (67593) BLE2MQTT: Discovered BLE device: 67:e6:83:e6:96:8b (RSSI: -50), connecting
D (67603) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)
E (67603) BT_BTM: BTM_BleScan scan not active

W (67613) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

W (67623) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1
D (67623) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)
E (67633) BLE: Scanning stopping failed, status: 0x1
D (72613) BLE: Received GATTC event 40 (ESP_GATTC_CONNECT_EVT), gattc_if 1
D (72613) BLE: GATTC event 40 wasn't handled
D (72613) BLE: Received GATTC event 2 (ESP_GATTC_OPEN_EVT), gattc_if 1
W (72623) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

E (72623) BT_HCI: btu_hcif_hdl_command_complete opcode 0x200c status 0xc
D (72633) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)
E (72643) BLE: Scanning starting failed, status: 0x1
W (73373) BT_GATT: gatt_if out of range [ = 0]
E (73373) BT_GATT: GATT_Write Illegal param: conn_id 0, type 02,
E (73383) BT_APPL: No pending command
D (73383) BLE: Received GATTC event 46 (Invalid GATTC event), gattc_if 1
D (73383) BLE: GATTC event 46 wasn't handled
D (73413) BLE: Received GATTC event 18 (ESP_GATTC_CFG_MTU_EVT), gattc_if 1
D (73413) BLE2MQTT: Queuing event BLE_DEVICE_CONNECTED (28:ff:3c:98:8f:94)
I (73423) BLE2MQTT: Connected to device: 28:ff:3c:98:8f:94, scanning
D (73433) MQTT: Subscribing to 28:ff:3c:98:8f:94/Connected

Yury777, 2021-03-06:
D (219873) BLE: GATTC event 41 wasn't handled
D (219883) BLE: Received GATTC event 2 (ESP_GATTC_OPEN_EVT), gattc_if 1
E (219893) BLE: Open failed, status = 0x85
D (219893) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)
D (220153) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)

Yury777, 2021-03-06:
only these errors
it's been running for 6 minutes, never rebooted

Yury777, 2021-03-06:
[log.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6095900/log.txt)

Yury777, 2021-03-06:
it's frozen again

shmuelzon, 2021-03-07:
Can you please run it again with your original configuration file (whitelisting devices). If it sill becomes unresponsive, I've pushed a commit with more debug messages.

Yury777, 2021-03-07:
[log2.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6096725/log2.txt)
it's frozen again

Yury777, 2021-03-07:
[log3.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6096833/log3.txt)
it's frozen again

shmuelzon, 2021-03-12:
Hey, log3 really helped find another issue that caused a deadlock.
Can you try now with the new update?

Once again, I really appreciate your help. I think the issues (and fixes) found here would greatly improve the stability of this application.

Yury777, 2021-03-12:
Hi!
I am very pleased to help you!
I deleted build... now I will compile and flash esp32... 5 minutes and I will write what came out of it.

Yury777, 2021-03-12:
after the last reboot, it works... but not for long:
D (174884) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
I (174884) Broadcaster: Matched a broadcaster Eddystone (0x3ffc157c) to:
I (174884) Broadcaster: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (174894) Broadcaster: 6f 2e 67 6c 2f 70 49 57 64 69 72
I (174894) BLE2MQTT: Queuing event BLE_BROADCASTER_DISCOVERED (c5:4a:d2:94:7d:13, 0x3fff4d2c, 27, -46, 0x3ffc15dc)
I (174914) BLE2MQTT: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (174914) BLE2MQTT: 6f 2e 67 6c 2f 70 49 57 64 69 72
I (174924) BLE: XXX Taking semaphore gap_cb:686
I (174924) BLE2MQTT: Handling broadcaster(c5:4a:d2:94:7d:13, 0x3fff4d2c, 27, -46, 0x3ffc15dc)
I (174924) BLE: XXX Giving semaphore gap_cb:693
I (174934) BLE2MQTT: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (174944) BLE2MQTT: 6f 2e 67 6c 2f 70 49 57 64 69 72
I (174954) BLE2MQTT: Discovered Eddystone broadcaster
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (58) boot: ESP-IDF v4.1 2nd stage bootloader
I (58) boot: compile time 21:56:21
I (58) boot: chip revision: 1
I (61) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (68) boot.esp32: SPI Speed      : 40MHz
I (73) boot.esp32: SPI Mode       : DIO
I (77) boot.esp32: SPI Flash Size : 4MB
W (82) boot.esp32: PRO CPU has been reset by WDT.
W (87) boot.esp32: WDT reset info: PRO CPU PC=0x4008d725
0x4008d725: panicHandler at /home/yury/esp/esp-idf/components/esp32/panic.c:247 (discriminator 1)

W (93) boot.esp32: WDT reset info: APP CPU PC=0x400824f4
0x400824f4: _xt_panic at /home/yury/esp/esp-idf/components/freertos/xtensa_vectors.S:385

Yury777, 2021-03-12:
D (10094) BLE2MQTT: Queuing event BLE_DEVICE_DISCOVERED (80:e6:50:ed:e9:c5, -92)
I (10094) BLE: XXX Giving semaphore ble_connect:298
I (10104) BLE2MQTT: Discovered BLE device: 80:e6:50:ed:e9:c5 (RSSI: -92), not connecting
D (10114) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
W (10514) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x003e
D (10514) BLE: Received GATTC event 41 (ESP_GATTC_DISCONNECT_EVT), gattc_if 1
D (10524) BLE: GATTC event 41 wasn't handled
D (10524) BLE: Received GATTC event 2 (ESP_GATTC_OPEN_EVT), gattc_if 1

E (10534) BLE: Open failed, status = 0x85

I (10534) BLE: XXX Taking semaphore esp_gattc_cb:769
I (10544) BLE: XXX Giving semaphore esp_gattc_cb:771

shmuelzon, 2021-03-12:
OK, so the good news is that I don't see the same memory corruption we so before. The same data that was queued:
```
I (174894) BLE2MQTT: Queuing event BLE_BROADCASTER_DISCOVERED (c5:4a:d2:94:7d:13, 0x3fff4d2c, 27, -46, 0x3ffc15dc)
I (174914) BLE2MQTT: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (174914) BLE2MQTT: 6f 2e 67 6c 2f 70 49 57 64 69 72
```
Is later retrieved from the queue:
```
I (174924) BLE2MQTT: Handling broadcaster(c5:4a:d2:94:7d:13, 0x3fff4d2c, 27, -46, 0x3ffc15dc)
I (174934) BLE2MQTT: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (174944) BLE2MQTT: 6f 2e 67 6c 2f 70 49 57 64 69 72
```
I still don't understand what the problem is since if I use the Eddystone code with the above data, it's parsed as expected.
I added more debugs specifically in that location now. Can you please run it again and post the logs?

Yury777, 2021-03-13:
D (176431) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
I (176431) Broadcaster: Matched a broadcaster Eddystone (0x3ffc157c) to:
I (176441) Broadcaster: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (176441) Broadcaster: 6f 2e 67 6c 2f 70 49 57 64 69 72
I (176451) BLE2MQTT: Queuing event BLE_BROADCASTER_DISCOVERED (c5:4a:d2:94:7d:13, 0x3fff4aa8, 27, -57, 0x3ffc15dc)
I (176461) BLE2MQTT: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (176471) BLE2MQTT: 6f 2e 67 6c 2f 70 49 57 64 69 72
I (176471) BLE: XXX Taking semaphore gap_cb:686
I (176471) BLE2MQTT: Handling broadcaster(c5:4a:d2:94:7d:13, 0x3fff4aa8, 27, -57, 0x3ffc15dc)
I (176481) BLE: XXX Giving semaphore gap_cb:693
I (176491) BLE2MQTT: 02 01 06 03 03 aa fe 13 16 aa fe 10 ee 03 67 6f
I (176501) BLE2MQTT: 6f 2e 67 6c 2f 70 49 57 64 69 72
I (176511) BLE2MQTT: Discovered Eddystone broadcaster
I (176511) BLE2MQTT: Sending brodcaster metadata
I (176521) BLE2MQTT:   Topic: BLE2MQTT-B960/c5:4a:d2:94:7d:13/Type
I (176521) BLE2MQTT:   Value: Eddystone
I (176531) BLE2MQTT: Done
I (176531) BLE2MQTT: Sending brodcaster metadata
I (176541) BLE2MQTT:   Topic: BLE2MQTT-B960/c5:4a:d2:94:7d:13/RSSI
I (176541) BLE2MQTT:   Value: -57
I (176551) BLE2MQTT: Done
I (176551) Broadcaster: eddystone_metadata_get:252
I (176561) Broadcaster: eddystone_metadata_get:254
I (176561) Broadcaster: eddystone_metadata_get:273
I (176571) BLE2MQTT: Sending brodcaster metadata
I (176571) BLE2MQTT:   Topic: BLE2MQTT-B960/c5:4a:d2:94:7d:13/Distance
I (176581) BLE2MQTT:   Value: 0.79
I (176591) BLE2MQTT: Done
I (176591) Broadcaster: eddystone_metadata_get:278
I (176591) BLE2MQTT: Sending brodcaster metadata
I (176601) BLE2MQTT:   Topic: BLE2MQTT-B960/c5:4a:d2:94:7d:13/URL
I (176611) BLE2MQTT:   Value: https://ga4:38:c1:51:77:01oa4:38:c1:51:77:01oa4:
I (176611) BLE2MQTT: Done
I (176621) Broadcaster: eddystone_metadata_get:287
I (176621) Broadcaster: eddystone_metadata_get:315
D (176911) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
I (176911) BLE: XXX Taking semaphore gap_cb:686
I (176921) BLE: XXX Giving semaphore gap_cb:693
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (58) boot: ESP-IDF v4.1 2nd stage bootloader
I (58) boot: compile time 21:23:02
I (58) boot: chip revision: 1
I (61) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (68) boot.esp32: SPI Speed      : 40MHz
I (73) boot.esp32: SPI Mode       : DIO
I (77) boot.esp32: SPI Flash Size : 4MB
W (82) boot.esp32: PRO CPU has been reset by WDT.
W (87) boot.esp32: WDT reset info: PRO CPU PC=0x4008d725
0x4008d725: panicHandler at /home/yury/esp/esp-idf/components/esp32/panic.c:247 (discriminator 1)

shmuelzon, 2021-03-13:
OK, found a problem that would also explain the original memory override we saw. Oddly enough, when testing the same code on my PC it didn't find any issues.
Anyways, I'm optimistic now...🤞

Can you please pull the latest changes and try again?

Yury777, 2021-03-14:
HURRAY!!!! you did it! IT WORKS! it's been 10 minutes without a single reboot!
THANK YOU SO MUCH!

shmuelzon, 2021-03-14:
That's great! Glad to hear that. Please do let me know if some problem does arise later on.
I'll merged the relevant fixes to the main branch properly later and, if everything still looks good, I'll close this issue.

Thanks again for sticking by and helping debug this!

Yury777, 2021-03-14:
4 hours, normal flight...
did not reboot, no errors

Yury777, 2021-03-14:
D (16647331) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
I (16647331) BLE: XXX Taking semaphore gap_cb:686
I (16647331) BLE: XXX Giving semaphore gap_cb:693

shmuelzon, 2021-03-14:
That's oddly satisfying :)

Yury777, 2021-03-14:
it still works! I saw this error:
(42301971) BLE: XXX Taking semaphore gap_cb:686
I (42301971) BLE: XXX Giving semaphore gap_cb:693
D (42303071) BLE2MQTT: Queuing event HEARTBEAT_TIMER
**E (42303201) BT_HCI: btu_hcif_hdl_command_complete opcode 0x2005 status 0xc**
D (42304341) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)

shmuelzon, 2021-03-14:
Hard to tell... The opcode `0x2005` is `HCI_BLE_WRITE_RANDOM_ADDR`, whatever that means, and that error isn't event sent to the application so there's not much I can do about it anyways.
I think we can ignore it for now

shmuelzon, 2021-03-14:
Merged to master. Thanks for all your help!

closed 2021-03-14 by shmuelzon

Yury777, 2021-03-14:
thank you! how do I switch to the main branch now and reduce the logs?

shmuelzon, 2021-03-14:
To switch to the main branch:
```
git checkout master
```
Get the latest changes:
```
git pull
```
Clean up everything and build again:
```
rm -rf build sdkconfig sdkconfig.old
idf.py build
```

Yury777, 2021-03-14:
ok. Thank you!

commit bee24479 (2021-03-14): Fix Eddystone URL decoding

Fixes #95