shmuelzon/esp32-ble2mqtt #99: [Feature] Implement Mqtt standard specification

slobberbone, 2021-02-17:
It would be interesting to implement in this project the specification describe here :
https://homieiot.github.io/

However, MQTT doesn’t define the structure and content of these messages and their relation. An IoT device publishes data and provides interaction possibilities but a controlling entity will need to be specifically configured to be able to interface with the device.

The Homie convention defines a standardized way of how IoT devices and services announce themselves and their data on the MQTT broker.

It is thereby a crucial aspect on top of the MQTT protocol for automatic discovery, configuration and usage of devices and services.

Some other xxx2mqtt already purpose this and I think specification for MQTT is the futur of home automation.
It would be not changing the way of this project work actually, but add another way to publish.

I'm from the open source project NextDom (home automation software) and we are working on another solution from scratch. We want to organize NextDom directly with MQTT and Node-Red (Nextdom becoming a kind of orchestrator. In this way, this standard for MQTT will be very interesting for us of course, but also for all others home automation software.

Thanks for reading and give me some feed back about that !

shmuelzon, 2021-02-23:
Hey,

While I'm not familiar with Homie, I did think about supporting [Home Assistant's MQTT discovery](https://www.home-assistant.io/docs/mqtt/discovery/) in the past. The problem I ran into is that this project, by design, isn't aware of the meaning/usage of each characteristic exposed by a BLE peripheral. Even if we rely only standard characteristics, i.e. the ones assigned by the Bluetooth SIG, you'd still need to map (and maybe group) characteristics to entities Homie supports.
For it to be truly useful, you'd have to somehow support user-defined characteristics since most cheap sensors decide use their own custom characteristics. In addition, there's also the issue of BLE broadcasters that you'll need to include as well, but those may be simpler to integrate as you're already handling vendor specific data.

Anyways, I'm not apposed to adding support for this but we'd need a plan on how to tackle it first. We can continue the discussion here, if it's relevant, and, obviously, any pull requests for this are welcome.

Thanks

unrelentingtech, 2021-05-23:
Well, this project *is* aware of the meaning of broadcasters' data! :)

For characteristics, probably defining a mapping in the JSON config would work…

shmuelzon, 2021-05-23:
True, broadcasters are a weird bunch. But since you don't really have an active connection with them and multiple instances of this project you'd probably have issues around that.

As for a JSON config, it might be possible but take into consideration that many devices include multiple bits of information for a single characteristic so the mapping would also need to take that into consideration. Also, what if a single device has multiple sensors / actuators, would we expose it as multiple devices in whatever MQTT standard we use?