shmuelzon/esp32-ble2mqtt #101: Show log in web interface

OKrause42, 2021-03-16:
**Is your feature request related to a problem? Please describe.**
Remote logging is really nice, but need to have it running all the time.

**Describe the solution you'd like**
When you want to only check log occasionally, it would be nice if you could see it in the web interface

shmuelzon, 2021-03-16:
Hey,

To be able to see the log at any point means you would have to save it on the ESP32 itself and that would take up much of your memory. I could limit the amount of log entries with a cyclic buffer of some sort but then the questions are:
1. Is it worth it? That memory could be "spent" on maintaining another BLE connection
2. How many log entries to keep? We could make it configurable
3. Which log level? Everything? Only errors?

What kind of information is it you want to extract from the logs?

Thanks

OKrause42, 2021-03-16:
Good point - I was mostly thinking about the last (few) errors.
E.g. if something goes wrong (while you hadn't remote-logging running), you have a place to check for hints

shmuelzon, 2021-03-16:
This is assuming that something went wrong, but the ESP is still alive, connected to WiFi and generally working.
In my experience, when something goes wrong, you pretty much have to reset the ESP to fix it or it resets itself and then the saved log is gone.
It makes me question the usefulness of such a feature.

Now, if you have a SD card installed, we could save the log to it and then, either take the card itself or allow to download/view the logs when the ESP is up again. Another option is to sacrifice flash memory from the SPIFFS (where the configuration and certificates are stored) to save a persistent log file which, again, can be downloaded/viewed later on.

What do you think?

OKrause42, 2021-03-21:
Yes, reducing memory is probably not worth it - but using the SD card sounds like a very good idea (especially as it could also be taken out after the fact...)