shmuelzon/esp32-ble2mqtt #102: how do I turn off notifications on a characteristic?

Yury777, 2021-03-22:
Hi!
I continue to configure the bridge with my thingy52...
what command should I send via BLE to disable the pressure, gas, and light/color intensity sensor?

In this documentation:
https://nordicsemiconductor.github.io/Nordic-Thingy52-FW/documentation/firmware_architecture.html
written:
"The environment module is responsible for reading the temperature, pressure, humidity, air quality/gas, and light intensity/color sensors, as well as for sending data and configuration. It is running on top of the environment BLE service (ble_tes), the environment sensor drivers, and the environment flash storage with base UUID EF68xxxx-9B35-4933-9B10-52FFA9740042. The configuration is stored in flash. To turn on sensor reading, enable notifications from the desired characteristic by writing 0x0001 to the Client Characteristic Configuration Descriptor (CCCD) for that characteristic."

shmuelzon, 2021-03-22:
Hey,

Well, you can't...
The idea of this project was to be zero-touch, meaning it would attempt to do all it can to make sure the BLE data on the MQTT bus is up-to-date. To do that it automatically registers on notifications so you don't have to manually issue get commands.
If you don't want the data from those characteristics, you can blacklist them. Otherwise, perhaps you can configure the Thingy52 to send the information less frequently?

Yury777, 2021-03-22:
Yes, I see! 
light / color intensity sensor blinks during the measurement, very bright : - (
maybe there is some simple solution? 
please tell me!
thingy52 software is too complex for me.

shmuelzon, 2021-03-22:
According to https://nordicsemiconductor.github.io/Nordic-Thingy52-FW/documentation/firmware_architecture.html#arch_env,
There's the `EF680200-9B35-4933-9B10-52FFA9740042` weather station service which has a `EF680206-9B35-4933-9B10-52FFA9740042` configuration characteristic.

You can add the following to your configuration:
```json
{
  "ble": {
    "services": {
      "definitions": {
        "ef680200-9b35-4933-9b10-52ffa9740042": {
          "name": "WeatherStation"
        }
      }
    },
    "characteristics": {
      "definitions": {
        "ef680206-9b35-4933-9b10-52ffa9740042": {
          "name": "Configuration",
          "types": [
            "uint16",
            "uint16",
            "uint16",
            "uint16",
            "uint8",
            "uint8",
            "uint8",
            "uint8"
          ]
        }
      }
    }
  }
}
```
Then, see what the `XX:XX:XX:XX:XX:XX/WeatherStation/Configuration` MQTT topic publishes and save that somewhere as it contains calibration values. The value which controls the interval of the color sensor is the forth one and its value is in milliseconds.
The published value would be `AA,BB,CC,DD,EE,FF,GG,HH`, issue a set command by publishing the same information to `XX:XX:XX:XX:XX:XX/WeatherStation/Configuration/Set` but replace the value in the forth field (`DD` in the above example) with the polling interval you want, e.g. `60000` for ` minute.

Yury777, 2021-03-22:
it worked!
I just removed the EF680206-9B35-4933-9B10-52FFA9740042 feature from the white list

closed 2021-03-22 by Yury777