shmuelzon/esp32-ble2mqtt #104: operation_queue concurrent access

kfishermail, 2021-04-26:
After looking at the code, it appears (to me?) that operation_queue is being accessed concurrently by two tasks, 
should this access be protected by some sort of access mechanism (i.e. semaphore)?

shmuelzon, 2021-04-26:
Hey,

You're probably right. I've recently added one for the `devices_list` so one might be in order here as well.
Another option is to use FreeRTOS's queue implementation (not sure why I didn't use it in the first place) which is probably* thread safe.

*I'd still verify that

Thanks!

kfishermail, 2021-04-26:
I have a few other bugs/suggestions for your code, are you interested in
hearing them? If so,
should I post them to public forum, or this private email?

shmuelzon, 2021-04-27:
> are you interested in hearing them?
Of course I am! Anything you think might help this project.
Publicly is fine, there aren't any secrets in the code. But, if it's not too much trouble, I would appreciate a separate issue for each so it would be easier to manage and fix one at a time.