shmuelzon/esp32-ble2mqtt #105: Race Condition

kfishermail, 2021-04-27:
regarding: ble_operation_enqueue
Background:
xTimerReset name is misleading, it should be named QueCommandToResetTimer.  It doesn't
restart the timer, it puts a command in a que (to be executed (when??)) to restart the timer.

The code is calling 'xTimerIsTimerActive' to see if there is a command in process.  xTimerIsTimerActive
will return true only after the system has processed the timer que (which could be who knows when)

I think the race condition is obvious.
One simple solution is to create a new flag 'new_timer_busy'  Set it to 1 in 'ble_operation_perform'
set it to 0 in 'ble_operation_dequeue' when operation_queue is empty

Check it instead of calling 'xTimerIsTimerActive'.

There are better solutions, but require more changes to code.
This problem, and my prev problem seem minor, but in my application,
I am sending 12 ble commands per-second repeatadly.  I noticed that I was missing
some of the commands, fixing these two issues solved my problems.

If you are only sending a single ble command, and not very often, these two changes
might not be needed.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and logs**
If applicable, add your configuration file (without passwords) and logs (preferably of DEBUG level) to help understand your problem.

**Additional context**
Add any other context about the problem here.

shmuelzon, 2021-04-28:
Hey,

> I think the race condition is obvious.

Unfortunately, it isn't for me...
Also, what I understood from your description isn't exactly what's going on, unless I'm misusing FreeRTOS's utilities.
The idea of the queue was not to overwhelm the ESP BLE library with many requests simultaneously, for example while we're discovering a new peripheral, we want to read all of the services, register on whatever supports notifications, etc. Plus the BLE library's queue wasn't large enough so I decided to handle it myself.

The timer was added so the BLE operations won't happen immediately and (is supposed to) work like so:
When the first device is connected, we detect it's first characteristic and want to queue an operation. This would be the first operation added to the queue and so we reset (restart) the timer, which is configured for 0.5s:
```c
    /* First item in queue or timer is already running, reset timer */
    if (!(*queue)->next || xTimerIsTimerActive(timer))
        xTimerReset(timer, 0);
```
Then we find the second characteristic and queue another operation. Again, we reset the timer since it's currently active, making it start counting again, from the beginning, the same 0.5 second. Since it takes a lot less than 0.5s to go through all of the services, we effectively wait until we finished handling all of the characteristics while not issuing a single BLE command to the BLE library, only queueing them internally. After handling the final characteristic, the timer is reset for the last time and now we wait the 0.5s.
Assuming no other peripherals were connected and scanned in this time, the timer expires, calling `ble_queue_timer_cb()` and we dequeue the first BLE operation. The (one-shot) timer is now inactive and we won't be using it until the next time an operation is queued. So, we dequeue an operation, send the request to the BLE library and wait until we receive an acknowledgement that would be handled in `esp_gattc_cb()`. Once we know the command successfully finished, we immediately dequeue the next operation, send it to the BLE library, wait for a response and that goes on until the queue is empty.
Now, if a new device is connected and scanned before the queue is empty, we add the new operation to the queue, the timer isn't active an so it will just be handled when its time comes after the previous operations have completed.
If the queue was emptied before the new peripheral was scanned then we're back to square one, i.e. we reset the timer (again to wait 0.5s) until the last operation is queued and then issue them one-by-one.

So, with all that in mind, can you please explain to me where the race condition is?
Also, without your modifications, where do you think the commands you send in succession disappear to? They should all go into the queue and handled at some point.
If possible, please enable the debug log level, and attach the log here. We can then try to follow the queue operations to understand why not all of your commands are being handled.

Thanks!

kfishermail, 2021-04-29:
In your code you take different paths if 'timer is already running'
What is your definition of 'timer is already running?'
a) xTimerIsTimerActive() returns true
or
b) I have called XTimerReset and the timer has not expired yet.

If your answer is a, I am totally off base here and apologize.

kfishermail, 2021-04-29:
''Also, without your modifications, where do you think the commands you send in succession disappear to? They should all go into the queue and handled at some point.''
This what I think:  By using xTimerIsTimerActive() to see if the timer is running,  you are getting 
an occaisional "false negative" (i.e. you think it is not active, but in fact it is).  In this
case you are sending it to BLE when it is processing a different operation, and BLE is
ignoring one of them.  (just a guess)

shmuelzon, 2021-04-29:
> What is your definition of 'timer is already running?'

I'm pretty sure (a) and (b) are the same, no? At least that's what I understand from the [documentation](https://www.freertos.org/FreeRTOS-timers-xTimerIsTimerActive.html).
As for your issue, could you enable debug logs and post it here? We should see if operations are added to the queue and not taken from it.

kfishermail, 2021-04-29:
First, how do I enable debug logs?

now the difference between a and b.
The way I understand it is that XTimerReset  doesn't really
restart the timer, It puts a command in the timer que (to
be executed at a later time) to start the timer.  Therefore
there is a period of time between calling XTimerReset,
and freertos starting the timer.  During this time xTimerIsTimerActive
will return false.  I have no idea of the length of this
time.

During this interval, a and b will be different.  Will this cause problems?
Yes, I definetly think so.

Other people have had the same issue:
https://www.freertos.org/FreeRTOS_Support_Forum_Archive/November_2015/freertos_Checking_xTimerIsTimerActive_after_xTimerStart_b75af088j.html

shmuelzon, 2021-04-29:
OK, I think I understand your concern now but let's take a look at what would happen if 2 commands are queued together.
Assuming an idle state with the timer being dormant, the first command is queued, and we'll reset the timer due to the first condition here:
```c
    /* First item in queue or timer is already running, reset timer */
    if (!(*queue)->next || xTimerIsTimerActive(timer))
        xTimerReset(timer, 0);
```
So we reset the timer, i.e. added the reset command to a timer queue. Let's assume the timer request is still in the timer queue and now a second BLE command is to be queued. Now, we get false on both conditions and we don't reset the timer again.
This should be fine since at some point later, when the timer task is woken, the timer will still be reset and start counting the 0.5s. The end result is the same, executing the BLE commands in the queue 0.5 seconds after the last one was entered.

In addition, in the link you've sent, there's the following:
> If the priority of the timer task is higher than the priority of the task starting the timer, then you will not get this problem as, as soon as the command goes into the command queue, the timer task will leave the Blocked state, pre-empt the task that posted the command, and process the command, before the function that started the timer even returns.

The timer task has priority 1 while the application task has priority 5.

shmuelzon, 2021-04-29:
I just noticed that the timer task is pinned to core 0 while the application is on core 1 so it won't pre-empt it but I still think we should be good in this case.

shmuelzon, 2021-04-29:
To enable debug logs, run `idf.py menuconfig` and go to "Component config" -> "Log output" and change "Default log verbosity" to "Debug"