shmuelzon/esp32-ble2mqtt #106: Recover From Loss of MQTT broker

kfishermail, 2021-04-28:
I had a problem when I lost connection to my MQTT broker, The system would not
work with my ble devices after the MQTT Broker became available.

To fix this, delete all code in 'mqtt_on_disconnected'

Appearantly, the esp32 mqtt code automatically reconnects

shmuelzon, 2021-04-28:
Hey,

> Appearantly, the esp32 mqtt code automatically reconnects

True. However, `mqtt_on_disconnected()` doesn't do anything with regards to MQTT. What is does is disconnect from the BLE peripherals and stops BLE scanning. This was done since we can't expose the BLE peripherals on the bus now and there's no reason to keep the connection to them. Even though there is a queue for MQTT publications in case someone wants to publish something while the client is disconnected from the broker, that queue might fill up, depending on how much data the BLE peripheral sends.
In addition, since you might have other ESP32 running this project nearby, disconnecting from the BLE peripherals would allow another ESP32 to take over and keep the BLE peripheral active on the MQTT bus.

Can you please explain what exactly happened when the broker became available again? Did the MQTT client fail to connect to it? Did BLE scanning not resume once the MQTT client has connected?

Thanks!

kfishermail, 2021-04-29:
Enclosed is a log file.  It does not fail everytime, this run it took three tries to fail.
Also, although I can not prove it, It appears to me that it a
[mqtlog.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6396035/mqtlog.txt)
lways recovers on the
first failure, but occasionally fails on susequent loss of mqtt broker.

kfishermail, 2021-04-29:
When it fails, the system fails to connect to my BLE device

kfishermail, 2021-04-29:
Not sure if scanning resumed

kfishermail, 2021-04-29:
I stand corrected, here is a log of a failure on first MQTT broker loss.
[log.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6396120/log.txt)

shmuelzon, 2021-04-29:
So the code does try to resume scanning after it connects to the MQTT broker, but that returns an error, for some reason. I should probably retry upon failure. I'll look into that.

There are issues here though with attempting to connect to multiple devices simultaneously. I'll see what I can do to sort it out.

unrelentingtech, 2021-10-05:
Maybe reboot when the scanning returns an error?