shmuelzon/esp32-ble2mqtt #108: Eddystone TLM data

fludstank, 2021-05-26:
**Question**
I have an ESP32 that I have flashed with [this](https://github.com/pcbreflux/espressif/blob/master/esp32/arduino/sketchbook/ESP32_Eddystone_TLM_deepsleep/ESP32_Eddystone_TLM_deepsleep.ino) code and am trying to get it to speak to another ESP32 that I have successfully flashed with the ble2mqtt code. However, I can't get make an initial connection nor can I read the TLM data. If I use nRF connect on my Android phone, I can see that the data is being broadcast even without connecting, so I do know that part of it is working. I tried whitelisting the MAC address for the Eddystone ESP32 so that it wouldn't even bother connecting to other devices, but that didn't help.
**Additional context**
I know it's a longshot but was hoping that you could help point me in the right direction.

shmuelzon, 2021-05-28:
Hey,

I'll try to run that example soon and see if I can find what might be the issue.
Are you interested just in the Eddystone broadcast data or do you also want to connect to it?

Thanks

fludstank, 2021-05-29:
I don't need to connect to it, I'd really just need to read the data. Thanks for getting back to me I really appreciate it!

shmuelzon, 2021-05-29:
So, I took a look at the example and I think it has a bug in it.
Could you please try to replace this line:
```c
pAdvertising->setScanResponseData(oAdvertisementData);
```
With this one:
```c
pAdvertising->setAdvertisementData(oAdvertisementData);
```
With that in place, I could get the advertisement data on the BLE2MQTT and it detected it as an Eddystone beacon. The temperature I publish isn't the same as is in the sketch, but I haven't dug into that to understand where the problem might be.

fludstank, 2021-05-30:
Progress! That indeed looks like a bug. I also found @h2zero's [NimBLE](https://github.com/h2zero/NimBLE-Arduino) repo that has refactored a lot of that old code and it had the `setAdvertisementData` updated as well! I'm going to test that code with this and see what I come up with. It's funny you mention the uptime not being accurate as that is the mission critical data for the project I'm working on (otherwise I would have just used the iBeacon protocol and called it a day) so that is super important for me to figure out. 
Again, I just want to say thank you very much for helping me along with this - it is GREATLY appreciated!

shmuelzon, 2021-06-02:
Hey,

I fixed the parsing of the temperature values.
Please let me know if you have any more issues.

Thanks!

commit 7c143c34 (2021-06-02): Fix parsing of Eddystone TLM temperature

Fixes #108
closed 2021-06-02 by shmuelzon