shmuelzon/esp32-ble2mqtt #109: Let device subscribe to MQTT topics for toggling GPIOs

gvdhoven, 2021-05-28:
**Is your feature request related to a problem? Please describe.**
Yes and no, i want to mount this device inside my garage door opener. Because that is where i am going to scan my barcodes and i have a 24v power supply i can step down to 3.3v. However, i would ideally also like to use this same ESP to control my garage door. It is a simple latching connection inside my housing. 

**Describe the solution you'd like**
So if we could pull up a GPIO by sending an MQTT message to some topic and pull down by sending another message (or pull up with duration maybe?) then i can embed everything within this same device.

**Describe alternatives you've considered**
Add another ESP but that seems overkill :)

gvdhoven, 2021-05-28:
p.s. some implementations using esp-idf
https://www.cyberhades.com/en/2018/09/19/controlling-your-garage-doors-with-an-esp32-part-1/
https://github.com/renanredel/esp-garage-homekit/blob/master/main/garage.c

Idea would be maybe to have an additional config section with gpios (maybe only of type output and default state so they can be initialized at startup)

and that the node listens to <devicename>/gpio
and on that topic messages can be sent with:
gpio_nr
state (0/1)
max_duration (in case it has to be reset to default state)

thoughts?

shmuelzon, 2021-05-30:
Hey,

Unfortunately, this kind of feature isn't a part of the project's scope so I cannot add it nor maintain it.
However, I can try to help you with adding the required modifications to fit your setup.
I wouldn't bother with going through the configuration file as you probably already know which GPIO you're after so just set it for output at the end of the `app_main()` function in `main/ble2mqtt.c` file.

As for subscribing to MQTT, the following should get you started:
```c
/* TODO: Add  EVENT_TYPE_GPIO_MQTT to the event_type_t enumeration */

/* Add the following functions */
static void _gpio_on_mqtt(const char *topic, const uint8_t *payload, size_t len, void *ctx)
{
    _mqtt_on_message(EVENT_TYPE_GPIO_MQTT, topic, payload, len, ctx);
}

static void ota_on_mqtt(const char *topic, const uint8_t *payload, size_t len, void *ctx)
{
  /* Do whatever you want here, check what the payload is (if needed), toggle the GPIO, you can safety use vTaskDelay() if needed here */
}

/* TODO: Add to the switch() in ble2mqtt_handle_event() */
case EVENT_TYPE_GPIO_MQTT:
    gpio_on_mqtt(event->mqtt_message.topic, event->mqtt_message.payload, event->mqtt_message.len, event->mqtt_message.ctx);
    free(event->mqtt_message.topic);
    free(event->mqtt_message.payload);
    break;

/* TODO: Add at the end of app_main() */
{
    char topic[32];
    sprintf(topic, "%s/gpio", device_name_get());
    mqtt_subscribe(topic, 0, _ota_on_mqtt, (void *)OTA_TYPE_FIRMWARE, NULL);<!--EndFragment-->
}
```

If you run into any issues, I can try to send you a proper patch for integrating the above points in the coming days.

closed 2021-05-30 by shmuelzon

gvdhoven, 2021-05-31:
I understand that this is not part of the project scope. But... since it is open source i thought why not try and code something myself in C :)

I already have it (the config part) working, with checking if the GPIOs are in fact valid output pins. You might want to check out my fork & branch and decide if that, if i would continue on that route, would be something you would be willing to add to the project; https://github.com/gvdhoven/esp32-ble2mqtt/commit/92b40f7bced91cb11e9637a96806e0be8356f1cb

shmuelzon, 2021-05-31:
> I understand that this is not part of the project scope. But... since it is open source i thought why not try and code something myself in C :)

That is the beauty of open source

Like I said, I don't think controlling GPIOs has a place in what this project tries to accomplish but, obviously, you're more than welcome to add your changes and it might be perfect for someone else as well.
In any case, I can still help if you have any questions or run into any issues.

gvdhoven, 2021-06-01:
Ok, i made a PR, hoping you can consider adding it. The changes are minimal and a fully working example of all my PRs can be seen in the fork i made and the [EspGarage](https://github.com/gvdhoven/esp32-ble2mqtt/tree/gvdhoven/espgarage) branch.

gvdhoven, 2021-06-01:
@shmuelzon ☝️