shmuelzon/esp32-ble2mqtt #111: minimal mqtt example

goosst, 2021-05-30:
**Question**
How do I need to structure /Get messages to read the value of a characteristic. I think I just seem to be looking for a minimal working example.

**Additional context**

I believe to have compiled successfully with (part of) these configurations and ESP-IDF 4.1
```
"mqtt": {
    "server": {
      "host": "192.168.0.xxx",
      "port": 1883,
      "username": "xxxx",
      "password": "xxx"
    },
    "publish": {
      "retain": true
    },
    "topics": {
      "prefix": "/ble2mqtt",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "ble": {
    "services": {
      "definitions": {
        "4fafc201-1fb5-459e-8fcc-c5c9c331914b": {
          "name": "Turbine"
        }
      }
    },
    "characteristics": {
      "definitions": {
        "beb5483e-36e1-4688-b7f5-ea07361b26a8": {
          "name": "power"
        }
      }
    }
  }
```

This is communicating with another ESP32 (acting as a BLE server).
When looking at the published messages, some things also seems to appear when rebooting the ESP32 "server".

```
/ble2mqttfc:f5:c4:55:10:82/GenericAccess/DeviceName ESP32_hover
/ble2mqttfc:f5:c4:55:10:82/GenericAccess/Appearance 0
/ble2mqttfc:f5:c4:55:10:82/GenericAccess/CentralAddressResolution 0
/ble2mqttfc:f5:c4:55:10:82/Turbine/power 48,46,48,48
/ble2mqttfc:f5:c4:55:10:82/Connected false
/ble2mqttfc:f5:c4:55:10:82/Connected true
/ble2mqttfc:f5:c4:55:10:82/Owner BLE2MQTT-6A28
/ble2mqttfc:f5:c4:55:10:82/GenericAccess/DeviceName ESP32_hover
/ble2mqttfc:f5:c4:55:10:82/GenericAccess/Appearance 0
/ble2mqttfc:f5:c4:55:10:82/GenericAccess/CentralAddressResolution 0
/ble2mqttfc:f5:c4:55:10:82/Turbine/power 48,46,48,48
```

But I don't seem to figure out how to get these messages appear by command by sending the relevant \Get mqtt-message.
The value in this test should actually just be zero, so not sure yet how this `48,46,48,48` translates yet (but I guess that's another decoding problem I guess).

Any suggestion or link to a reference?

Thanks!

shmuelzon, 2021-05-30:
Hey,

It seems you do have the hard part behind you. By default, this application will read all the characteristics it detects and subscribe for notifications if characteristic supports it. This means that in most cases, you will not need to manually request updated values from the BLE peripheral rather it will push updates to you.

If you're BLE device doesn't support notifications, you can manually issue a read request by publishing anything to the MQTT topic of relevance with the `get_suffix` suffix.
In your case, manually reading new values for the turbine power characteristic means publishing anything to the `/ble2mqttfc:f5:c4:55:10:82/Turbine/power/Get` topic, e.g.:
```bash
mosquitto_pub -t "/ble2mqttfc:f5:c4:55:10:82/Turbine/power/Get" -m "1"
```

As for the actual payload of that characteristic, it looks like a string to me (which would translate to `0.00`. You can tell the project to parse that characteristic as such by using the following characteristic definition in the configuration file:
```json
      "definitions": {
        "beb5483e-36e1-4688-b7f5-ea07361b26a8": {
          "name": "power",
          "types": [ "utf8s" ]
        }
```

As a side note, please note that it's not customary to prefix MQTT topics with a forward slash and, with you current configuration, there's no separator before the MAC address of you BLE peripheral. I would suggest replacing the `prefix` field in the configuration file with `ble2mqtt/` to address both points.

Let me know if that helps.

goosst, 2021-05-30:
- aaah suffix ... not prefix !
- thanks for the hint on the conversion

thanks a lot :)

closed 2021-05-30 by goosst