shmuelzon/esp32-ble2mqtt #113: Possible issue in ota_on_mqtt function

gvdhoven, 2021-06-01:
Currently there is this code; where you allocate an additional character for the '\0' character.
```
    char *url = malloc(len + 1);
    ota_type_t type = (ota_type_t)ctx;
    ota_err_t err;

    memcpy(url, payload, len);
    url[len] = '\0';
    ...
```

The possible error is at the last line. E.g. should the above code not read:
```
    url[len + 1] = '\0';
```

(or, with some additional tweaks maybe even this)

```
    char *url = malloc(len + sizeof(char));
    ota_type_t type = (ota_type_t)ctx;
    ota_err_t err;

    memcpy(url, payload, len);
    url[len + sizeof(char)] = '\0';
    ...
```

?

shmuelzon, 2021-06-01:
I think you have an off-by-one error in your thought process.
For simplicity, let's say the payload is `bar` making `len` 3. I allocate 3+1=4 bytes. I then copy the first 3 bytes (`b`, `a` and `r`). Finally, I need to NULL-terminate the string.

Since arrays start from index 0, index 0 is `b`, index 1 is `a`, index 2 is `r` and I need to set index 3 (same value as `len`) to the NULL termination character.
Had I tried to set index 4 (`len` + 1), that would have been the fifth element in the array and would be out of bounds for the 4-element array I allocated.

gvdhoven, 2021-06-01:
ok, check, i will close the PR then :)

closed 2021-06-01 by gvdhoven