shmuelzon/esp32-ble2mqtt #117: build error at esp32c3

guachong133, 2021-06-06:
**Question**
hi thanks a lot for you code!
it works fine when "set-target esp32 " while show error when "set-target esp32c3". 
I try to find difference between sdk_config.h in that two version , find the major difference is "CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y" is included at esp32 version but disappear at esp32c3 version. 
but not sure whether it is the reason.
could you please help or guide ,thanks very much. 
the complier error are list below:

**Additional context**
/home/esp32/esp-tools/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(ble.c.obj): in function `gap_cb':
/home/esp32/esp32c3-ble2mqtt/build/../main/ble.c:603: undefined reference to `esp_ble_gap_set_scan_params'
/home/esp32/esp-tools/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(ble.c.obj): in function `esp_gattc_cb':
/home/esp32/esp32c3-ble2mqtt/build/../main/ble.c:716: undefined reference to `esp_ble_gap_start_scanning'
/home/esp32/esp-tools/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(ble.c.obj): in function `ble_scan_start':
/home/esp32/esp32c3-ble2mqtt/build/../main/ble.c:257: undefined reference to `esp_ble_gap_start_scanning'
/home/esp32/esp-tools/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(ble.c.obj): in function `ble_scan_stop':
/home/esp32/esp32c3-ble2mqtt/build/../main/ble.c:267: undefined reference to `esp_ble_gap_stop_scanning'
/home/esp32/esp-tools/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(ble.c.obj): in function `ble_connect':
/home/esp32/esp32c3-ble2mqtt/build/../main/ble.c:277: undefined reference to `esp_ble_gap_stop_scanning'
/home/esp32/esp-tools/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/esp32/esp32c3-ble2mqtt/build/../main/ble.c:282: undefined reference to `esp_ble_gattc_open'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

guachong133, 2021-06-06:
got it, god knows why ble4.2 futures are disabled by default so related functions are disabled.....
thanks a lot ~~

closed 2021-06-06 by guachong133

shmuelzon, 2021-06-06:
That **is** weird, especially considering that scanning is generic feature for all Bluetooth versions.
Which ESP-IDF version were you using? Perhaps Espressif is pushing to use NimBLE instead of Bluedroid?

guachong133, 2021-06-24:
hi my esp-idf version is ESP-IDF v4.4-dev-1405-gb5eef84bc-dirty, it is sitll using Bluedroid , all you code runs perfect well!
i added some feture to change your code from automatically blescan-bleconnect-mqtt publish to manually trigger by console. your code is  really a fantastic job bro thanks never never ending~~

shmuelzon, 2021-06-24:
Thanks for the update, glad it's of use to you