shmuelzon/esp32-ble2mqtt #118: Auto reboot on too many Wi-Fi disconnections in a row?

unrelentingtech, 2021-06-23:
**Is your feature request related to a problem? Please describe.**

After a day or two, my device falls into an endless reconnect loop:

```
I (122381671) WiFi: Disconnected
I (122381671) BLE2MQTT: Disconnected from the network, stopping MQTT
I (122381671) MQTT: Disconnecting MQTT client
W (122381671) wifi:m f probe req l=0

W (122381801) wifi:m f probe req l=0

W (122381921) wifi:m f probe req l=0

W (122382041) wifi:m f probe req l=0

W (122382171) wifi:m f probe req l=0

W (122382291) wifi:m f probe req l=0

W (122382411) wifi:m f probe req l=0

W (122382531) wifi:m f probe req l=0

W (122382651) wifi:m f probe req l=0

W (122382771) wifi:m f probe req l=0

W (122382891) wifi:m f probe req l=0

I (122383261) WiFi: Disconnected
I (122383261) BLE2MQTT: Disconnected from the network, stopping MQTT
I (122383261) MQTT: Disconnecting MQTT client
```

No other errors like malloc ones that random logs on the internet show next to this, here I have *only* `wifi:m f probe req l=0`.

**Describe the solution you'd like**

Count disconnections-in-a-row (increment counter on disconnect, reset to zero on connect), `esp_restart()` if too many, I guess. I can send a PR.

**Describe alternatives you've considered**

I've just noticed that I've been using some random git master rev of esp-idf.. Maybe a tagged release will be more stable. But extra reliability may be desirable anyway.

shmuelzon, 2021-06-23:
There's [something similar](https://github.com/shmuelzon/esp32-ble2mqtt/blob/master/main/ble2mqtt.c#L222) where after 3 MQTT disconnections I reconnect to WiFi.
I guess it can, just as easily, do a complete restart, maybe it's even safer to do that. It won't really add much delay as we boot pretty quickly.

gvdhoven, 2021-08-06:
@shmuelzon got a question on that code;
```
static void mqtt_on_disconnected(void)
{
    static uint8_t num_disconnections = 0;

    ESP_LOGI(TAG, "Disconnected from MQTT, stopping BLE");
    cleanup();

    if (++num_disconnections % 3 == 0)
    {
        ESP_LOGI(TAG,
            "Failed connecting to MQTT 3 times, reconnecting to the network");
        wifi_reconnect();
    }
}
```

This code (currently) does not 'reset' the `num_disconnections` variable when a successful reconnection was done. So if you have 2 disconnects, and then a succesfull connect, the next disconnect would trigger the `wifi_reconnect()`, right? Do you mean to say that it would be possible to replace the `wifi_reconnect()` in this method with an `esp_restart()` ?

shmuelzon, 2021-08-07:
Yes, on both counts. The first was simply to avoid yet another global variable but I didn't think it would matter much since reboots are fairly quick anyways.
As for the  second question, I've recently changed `esp_restart()` calls with `esp_abort()` since the former doesn't always succeed, somehow depending on if WiFi and/or BLE is running...