shmuelzon/esp32-ble2mqtt #119: Does not build - jquery-3.5.1.min.js.gz: No such file or directory

metronidazole, 2021-06-24:
Whenever I try to build this I receive errors at the end of the compile:

> [ 97%] Built target __idf_wifi_provisioning
> [ 97%] Generating ../../www/js/ble2mqtt.js.gz
> [ 97%] Generating ../../www/css/ble2mqtt.css.gz
> [ 97%] Generating ../../www/js/jquery-3.5.1.min.js.gz
> [ 97%] Generating ../../www/index.html.gz
> [ 97%] Generating ../../www/images/spinner.gif.gz
> /bin/sh: line 1: /foo/esp32-ble2mqtt/build/www/js/ble2mqtt.js.gz: No such file or directory
> make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:111: www/js/ble2mqtt.js.gz] Error 1
> make[2]: *** Waiting for unfinished jobs....
> /bin/sh: line 1: /foo/esp32-ble2mqtt/build/www/css/ble2mqtt.css.gz: No such file or directory
> make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:99: www/css/ble2mqtt.css.gz] Error 1
> /bin/sh: line 1: /foo/esp32-ble2mqtt/build/www/js/jquery-3.5.1.min.js.gz: No such file or directory
> /bin/sh: line 1: /foo/esp32-ble2mqtt/build/www/index.html.gz: No such file or directory
> /bin/sh: line 1: /foo/esp32-ble2mqtt/build/www/images/spinner.gif.gz: No such file or directory
> make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:115: www/js/jquery-3.5.1.min.js.gz] Error 1
> make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:107: www/index.html.gz] Error 1
> make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:103: www/images/spinner.gif.gz] Error 1
> make[1]: *** [CMakeFiles/Makefile2:4856: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
> make: *** [Makefile:136: all] Error 2
> make failed with exit code 2
> 

https://pastebin.com/dhFFYDxD

If I manually gzip those files, i receive a different error

> [ 97%] Built target __idf_wifi_provisioning
> make[2]: *** No rule to make target '../www/js/jquery-3.5.1.min.js', needed by 'www/js/jquery-3.5.1.min.js.gz'.  Stop.
> make[1]: *** [CMakeFiles/Makefile2:4856: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
> make: *** [Makefile:136: all] Error 2
> make failed with exit code 2
> 

https://pastebin.com/cqAV2tCy

I'm using esp-idf v4.2.1 which might be the culprit, but might not be.  I have not tried idf v4.1

Config used

```
{
  "network": {
    "wifi": {
      "ssid": "foo",
      "password": "bar"
    }
  },
  "mqtt": {
    "server": {
      "host": "192.168.1.5",
      "port": 1883,
      "ssl": false,
      "username": "bar",
      "password": "foo"
    },
    "publish": {
      "qos": 0,
      "retain": true
    },
    "topics": {
      "prefix": "foobar/"
    }
  }
}

```

shmuelzon, 2021-06-25:
That is odd, since `../www/js/jquery-3.5.1.min.js` is a source and not a target.
Anyways, I see you're using `cmake` + `make`, can you please try to install `ninja` so that would be used (it's the default). That's the combo I use and didn't see these issues.
I'll try to reproduce with `make`.

metronidazole, 2021-06-25:
@shmuelzon 

Cool, it does build using ninja (idf.py -G Ninja build) - thanks. Feel free to close this.

shmuelzon, 2021-07-17:
Hey,

I tested this with ESP-IDF v4.3 and saw a different error, which I fixed.
Compiling with `make` works now, at least on my WSL environment.
Please let me know if you still see an issue on your setup.

Thanks!

commit 29859075 (2021-07-17): Manually generate output directory for www

For some reason, when compiling this project with `make`, the output directory
for the compressed web files wasn't created and generating them failed. `Ninja`
doesn't suffer from this.

Fixes #119
closed 2021-07-17 by shmuelzon