shmuelzon/esp32-ble2mqtt #120: Persistance of bonding information

tgillbe, 2021-07-03:
ESP-IDF version: ESP-IDF v4.1-beta2-247-gb3b7fcf51-dirty

**Describe the bug**
When trying to connect to a device which requires BLE pairing, I get an authentication error. The device is not in pairing mode but was paired previously.

Ther error is "BLE: Authentication failed, status: 0x52"

**To Reproduce**
Bit of context - I am trying to use this to connect to a Flymo EasiLife Go robot lawnmower. It seems to require a LE Secure Connection. The pairing can only occur for the first 3 minutes after the device is powered up.

If I turn off the lawnmower, reboot esp32-ble2mqtt and then turn on the lawnmower the connection works fine. If esp32-ble2mqtt restarts for whatever reason I get the authentication failed message, and need to reboot both the esp32 and the lawnmower to get the connection started up again.

It seems that the bonding key is not being persisted or reused correctly the next time the connection is initiated.

I deleted the call to ble_clear_bonding_info when the config is uploaded over OTA in case it was the problem, but it has not solved the problem.

**Expected behavior**
It should be possible to pair the esp32 once, reboot and persist the bonding key indefinitey.

**Configuration and logs**
Here's a log with a random reboot that I encountered while talking to the lawnmower (not sure what the cause of that reboot was either). The lawnmower is detected but fails to connect:

[4 random-reboot-and-connect-fail.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6758377/4.random-reboot-and-connect-fail.txt)

Here is the pairing working after the esp32 and the lawnmower are rebooted. It looks like the key is persisted in NVS given by the nvs_set_blob bt_cfg_key0 messages:

[working.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6758385/working.txt)

shmuelzon, 2021-07-05:
Hey,

Unfortunately, I don't have any answers yet, just more questions...

> The pairing can only occur for the first 3 minutes after the device is powered up.

Is this true regardless of this project? Does it also happen with a mobile app running nRF Connect / LightBlue?

I don't see anywhere where we initiate the secured connection, so I'm assuming the lawnmower does that, but I also don't see any events being sent our way by the BLE stack that we can respond to. Perhaps it's using the 'just works' method.
Anyways, you say that pairing is only possible during the first 3 minutes so, if the ESP resets for some reason you cannot connect to the lawnmower. Could this be because the lawnmower was already on for more than 3 minutes?

What happens if they are both restarted. Connect and do whatever's needed. Then you turn off the lawnmower, wait for ESP to notice that and turn it on again, will the ESP be able to pair?

BTW - How does a mobile phone behave in the scenario you described? Using generic BLE apps like the nRF of LightBlue, are they able to connect and pair?

Thanks

tgillbe, 2021-07-05:
First of all thanks for your help and for this project - it's very much appreciated.

> > The pairing can only occur for the first 3 minutes after the device is powered up.
> 
> Is this true regardless of this project? Does it also happen with a mobile app running nRF Connect / LightBlue?

Yes - it is true regardless of this project. The lawnmower appears in the paired devices in my mobile and I have paired it on an ubuntu machine and it shows up under bluetoothctl:

```
[bluetooth]# paired-devices
Device 0C:EC:80:DC:3D:63 Mozo
```

If I remove the pairing in bluetoothctl then I get the same issue where I can't connect until I restart the lawnmower again and initiate the connection in the first 3 minutes. Then it is paired and works indefinitely. Works the same on the manufacturer's app - I haven't tried nRF Connect / LightBlue.

> I don't see anywhere where we initiate the secured connection, so I'm assuming the lawnmower does that, but I also don't see any events being sent our way by the BLE stack that we can respond to. Perhaps it's using the 'just works' method.

I'm not too sure about how BLE encryption / pairing works but I think one of the characteristics requires an encrypted connection which is handled here:

https://github.com/shmuelzon/esp32-ble2mqtt/blob/50d9fe3540bd7f623aa6850c5d1414455d517125/main/ble.c#L823-L842

> Anyways, you say that pairing is only possible during the first 3 minutes so, if the ESP resets for some reason you cannot connect to the lawnmower. Could this be because the lawnmower was already on for more than 3 minutes?

The ESP should persist the key through a reboot, so it should work after the lawnmower is on for more than 3 minutes (as it works on my phone and ubuntu computer). Apparently [esp-idf takes care of storing the key in NVS](https://github.com/nkolban/esp32-snippets/issues/414#issuecomment-368281667) and it looks like it is storing the key in the log where the connection is successful. I think the issue is that it does not reuse the key when it tries to reconnect later.

> What happens if they are both restarted. Connect and do whatever's needed. Then you turn off the lawnmower, wait for ESP to notice that and turn it on again, will the ESP be able to pair?

I haven't tried that - I'll give it a go.

If it helps, I have a BLE sniffer and I sniffed the connection initialization between the lawnmower and my mobile while it was initiating the connection. This is a successful connection flow:

[image]

Here is the encryption request from the host:

[image]

And the response from the slave:

[image]

This appears to be this flow: https://fte.com/WebHelpII/BPA600/Content/Documentation/WhitePapers/BTLE/EncryptingTheDataTransmission.htm

I have sniffed the connection failing because it was after 3 minutes after the lawnmower restarted but unfortuantley I can't find it right now. I'll rerun it when I get a chance.

shmuelzon, 2021-07-05:
Hey,

Could you please try applying the following patch:
```patch
diff --git a/main/ble.c b/main/ble.c
index 6b888be..c52c23c 100644
--- a/main/ble.c
+++ b/main/ble.c
@@ -824,11 +824,12 @@ static void esp_gattc_cb(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if,
             if (param->read.status == ESP_GATT_INSUF_AUTHENTICATION ||
                 param->read.status == ESP_GATT_INSUF_ENCRYPTION)
             {
+                ESP_LOGE(TAG, "Failed reading characteristic, %s (is_authenticating: %d", param->read.status == ESP_GATT_INSUF_AUTHENTICATION ?  "INSUF_AUTHENTICATION" : "INSUF_ENCRYPTION", device->is_authenticating);
                 if (!device->is_authenticating)
                 {
                     device->is_authenticating = 1;
                     esp_ble_set_encryption(device->mac,
-                        ESP_BLE_SEC_ENCRYPT_MITM);
+                        ESP_BLE_SEC_ENCRYPT_NO_MITM);
                 }
                 /* Try again */
                 esp_ble_gattc_read_char(g_gattc_if, param->read.conn_id,
```
The first will just add an additional log entry so I'll better understand what going on. The second is just a guess... Since, if I understand correctly, you're not using a pin-code for pairing, man-in-the-middle (MITM) protection won't help, so we'll just try without it.

Let me know...

tgillbe, 2021-07-15:
Thank you very much, that seems to work! Here's the log when it connects:
[connect-ok-trimmed.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6825155/connect-ok-trimmed.txt)

Got a new weird bug now though, if I do the OTA config update the log file gets spammed with errors: 
[connect-bad-trimmed.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6825164/connect-bad-trimmed.txt). The error in question is these two repeated over and over again:

```
D (11111) BLE: Received GATTC event 3 (ESP_GATTC_READ_CHAR_EVT), gattc_if 1
E (11111) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
```

The messages go on until the lawnmower is turned off and on again. I presume it is caused by the `/* Try again */ esp_ble_gattc_read_char` block.

I can avoid getting into the situation by removing the code which clears the ble bonding info on ota config upgrade:

```diff
diff --git a/main/ble2mqtt.c b/main/ble2mqtt.c
index 777dd5a..08d3d44 100644
--- a/main/ble2mqtt.c
+++ b/main/ble2mqtt.c
@@ -107,8 +107,6 @@ static void ota_on_completed(ota_type_t type, ota_err_t err)
     /* All done, restart */
     if (err == OTA_ERR_SUCCESS)
     {
-        if (type == OTA_TYPE_CONFIG)
-            ble_clear_bonding_info();
         vTaskSuspendAll();
         esp_restart();
         xTaskResumeAll();
```

By the way - I would be interested in adding a "command" interface with a basic console which allows you to do things though the web interface, like reboot the device, clear the bonding info and list the devices which are paired etc. Thinking of something like /api/command?q=reboot or /api/command?q=ble list. Is this something you may consider upstreaming if I can get it working? It could possibly be accessed thorugh a MQTT topic too. Then the automatic clear of bonnding information on OTA could be removed.

shmuelzon, 2021-07-15:
Hey,

Glad to hear things are working out for you.
The reason I'm clearing the bonding information when updating the configuration file is because the pin codes might have changed and would now require pairing again. However, now that we have a web interface (and I use that term very loosely), I'm all for adding a button to do that and remove it on OTA updates.
As for the reboot button, it was just added since I needed a way to restart the ESP after modifying files on the SPIFFS partition. Perhaps I should change it to restart or reboot, it would be better than the reset I used.

Anyways, I need to see if there's a way to know in runtime of one authentication mode failed and I should try the other.

Thanks!

tgillbe, 2021-07-15:
Ah - reset is very new! Will be useful. Thank you for sorting out my issue!

closed 2021-07-15 by tgillbe

shmuelzon, 2021-07-17:
Hey @tgillbe,

I'm trying to find a way to make this work without the custom change you have which changes all encryption requests to not use man-in-the-middle protection.
Could you please try to add the following change:
```patch
diff --git a/main/ble.c b/main/ble.c
index 8f91ecf..d96c712 100644
--- a/main/ble.c
+++ b/main/ble.c
@@ -712,6 +712,7 @@ static void gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
 
         xSemaphoreGiveRecursive(devices_list_semaphore);
 
+        ESP_LOGE(TAG, "Authentication complete. key_present: %d, dev_type: %d," " auth_mode: %d", param->ble_security.auth_cmpl.key_present, param->ble_security.auth_cmpl.dev_type, param->ble_security.auth_cmpl.auth_mode);
         if (!param->ble_security.auth_cmpl.success)
         {
             ESP_LOGE(TAG, "Authentication failed, status: 0x%x",
@@ -857,12 +858,16 @@ static void esp_gattc_cb(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if,
             if (param->read.status == ESP_GATT_INSUF_AUTHENTICATION ||
                 param->read.status == ESP_GATT_INSUF_ENCRYPTION)
             {
+                ESP_LOGE(TAG, "Failed reading characteristic, %s (is_authenticating: %d", param->read.status == ESP_GATT_INSUF_AUTHENTICATION ?  "INSUF_AUTHENTICATION" : "INSUF_ENCRYPTION", device->is_authenticating);
                 if (!device->is_authenticating)
                 {
                     device->is_authenticating = 1;
                     esp_ble_set_encryption(device->mac,
                         ESP_BLE_SEC_ENCRYPT_MITM);
                 }
+                else
+                    esp_ble_set_encryption(device->mac,
+                        ESP_BLE_SEC_ENCRYPT_NO_MITM);
                 /* Try again */
                 esp_ble_gattc_read_char(g_gattc_if, param->read.conn_id,
                         param->read.handle, ESP_GATT_AUTH_REQ_NONE);

```
And reproduce the same scenario you had issues with before? When, after the initial pairing, you restart the ESP without restarting the lawn mower.
I want to see if I can use any information I can get understand if I need to use MITM or not.

Thanks in advance!

tgillbe, 2021-07-17:
Updated to lastest master, new patch does not seem to work.

ESP on, lawnmower turned on:
```
I (20591) BLE2MQTT: Discovered BLE device: 0c:ec:80:dc:3d:63 (RSSI: -60), connecting
I (21791) BLE2MQTT: Connected to device: 0c:ec:80:dc:3d:63, scanning
I (22391) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/GenericAccess/DeviceName = Mozo
I (22481) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/GenericAccess/Appearance = 0
I (22591) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/GenericAccess/PeripheralPreferredConnectionParameters = 80,160,0,1000
E (22681) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 0
E (22791) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
E (22791) BT_APPL: earlier enc was not done for same device

E (22891) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
E (22891) BT_APPL: earlier enc was not done for same device

E (22991) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
E (23001) BT_APPL: earlier enc was not done for same device

E (23251) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
E (23261) BT_APPL: earlier enc was not done for same device

E (23321) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
E (23641) BLE: Authentication complete. key_present: 0, dev_type: 2, auth_mode: 1
I (23721) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/98bd0001-0b0e-421a-84e5-ddbf75dc6de4/98bd0003-0b0e-421a-84e5-ddbf75dc6de4 = 80,160,0,1000
I (23901) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/98bd0001-0b0e-421a-84e5-ddbf75dc6de4/98bd0004-0b0e-421a-84e5-ddbf75dc6de4 = 65,117,116,111,109,111,119,101,114,0,0,0,0,0,0,0,0,0,0,0
```

ESP rebooted:

```
I (6250) BLE2MQTT: Connected to device: 0c:ec:80:dc:3d:63, scanning
I (6790) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/GenericAccess/DeviceName = Mozo
I (6880) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/GenericAccess/Appearance = 0
I (6980) BLE2MQTT: Publishing: 0c:ec:80:dc:3d:63/GenericAccess/PeripheralPreferredConnectionParameters = 80,160,0,1000
E (7000) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 0
E (7090) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
E (7100) BLE: Authentication complete. key_present: 0, dev_type: 2, auth_mode: 0
E (7100) BLE: Authentication failed, status: 0x52
W (7110) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (7200) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 0
W (7200) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (7300) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (7310) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (7480) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (7480) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (7510) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (7520) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (7620) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (7620) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (7720) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (7730) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (7900) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (7910) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (8010) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (8010) BT_BTM: Security Manager: BTM_SetEncryption not connected

E (8110) BLE: Failed reading characteristic, INSUF_ENCRYPTION (is_authenticating: 1
W (8120) BT_BTM: Security Manager: BTM_SetEncryption not connected
```

I see you've just edited your post so I'll try with the new patch.

tgillbe, 2021-07-17:
Something is bad with that patch, I get a boot loop with a panic when the lawnmower is on!

```
I (8841) BLE2MQTT: Discovered BLE device: 0c:ec:80:dc:3d:63 (RSSI: -68), connecting
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dbc49  PS      : 0x00060d30  A0      : 0x800dc412  A1      : 0x3ffed710
A2      : 0x00000000  A3      : 0x3ffbb43c  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3ffbb45c  A8      : 0x00000000  A9      : 0x3ffbb43c
A10     : 0x00002293  A11     : 0x3f445be2  A12     : 0x00000006  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

ELF file SHA256: 03dc656260a3abde

Backtrace: 0x400dbc46:0x3ffed710 0x400dc40f:0x3ffed790 0x400d86cd:0x3ffed7b0 0x400d88e1:0x3ffed900

Rebooting...
```

tgillbe, 2021-07-17:
With backtrace decoded @shmuelzon:

```
I (8131) BLE2MQTT: Discovered BLE device: 0c:ec:80:dc:3d:63 (RSSI: -61), connecting
D (8131) MQTT_CLIENT: msg_type=9, msg_id=15589
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dbc49  PS      : 0x00060b30  A0      : 0x800dc412  A1      : 0x3ffed710
0x400dbc49: ble_operation_enqueue at ...\esp32-ble2mqtt\build/../main/ble.c:239

A2      : 0x00000000  A3      : 0x3fff0e64  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3fff0f90  A8      : 0x00000000  A9      : 0x3fff0e64
A10     : 0x00001fcd  A11     : 0x3f445be2  A12     : 0x00000006  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

ELF file SHA256: 03dc656260a3abde

Backtrace: 0x400dbc46:0x3ffed710 0x400dc40f:0x3ffed790 0x400d86cd:0x3ffed7b0 0x400d88e1:0x3ffed900
0x400dbc46: ble_operation_enqueue at ...\esp32-ble2mqtt\build/../main/ble.c:239

0x400dc40f: ble_connect at ...\esp32-ble2mqtt\build/../main/ble.c:309

0x400d86cd: ble_on_device_discovered at ...\esp32-ble2mqtt\build/../main/ble2mqtt.c:348
 (inlined by) ble2mqtt_handle_event at ...\esp32-ble2mqtt\build/../main/ble2mqtt.c:606

0x400d88e1: ble2mqtt_task at ...\esp32-ble2mqtt\build/../main/ble2mqtt.c:660
```

shmuelzon, 2021-07-17:
Thanks, that's because I'm stupid... It's caused by a different commit that I haven't tested with debug logs.
I've updated master (with force, so you'll need to `git fetch && git reset --hard origin/master`)
After that the patch should work, or at least not crash :)

Thanks again for your help, it's much appreciated!

tgillbe, 2021-07-17:
You are fixing my issue, don't thank me, thank you!

Seems to authenticate and stay authenticated correctly with the new patch. Still have the infinite loop if the authentication fails (see the clear-bonding.txt log) but it's not a big deal. Procedure and logs:

- Flash esp32, clear bonding info
- Turn on lawnmower [connect-new.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6835024/connect-new.txt) - works
- Reboot esp32 [esp-reboot.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6835027/esp-reboot.txt) - works
- Reboot lawnmower [mower-reboot.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6835029/mower-reboot.txt) - works
- Clear esp32 bonding info [clear-bonding.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6835040/clear-bonding.txt) - bug

Seems like perhaps an additional flag is needed when authenticating to prevent the esp_ble_gattc_read_char infinite loop?

shmuelzon, 2021-07-17:
Yeah, the infinite loop is a problem since it will block the BLE queue and nothing else will work.
Just to make sure I understand, the "Clear esp32 bonding info" scenario is when the lawn mower was already on for more than 3 minutes, i.e. it doesn't accept new pairing requests?

I'm wondering what would be the correct behavior in this case. Ignore the error and continue to the next characteristics? Disconnect from the BLE peripheral and let it try again later?
Any thoughts?

tgillbe, 2021-07-17:
You are right,I left it 3 mins mins before clearing the bonding info.

I presume the correct thing to do would be to skip over that characteristic and publish some kind of message to MQTT to say that the characteristic cannot be accessed? There must be some BLE devices out there which have security requirements on some characteristics but not others. Could have a "<mac>/<serive>/<characteristic>/Status" topic with a json value { "authenticated": true|false, "access": "readable"|"writable"|"readable and writiable", "encryption": "level 1" | "level 2" | "level 3", etc... }. May add a lot of clutter to the broker though. What do you think?

shmuelzon, 2021-07-18:
I agree the status topic would be overkill. Also, I try to avoid exposing BLE specific things on the MQTT bus since part of the point of this project was to abstract and hide the BLE part.
I think that if pairing fails, I'll just ignore the characteristics that require it and log an error message about it.
I'll try to upload a version with that and would be happy if you could test it again once it's ready.

shmuelzon, 2021-07-23:
Hey @tgillbe,

If/When you can, please try out the [ignore_characteristics_that_require_bonding_when_that_fails](https://github.com/shmuelzon/esp32-ble2mqtt/tree/ignore_characteristics_that_require_bonding_when_that_fails).
It should avoid the endless loop of retrying a characteristic if bonding already failed.

Thanks!

tgillbe, 2021-07-29:
That last patch seems to have sorted that issue. Same sequence as before:

- Flash esp32, clear bonding info
- Turn on lawnmower [connext-new.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6902206/connext-new.txt) - works
- Reboot esp32 [esp32-reboot.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6902221/esp32-reboot.txt) - works
- Reboot lawnmower [mower-reboot.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6902235/mower-reboot.txt) - works
- Clear esp32 bonding info [clear-bonding.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6902252/clear-bonding.txt) - works, get the new error message `Failed reading characteristic 98bd0003-0b0e-421a-84e5-ddbf75dc6de4 as it requires bonding but that failed`

Still a bit of lingering weirdness though. Straight afterwards doing the above:
- Turn off lawnmower. I expected to get the "connection closed" and "disconnected from device" messages but nothing comes through. Turning the lawnmower back on and the esp does not find discover the device again or try to reconnect, presumably because it still thinks it's connected?
- Rebooting both the esp32 and the lawnmower gives ESP_GATTC_READ_CHAR_EVT spam again [second-connect-new.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6902317/second-connect-new.txt). There's also a BLE_DEVICE_DISCOVERED event part way through the connection sequence - don't think I've seen that before but I could be wrong.

Not entirely sure how but I got it working again. I think I turned off the mower, rebooted the esp, cleared bonding data a couple times, rebooted the esp again and turned on the mower.

dharvey242, 2022-01-26:
Hi TGillbe,
Just found this closed thread.  Are you Tgillbe still working on this project?  I have a Easilife 250 and a MQTT HA hub so keen to understand if you made any progress.  Its really annoying to only have the features available in the APP when in range of the mower with Bluetooth and my phone.

shmuelzon, 2022-01-27:
Hey @dharvey242,

Can you also try the [ignore_characteristics_that_require_bonding_when_that_fails](https://github.com/shmuelzon/esp32-ble2mqtt/tree/ignore_characteristics_that_require_bonding_when_that_fails) branch and let me know how that works out for you?

Thanks!

tgillbe, 2022-01-27:
@dharvey242 unfortunately I'm not working on this project any more. I've gotten decently far reverse engineering the protocol the lawnmower uses to communicate so I can send it messages and understand the response, but I could never establish a reliable bluetooth connection. I tried lots approaches (usb dongle on windows / linux) as well as this project but it always required a manual restart of something every 10 or minutes to keep working or the connection would just drop which kind of defeats the point.

I can share what I've learned if you are interested in continuing development? If you're got a way of maintaining a bluetooth connection to the damn thing I would also be interested!

dharvey242, 2022-01-27:
It would be great to see where you got up to code wise.  Do you have a 
GIT?  Odd it timed out.  If you're not killing the connection I assume 
the firmware on their side is doing it.  I will test it with my phone to 
see if the connection is cut automatically and I need to log back in.  
They may have done that for battery saving measures?

tgillbe, 2022-01-30:
@dharvey242 I've thrown it up here: https://github.com/tgillbe/flymo2mqtt

Bear in mind it's a total mess and not at all production ready. I think it'll try to connect to the lawnmower's mqtt interface (exposed using ble2mqtt), and then update the activity every 60 seconds, but you'll really need to take a look at the code to figure out what's going on. I also had it connecting directly using pygatt (see BlePacketInterface) but that wasn't any better. Not sure if the connection link ID will work for you or if it was unique to my lawnmower. If you manage to get it working let me know!

The part hardest to figure out was the packet structure, so if you're looking to do the same, the packets are here:

https://github.com/tgillbe/flymo2mqtt/blob/master/src/mower/packets.py

And they are serialized / deserialized by this:

https://github.com/tgillbe/flymo2mqtt/blob/master/src/mower/packet_framer.py

The client sends "requests" and gets "responses", for example ConnectRequest returns ConnectResponse etc.