shmuelzon/esp32-ble2mqtt #121: Naming characteristics and services doesn't appear to be reflected in MQTT

cheard, 2021-07-04:
How do I rename characteristics and services? I have made the change in my config file, and re-flashed the device, however the characteristics appear by their old ID's, rather than the 'name' as intended.

Below is the snippet from my config.json

```json
       "topics":
        {
            "prefix": "",
            "get_suffix": "/Get",
            "set_suffix": "/Set"
        },
        "ble":
        {
            "services":
            {
                "definitions":
                {
                    "0000ff12-0000-1000-8000-00805f9b34fb":
                    {
                        "name": "device-service"
                    }
                }
            },
            "characteristics":
            {
                "definitions":
                {
                    "0000ff02-0000-1000-8000-00805f9b34fb":
                    {
                        "name": "height-read"
                    },
                    "0000ff01-0000-1000-8000-00805f9b34fb":
                    {
                        "name": "height-write-preset"
                    }
                }
            }
        }
```

The below is a screenshot of the result in MQTT-Explorer.
[image]

Thanks for producing a great bit of software, I'm working on controlling a standing desk from Home-Assistant and would love to share the config once it is complete.

Thanks,

cheard, 2021-07-05:
I realised my error - The 'ble' object had been nested under the 'mqtt' object, which is incorrect. Correcting this meant that the service definition appeared as expected in MQTT. The characteristics didn't update, and changing the names meant that they did not appear in MQTT at all.

shmuelzon, 2021-07-05:
Hey,

I don't see any reason why the services would work but not the characteristics.
As far as I can tell, you've configured the correct UUIDs. Did all of the characteristics stop appearing on the MQTT bus or only the ones you've added definitions for?
Could you please attach your complete configuration file (removing any sensitive information, WiFi, passkeys, etc.)?
Adding a log from when you connect to the BLE peripheral might also help.

Thanks

cheard, 2021-07-05:
Hi Shmeulzon,
Both issues appear to be resolved after flashing the device again - I think it was my own error.

Thanks for your assistance and producing this.

cheard, 2021-07-05:
Apologies - this appears to behave intermittently.

After a device restart, the esp32 will reconnect to the correct device, and publish all of the characteristics, other than the named ones.
[image]

Below is a snippet of my config.json
```json
{
	"wifi": {
		"ssid": "",
		"password": ""
	},
	"mqtt": {
		"server": {
			"host": "",
			"port": 1883,
			"username": "",
			"password": ""
		},
		"topics": {
			"prefix": "ESP32/",
			"get_suffix": "/Get",
			"set_suffix": "/Set"
		},
		"publish": {
			"retain": true
		}
	},
	"ble": {
		"services": {
			"definitions": {
				"0000ff12-0000-1000-8000-00805f9b34fb": {
					"name": "device-service"
				}
			}
		},
		"characteristics": {
			"definitions": {
				"0000ff02-0000-1000-8000-00805f9b34fb": {
					"name": "height-read"
				},
				"0000ff01-0000-1000-8000-00805f9b34fb": {
					"name": "height-write-preset"
				}
			}
		}
	},
	"log": {
		"host": "192.168.50.52",
		"port": 5001
	}

}
```

Logs listed below.
```
Listening on port 5001
2021-07-05 18:50:50.828915 (BLE2MQTT-7FB8): I (4461) BLE2MQTT: Connected to the network, connecting to MQTT
2021-07-05 18:50:50.836006 (BLE2MQTT-7FB8): I (4471) MQTT: Connecting MQTT client
2021-07-05 18:50:50.855540 (BLE2MQTT-7FB8): I (4711) MQTT: MQTT client connected
2021-07-05 18:50:50.857536 (BLE2MQTT-7FB8): I (4711) BLE2MQTT: Connected to MQTT, scanning for BLE devices
2021-07-05 18:50:50.895910 (BLE2MQTT-7FB8): I (4741) BLE2MQTT: Discovered BLE device: f8:30:02:3e:bb:93 (RSSI: -78), connecting
2021-07-05 18:50:51.504321 (BLE2MQTT-7FB8): I (5331) BLE2MQTT: Discovered BLE device: 66:c8:c8:82:e8:26 (RSSI: -84), connecting
2021-07-05 18:50:52.857818 (BLE2MQTT-7FB8): I (6631) BLE2MQTT: Connected to device: f8:30:02:3e:bb:93, scanning
2021-07-05 18:50:53.449651 (BLE2MQTT-7FB8): I (7281) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/GenericAccess/DeviceName = test
2021-07-05 18:50:53.485549 (BLE2MQTT-7FB8): I (7341) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/GenericAccess/Appearance = 0
2021-07-05 18:50:53.551808 (BLE2MQTT-7FB8): I (7381) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/GenericAccess/PeripheralPrivacyFlag = false
2021-07-05 18:50:53.590996 (BLE2MQTT-7FB8): I (7441) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/GenericAccess/PeripheralPreferredConnectionParameters = 80,160,0,1000
2021-07-05 18:50:53.696981 (BLE2MQTT-7FB8): I (7551) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/DeviceInformation/SystemID = 4111251,16265218
[... 4 lines cut ...]
2021-07-05 18:50:53.963575 (BLE2MQTT-7FB8): I (7781) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/DeviceInformation/SoftwareRevisionString = v1.0.170508
2021-07-05 18:50:53.963575 (BLE2MQTT-7FB8): I (7801) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/DeviceInformation/ManufacturerNameString = Manufacturer Name
2021-07-05 18:50:54.063522 (BLE2MQTT-7FB8): I (7861) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/DeviceInformation/IEEE11073-20601RegulatoryCertificationDataList = 254,0,101,120,112,101,114,105,109,101,110,116,97,108
2021-07-05 18:50:54.064655 (BLE2MQTT-7FB8): I (7891) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/DeviceInformation/PnPID = 1,766,0,272
2021-07-05 18:50:54.171037 (BLE2MQTT-7FB8): I (8001) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff03-0000-1000-8000-00805f9b34fb = 2
2021-07-05 18:50:54.206784 (BLE2MQTT-7FB8): I (8061) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff04-0000-1000-8000-00805f9b34fb = 12,0
2021-07-05 18:50:54.270408 (BLE2MQTT-7FB8): I (8101) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff05-0000-1000-8000-00805f9b34fb = 48,48,48,48,48,49
2021-07-05 18:50:54.312048 (BLE2MQTT-7FB8): I (8161) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff06-0000-1000-8000-00805f9b34fb = 116,101,115,116
2021-07-05 18:50:54.376020 (BLE2MQTT-7FB8): I (8211) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff07-0000-1000-8000-00805f9b34fb = 1
2021-07-05 18:50:54.473833 (BLE2MQTT-7FB8): I (8271) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff08-0000-1000-8000-00805f9b34fb = 160,0
2021-07-05 18:50:54.474833 (BLE2MQTT-7FB8): I (8301) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff09-0000-1000-8000-00805f9b34fb = 2
2021-07-05 18:50:54.482114 (BLE2MQTT-7FB8): I (8331) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000ff0a-0000-1000-8000-00805f9b34fb = 0
2021-07-05 18:50:54.576990 (BLE2MQTT-7FB8): I (8391) BLE2MQTT: Publishing: ESP32/f8:30:02:3e:bb:93/device-service/0000fff2-0000-1000-8000-00805f9b34fb = 2,1
2021-07-05 18:51:21.513421 (BLE2MQTT-7FB8): W (35341) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=257 reason=0x0100
2021-07-05 18:51:21.514896 (BLE2MQTT-7FB8): E (35351) BLE: Open failed, status = 0x85
2021-07-05 18:51:22.019859 (BLE2MQTT-7FB8): I (35851) BLE2MQTT: Discovered BLE device: 69:93:fc:0c:78:af (RSSI: -75), connecting
2021-07-05 18:51:52.028257 (BLE2MQTT-7FB8): W (65871) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=257 reason=0x0100
2021-07-05 18:51:52.035466 (BLE2MQTT-7FB8): E (65881) BLE: Open failed, status = 0x85
2021-07-05 18:51:52.126364 (BLE2MQTT-7FB8): I (65971) BLE2MQTT: Discovered BLE device: 71:31:31:95:62:4e (RSSI: -88), connecting
2021-07-05 18:52:22.148331 (BLE2MQTT-7FB8): W (95991) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=257 reason=0x0100
2021-07-05 18:52:22.150328 (BLE2MQTT-7FB8): E (96001) BLE: Open failed, status = 0x85
2021-07-05 18:52:22.436818 (BLE2MQTT-7FB8): I (96261) BLE2MQTT: Discovered BLE device: 69:93:fc:0c:78:af (RSSI: -74), connecting
2021-07-05 18:52:52.440068 (BLE2MQTT-7FB8): W (126281) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=257 reason=0x0100
2021-07-05 18:52:52.441379 (BLE2MQTT-7FB8): E (126281) BLE: Open failed, status = 0x85
```

shmuelzon, 2021-07-05:
That is a bit strange. Just to be sure, if you remove the `characteristics` field from the configuration file, they are published as expected?
Could you, perhaps, compile the application with debug log levels and publish the log here? That will list all of the detected characteristics and their flags (read/write/etc.) and might shed some light.

Thanks

cheard, 2021-07-05:
After a bit more troubleshooting it appears that the named characteristics don't come through in the initial discovery, even though the characteristics did come through before being named.
If that characteristic refreshes, the new value is pushed using the name from the config. 
[debug_log_characteristics.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6764561/debug_log_characteristics.txt)

Log attached but I think I missed the startup - sorry I am just a beginner! You can see the named characteristics being sucessfully published.

Thanks for your help

shmuelzon, 2021-07-05:
Hey,

You're doing great, don't worry about it!

As for the two characteristics you've named see the logs:
```
D (75382) BLE2MQTT: Found new characteristic: service: 0000ff12-0000-1000-8000-00805f9b34fb, characteristic: 0000ff01-0000-1000-8000-00805f9b34fb, properties: 0x4
```
The `0x4` property means `CHAR_PROP_WRITE_NR` so it only supports writing (with no response). That means it won't publish data but you can publish on the MQTT bus (with the `ESP32/f8:30:02:3e:bb:93/device-service/height-write-preset/Set` topic) to make the application write to it.
The second characteristic:
```
D (75522) BLE2MQTT: Found new characteristic: service: 0000ff12-0000-1000-8000-00805f9b34fb, characteristic: 0000ff02-0000-1000-8000-00805f9b34fb, properties: 0x10
```
has a property of `0x10` which would translate to `CHAR_PROP_NOTIFY` meaning the characteristic only supports notifying (pushing) information. The application will register to get notifications and, once the BLE peripheral sends something, you should see it on the MQTT bus.

FYI, the properties value is a bit field and you can see what each bit represents here:
https://github.com/shmuelzon/esp32-ble2mqtt/blob/50d9fe3540bd7f623aa6850c5d1414455d517125/main/ble.h#L10-L17

cheard, 2021-07-06:
Thanks so much for providing this information. 

As best as I can tell, the values are coming through correctly when that characteristic is updated (for the height). I was confused since previously, the characteristic ID would come through when the device initially connected. When using the named characteristics, it doesn't come through with the initial data, only after it is updated. 

The `height-write-preset` does not appear to work correctly all the time. I have attached two logs. I the first log, writing to the `height-write-preset` does not work. I can see that the device is still sending values on `height-read`, so I assume it is still connected.
[log.ble2mqtt.20210706.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6766698/log.ble2mqtt.20210706.txt)

After a reset via the monitor, the `height-write-preset` appears to work again. 

[log.ble2mqtt.20210706_working.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6766699/log.ble2mqtt.20210706_working.txt)

Thanks again for your help.

shmuelzon, 2021-07-06:
The first log had another issue I've yet to fix and doesn't actually connect to your device at all.
I suggest adding a whitelist of device that would disconnect only to your desk (or any other specific devices you want) instead of any other random peripheral. In your configuration file, add:
```json
{
  "ble": {
    "whitelist": [
      "f8:30:02:3e:bb:93"
    ]
  }
}
```

cheard, 2021-07-06:
I've added this in and it appears to be working, thanks.

shmuelzon, 2021-07-17:
Hey,

As I understand, your issues were resolved and I've also push a fix for the issue that appeared in [log.ble2mqtt.20210706.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/6766698/log.ble2mqtt.20210706.txt), so I'm closing this ticket.
If you think there's something that wasn't addressed here, feel free to reopen it.

Also, I'd also appreciate it if you could add the services/characteristic definitions of your standing desk to the [Wiki](https://github.com/shmuelzon/esp32-ble2mqtt/wiki).

Thanks!

closed 2021-07-17 by shmuelzon