shmuelzon/esp32-ble2mqtt #128: Does not build

almirus, 2021-10-08:
I can't build a project on windows
Get a error

> Running ninja in directory d:\distrib\esp\build
> Executing "ninja all"...
> [1/39] Generating ../../www/js/ble2mqtt.js.gz
> FAILED: www/js/ble2mqtt.js.gz
> cmd.exe /C "cd /D D:\Distrib\esp\build\esp-idf\main && gzip -c D:/Distrib/esp/www/js/ble2mqtt.js > D:/Distrib/esp/build/www/js/ble2mqtt.js.gz"
> "gzip" не является внутренней или внешней
> командой, исполняемой программой или пакетным файлом.
> [2/39] Generating ../../www/images/spinner.gif.gz
> FAILED: www/images/spinner.gif.gz
> cmd.exe /C "cd /D D:\Distrib\esp\build\esp-idf\main && gzip -c D:/Distrib/esp/www/images/spinner.gif > D:/Distrib/esp/build/www/images/spinner.gif.gz"
> "gzip" не является внутренней или внешней
> командой, исполняемой программой или пакетным файлом.
> [3/39] Generating ../../www/index.html.gz
> FAILED: www/index.html.gz
> cmd.exe /C "cd /D D:\Distrib\esp\build\esp-idf\main && gzip -c D:/Distrib/esp/www/index.html > D:/Distrib/esp/build/www/index.html.gz"
> "gzip" не является внутренней или внешней
> командой, исполняемой программой или пакетным файлом.
> [4/39] Generating ../../www/js/jquery-3.5.1.min.js.gz
> FAILED: www/js/jquery-3.5.1.min.js.gz
> cmd.exe /C "cd /D D:\Distrib\esp\build\esp-idf\main && gzip -c D:/Distrib/esp/www/js/jquery-3.5.1.min.js > D:/Distrib/esp/build/www/js/jquery-3.5.1.min.js.gz"
> "gzip" не является внутренней или внешней
> командой, исполняемой программой или пакетным файлом.
> [5/39] Generating ../../www/css/ble2mqtt.css.gz
> FAILED: www/css/ble2mqtt.css.gz
> cmd.exe /C "cd /D D:\Distrib\esp\build\esp-idf\main && gzip -c D:/Distrib/esp/www/css/ble2mqtt.css > D:/Distrib/esp/build/www/css/ble2mqtt.css.gz"
> "gzip" не является внутренней или внешней
> командой, исполняемой программой или пакетным файлом.
> [6/39] Performing build step for 'bootloader'
> ninja: no work to do.
> [8/37] cmd.exe /C "cd /D D:\Distrib\esp\build && C:\Users\...-obj-name-len=32 --meta-len=4 --use-magic --use-magic-len"
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1

shmuelzon, 2021-10-09:
Hey, from what I gather, you don't have gzip installed on your machine, or it's not in your PATH.

almirus, 2021-10-09:
@shmuelzon maybe, but I have windows

closed 2021-10-09 by almirus