shmuelzon/esp32-ble2mqtt #129: Amazfit GTR isn't discovered

almirus, 2021-10-09:
[image]
[image]

>2021-10-09 16:59:55.480608 (192.168.1.86): I (499052) BLE2MQTT: Discovered BLE device: c6:82:50:9f:66:36 (RSSI: -73), connecting
> 2021-10-09 17:00:00.428207 (192.168.1.86): W (503552) BT_GATT: gatt_rsp_timeout retry discovery primary service
> 2021-10-09 17:00:05.398220 (192.168.1.86): W (504542) BT_HCI: DiscCmpl evt: hdl=3, rsn=0x8
> 2021-10-09 17:00:10.347947 (192.168.1.86): W (504552) BT_APPL: gattc_conn_cb: if=1 st=0 id=769 rsn=0x8
> 2021-10-09 17:00:15.315009 (192.168.1.86): I (504552) BLE: Connection closed, reason = 0x8

shmuelzon, 2021-10-10:
Technically, yes, and it was discussed [here](https://github.com/shmuelzon/esp32-ble2mqtt/issues/12#issuecomment-459939075).
I don't want to publish everything that's broadcasted as I believe it will be too much but, as a compromise, I suggested broadcasting the information of non-connectable devices, such as your Amazfit.
While I do know now how to understand if a device is connectable or not, I've never gotten around to implementing it.

almirus, 2021-10-10:
@shmuelzon It would be cool!
make two topics by default:
1.name
2.signal level
other topics are public services (to which we can to connect)