shmuelzon/esp32-ble2mqtt #130: Nvidia Shield have different MAC

almirus, 2021-10-09:
> 2021-10-09 17:42:06.823505 (192.168.1.86): I (1553362) BLE2MQTT: Discovered BLE device: 77:80:62:5a:88:c2 (RSSI: -86), connecting
> 2021-10-09 17:42:11.755163 (192.168.1.86): W (1556472) BT_HCI: DiscCmpl evt: hdl=2, rsn=0x3e
> 2021-10-09 17:42:16.791369 (192.168.1.86): I (1560892) BLE2MQTT: Connected to device: 77:80:62:5a:88:c2, scanning
> 2021-10-09 17:42:21.732223 (192.168.1.86): I (1561492) BLE2MQTT: Publishing: 77:80:62:5a:88:c2/GenericAccess/DeviceName = SHIELD
> 2021-10-09 17:42:26.692518 (192.168.1.86): I (1561582) BLE2MQTT: Publishing: 77:80:62:5a:88:c2/GenericAccess/Appearance = 0
> 2021-10-09 17:42:31.620617 (192.168.1.86): I (1561682) BLE2MQTT: Publishing: 77:80:62:5a:88:c2/GenericAccess/CentralAddressResolution = 1
> 2021-10-09 17:42:36.563256 (192.168.1.86): I (1561772) BLE2MQTT: Publishing: 77:80:62:5a:88:c2/00009999-0000-1000-8000-00805f9b34fb/00009996-0000-1000-8000-00805f9b34fb = 2,4,6
> 2021-10-09 17:42:41.555863 (192.168.1.86): I (1562482) BLE2MQTT: Discovered BLE device: 15:4e:f2:2f:1a:50 (RSSI: -39), connecting

[image]

shmuelzon, 2021-10-09:
Some more information available [here](https://www.novelbits.io/bluetooth-address-privacy-ble/).

almirus, 2021-10-10:
oh, thanks!

closed 2021-10-10 by almirus