shmuelzon/esp32-ble2mqtt #132: fatal error: eth_phy/phy.h: No such file or directory

ZhidongYa, 2021-11-03:
**Question**
i run an old projekt with latest Version ESP-IDF, and then this problem comes. i find that the name of headfile has changed, so i change it into esp_wifi/include/phy.h, but this problem still here. why does it happen?

**Additional context**
[image]

shmuelzon, 2021-11-03:
Hey,

Things generally change between ESP-IDF versions. I suggest you use the ESP-IDF version specified in the README file.

ZhidongYa, 2021-11-03:
thank you,

 @shmuelzon I have spend two hour to find the ESP-IDF version specified in the README file. but i still dont find it. Can you recommend one version or introduce the way to find the ESP-IDF version specified in the README file? By the way there is a requirement(cmake_minimum_required(VERSION 3.5)).

shmuelzon, 2021-11-03:
[image]

ZhidongYa, 2021-11-03:
Hey, @shmuelzon 
until now i have tried 3 kinds of version:v4.3.1 ; v4.3 ; v4.0.3. the problem is just same as before. i upload  this old projekt and you can find the reason with it. By the way this projekt was finished before October.2019 and there is no information about Version in README file. i will keep trying different Versions and later write  the result here.

[MyOldProjekt.zip](https://github.com/shmuelzon/esp32-ble2mqtt/files/7469302/MyOldProjekt.zip)

shmuelzon, 2021-11-03:
I can't really help you with projects that I'm not related to in any way.
I suggest you try to contact whoever developed that application.

closed 2021-11-03 by shmuelzon