shmuelzon/esp32-ble2mqtt #137: Figure Values send by BLE Broadcast

JoeRu, 2021-11-24:
**Question**
Hi,
first of - thanks for putting this project here. Flashing and making it work - was quite easy.
I do have some new bee-scale (beehivemonitoring) which uses the weight and temperature and humidity; And Information about Battery and RSSI in another.

This results in this MQTT Information - i figured that the information is packed in the one characteristic. But the generic way of putting it on the MQTT - makes me need to understand how this values where transformed from its original value? If you have some insights on how to reverse engineer this - i really would love to hear some advises.

Many Thanks again. 
 
**Additional context**
Values send by MQTT
I *think* this may be Information about battery levels and RSSI
ec:1b:bd:1c:55:41/0d01c3b8-eff2-44bc-9260-3256eb957268/e11b41bc-6a89-4cf2-828b-9e88b33994c1
19,142,198,37,64,72,220,52

And this could be Weight, Humidity and Temperature.
ec:1b:bd:1c:55:41/0d01c3b8-eff2-44bc-9260-3256eb957268/513849eb-913d-4f80-8c44-3f0685533d6e
146,162,80,0,201,190,65,0,128,46,10,125,105,251

The Values should be something in this range : 25.88Kg, 3.5C, 78.8% hum; Battery 4.076 V, Signal 30%

shmuelzon, 2021-11-24:
Hey,

By default, new characteristics are published with their payloads, byte-by-byte, as unsigned numbers.
So, using the second characteristic as an example, `146,162,80,0,201,190,65,0,128,46,10,125,105,251`, the characteristic is 14 bytes long. The first by is `146`, or `0x92` in hex. The second, `162` is `0xa2` and so on giving us the following bytes:
```
92 a2 50 00 c9 be 41 00 80 2e 0a 7d 69 fb
```
Now the question is how the data is split up between those bytes and there are many options. If we ignore strings (as the payload isn't all in the range of printable characters) we're usually talking a about 1-byte, 2-byte or 4-byte numbers. Each of which can be either unsigned numbers or signed numbers. The humidity would probably be unsigned, but temperature can be negative. Technically the weight might also be if there's an option to zero/tare the scale.
What I would do (after failing to find someone else who already did reversed engineered it), is try to group the above hex string to groups of 1, 2 and 4 in a calculator, convert them to a number (signed or unsigned) and see if that correlates with any known value.
There are two things you need to consider, the order of the bytes which is usually from left to right, to the first 4 bytes, converted to a single 32 bit number, would be `0x0050a292` or, in decimal `5284498`. The other thing to consider is that often these microcontrollers don't use floating point operations and use integers. So what you see on your phone/app as `25.88`, might have actually been sent as and integer `2,588` and divided by 100 on your phone.
The BLE standard does specify some kind of floating point but I've yet to see that in use anywhere.
I've also noticed the seeing a zero byte in the data would often be the highest byte of that group and can be used to separate between groups.

Unfortunately, even with all of the above, I wasn't able to find any correlation with the values you've posted so I have no idea how they should be read. I should try yourself as well as you might find something I didn't. If you are able to find even some of the values, you can configure the application and tell it how to parse the characteristic and it will be sent, as such, on the MQTT bus.

BTW - You can always try to contact the manufacturer, unless they have a specific reason to tie you into their app, they might be willing to provide you with the information you need.

Good luck!

JoeRu, 2021-11-29:
Thanks for looking into it - i did choose then - another way, which may be helpful to others debugging hardware; 
I did download the apk file out of the app-store connected to the hardware und dissembled the xamarin projekt (c# DLLs) with a decompiler. 

The logic to the sensor-data were more complex that anticipated and otherwise - I had to work much much more to figure it out. This way - I just need to translate it to a code - more appropriate to my needs.

shmuelzon, 2021-11-29:
Glad to hear you've got a lead. If you do find a simple mapping and want to share, please add your device to the [Wiki](https://github.com/shmuelzon/esp32-ble2mqtt/wiki).
In the meantime, I'm closing this issue but feel free to reopen it if you run into any issues

closed 2021-11-29 by shmuelzon