shmuelzon/esp32-ble2mqtt #138: BLE stops scanning - "BLE: Scanning starting failed, status: 0x1"

dmascord, 2021-11-25:
esp-idf v4.3.1 is used as recommended.

**Describe the bug**
BLE scanning eventually fails after some time, and does not connect to all 4 thermometers.

**To Reproduce**
Restart BLE2MQTT, have multiple BLE devices within listening range (for example 4 x Mi Thermometer with ATC firmware)

**Expected behavior**
Able to listen to advertising broadcasts continuously, and restart if failed for whatever reason.

**Configuration and logs**
[log1.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/7600137/log1.txt)
[log2.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/7600138/log2.txt)

{
  "network": {
    "wifi": {
      "ssid": "SSID",
      "password": "PASSWORD"
    }
  },
  "mqtt": {
    "server": {
      "host": "mqtt.hostname",
      "port": 1883
    },
    "publish": {
      "retain": true
    }
  },
  "log": {
    "host": "LOCALIP",
    "port": 5000
  }
}

**Additional context**

The ESP32 used was originally a Xiaomi BLE Gateway, a single core ESP-WROOM-32U, and reflashed with this firmware.
CONFIG_FREERTOS_UNICORE=y
And MAC CRC commented out so that it doesn't fail on startup.

dmascord, 2021-11-25:
[debug.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/7600342/debug.log)
This time it looks like WiFi fell over and didn't attempt to reconnect.  Will power off/on and retest again

dmascord, 2021-11-25:
[debug.log.2021-11-25T17:28.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/7600680/debug.log.2021-11-25T17.28.log)
WiFi up, but BLE has stopped

dmascord, 2021-11-25:
[debug.log.2021-11-25T18:27.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/7601076/debug.log.2021-11-25T18.27.log)
Further logs in case useful

shmuelzon, 2021-11-27:
Hey,

I've never tried running this project on a single core, but it doesn't seem like it's working all that much.
Are you sure the ESP-WROOM-32U is a single core? I've seen it referenced as a dual CPU?

In any case, the BLE scan errors of 0x03 seem to be "not enough resources". The WiFi disconnections with event 21 are "station beacon timeout". Not sure what to do with these thought.

Do you have the option of gathering logs from serial? Maybe they'll include some logs that didn't make it over WiFi?

Thanks

dmascord, 2021-11-28:
Hi,

The ESP-WROOM-32U which is shipped inside Xiaomi products seems to be single core, which contradicts the commercial product offering.  Either it is a cost saving thing, or the thing is really dual core, but needs some kind of activation.

I have another ESP-WROOM-32U dev board, which I am trying the same code on (ie, configured as UNICORE) and the behaviour is pretty similar.  It only finds 1 MiThermometer.

[debug.2021-11-28.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/7612596/debug.2021-11-28.log)

I will let it run for a long time, and see whether BLE or WiFi stops.

Cheers,

Damien

dmascord, 2021-11-28:
Ah ha, everything stops eventually:

```
D (29696324) BLE: Received GATTC event 10 (ESP_GATTC_NOTIFY_EVT), gattc_if 1␛[0m
D (29696334) BLE2MQTT: Queuing event BLE_DEVICE_CHARACTERISTIC_VALUE (a4:c1:38:43:48:bb, 00002a19-0000-1000-8000-00805f9b34fb, 0x3fff0020, 1)␛[0m
D (29696344) BLE: Received GATTC event 10 (ESP_GATTC_NOTIFY_EVT), gattc_if 1␛[0m
D (29696344) BLE2MQTT: Queuing event BLE_DEVICE_CHARACTERISTIC_VALUE (a4:c1:38:43:48:bb, 00002a1f-0000-1000-8000-00805f9b34fb, 0x3ffebd28, 2)␛[0m
D (29696364) BLE: Received GATTC event 10 (ESP_GATTC_NOTIFY_EVT), gattc_if 1␛[0m
D (29696364) BLE2MQTT: Queuing event BLE_DEVICE_CHARACTERISTIC_VALUE (a4:c1:38:43:48:bb, 00002a6e-0000-1000-8000-00805f9b34fb, 0x3ffeb154, 2)␛[0m
D (29696384) BLE: Received GATTC event 10 (ESP_GATTC_NOTIFY_EVT), gattc_if 1␛[0m
D (29696384) BLE2MQTT: Queuing event BLE_DEVICE_CHARACTERISTIC_VALUE (a4:c1:38:43:48:bb, 00002a6f-0000-1000-8000-00805f9b34fb, 0x3ffeb3c0, 2)␛[0m
␛[0;32mI (29696404) BLE2MQTT: Publishing: a4:c1:38:43:48:bb/BatteryService/BatteryLevel = 57␛[0m
␛[0;32mI (29696414) BLE2MQTT: Publishing: a4:c1:38:43:48:bb/EnvironmentalSensing/TemperatureCelsius = 180␛[0m
␛[0;32mI (29696424) BLE2MQTT: Publishing: a4:c1:38:43:48:bb/EnvironmentalSensing/Temperature = 1802␛[0m
␛[0;32mI (29696434) BLE2MQTT: Publishing: a4:c1:38:43:48:bb/EnvironmentalSensing/Humidity = 6759␛[0m
D (29703564) BLE2MQTT: Queuing event HEARTBEAT_TIMER␛[0m
D (29703934) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)␛[0m
Failed sending remote log: 12
␛[0;31mE (29960634) BT_HCI: host_recv_pkt_cb couldn't aquire memory for inbound data buffer.
␛[0m
Failed sending remote log: 12
␛[0;31mE (29960634) BT_OSI: list_append osi_calloc failed.
␛[0m
Failed sending remote log: 12
␛[0;31mE (29960644) BT_OSI: list_append osi_calloc failed.
␛[0m
```

dmascord, 2021-11-28:
Full log for the WiFi failure:
[debug.2021-11-28.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/7614665/debug.2021-11-28.log)
Full log for BLE failure:
[debug.2021-11-29.log](https://github.com/shmuelzon/esp32-ble2mqtt/files/7614663/debug.2021-11-29.log)

shmuelzon, 2021-11-29:
In `debug.2021-11-28.log` I can see that the device ran out of memory, which would explain why things started to fail.
The second shows that it failed to start scanning. I'm not sure why but we can add a retry and/or reboot when that happens.

dmascord, 2021-11-29:
Hi,

Yes, we could probably reboot in both cases, but I think there is something fundamentally broken when running in UNICORE mode.  I am running an ESPHome firmware on a dual core ESP32, which gets attributes from the advertisements from all ATC devices that are broadcasting, and I can get consistent results.

Is it possible that because you are pinning on CPU 0 and CPU 1 that when it is running on CPU 0 only, the interrupts collide ?

Cheers,

Damien