shmuelzon/esp32-ble2mqtt #141: Owner has non-ascii characters in value

GrumpyMeow, 2022-01-15:
**Describe the bug**
The value of discovered BLE devices contain non-ascii characters: `ble2mqtt�?(`. I expected this value to be the hostname of the esp32. As such it seems there is an encoding issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and logs**
If applicable, add your configuration file (without passwords) and logs (preferably of DEBUG level) to help understand your problem.

**Additional context**
Add any other context about the problem here.

shmuelzon, 2022-01-16:
That's an interesting one... Coincidently, or not, that's the [only place in the code](https://github.com/shmuelzon/esp32-ble2mqtt/blob/4bb9b11370bec858993c2402b47118f4d5a83261/main/ble2mqtt.c#L295) where I don't treat the device name as a null-terminated string. Still, I would have expected to see there `ble2mqtt` followed by a dash and the last two bytes of the MAC address. Does the device name appear correctly in other places? Such as the uptime and/or free memory topics?

In any case, there is a small bug there that I'll fix soon but please let me know if it fixes the issue you're seeing as well.

closed 2022-01-16 by shmuelzon

GrumpyMeow, 2022-01-16:
May i suggest a better fix? My C knowledge is limited, but:
`
        const char * owner_name = device_name_get();
        mqtt_publish(topic, (uint8_t *)owner_name, strlen(owner_name),
            config_mqtt_qos_get(), config_mqtt_retained_get());
`
This should be a more robust fix. I think that otheriwse configuring a short hostname might still introduce weird characters..

shmuelzon, 2022-01-16:
Generally speaking, you're right that it's more robust that way. However, the device name is of a constant length so, in theory, there's no need to to try an calculate the length.
To be honest, I'm a bit surprised you're seeing this issue in the first place and have my doubts that the change will actually fix what you're seeing since at no time was there garbage at that memory. The 14's character is a null byte and shouldn't cause what you posted.
Can you please tell me if the device name appears correctly in other places? Such as the uptime and/or free memory topics?

shmuelzon, 2022-01-16:
Oh, scratch that... The host name is also configurable by the user via the configuration file. So yes, I have no way of knowing in advance what the length would be. Seeing how I've always left the default, I never saw this problem...

GrumpyMeow, 2022-01-16:
In other places the name is correct. Only the owner value had a “square character” appended at the end. The “square character” is for an character which the font does not support. I’m using MQTT-explorer.
The owner value is “BLE2MQTT-BD60”. I did not override the hostname in the configuration.

PR #142 (2022-01-16): Fix owner payload size
Fixes #141
commit 30731a41 (2022-01-16): Fix owner payload size

Fixes #141
commit 3ca074c9 (2022-01-16): Really fix owner payload size

Really fixes #141