shmuelzon/esp32-ble2mqtt #148: Unsubscribe from MQTT doesn

jumakki, 2022-02-13:
https://github.com/shmuelzon/esp32-ble2mqtt/blob/774b516060e23b976e7bb194daea7bf5161ca15d/main/mqtt.c#L175-L189

shmuelzon, 2022-02-13:
Hey,

Thanks for that, I believe you're right.
Have you tried to modify it locally? Does it fix the memory issue you're seeing?

Thanks

closed 2022-02-13 by shmuelzon

shmuelzon, 2022-02-13:
Great, thanks again for letting me know about this. I've already pushed the fix.
Any help, especially in the form of a pull request, is appreciated.
As for writing characteristics when connected, I haven't had the need for it yet, but it seems to me that publishing to the set topic of a characteristic with a retained flag once, would cause the ESP to write to that characteristics whenever it subscribes to that topic, which is right when the characteristic was discovered.

commit 4c857508 (2022-02-13): Fix MQTT unsubscribing by topic prefix

Fixes #148