shmuelzon/esp32-ble2mqtt #150: Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

Yersi88, 2022-02-22:
**Question**
How do I flash this onto ESP32? I tried to flash ble2mqtt-full.bin on my ESP32 but got the error? 

**Additional context**
```
Using 'COM6' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 58:BF:25:8B:DC:E8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
```

shmuelzon, 2022-02-22:
Hey,

You didn't post what command you've used to flash your devices but I'm guessing that's the issue.
You can find one that should work in the [releases ](https://github.com/shmuelzon/esp32-ble2mqtt/releases) page:
```bash
esptool.py -p <Serial / COM device> -b 115200 --chip esp32 write_flash 0 build/ble2mqtt-full.bin
```

Good day

Yersi88, 2022-02-22:
Hello, 

I tried to use this to flash: https://github.com/esphome/ESPHome-Flasher 

Python command worked tho, many thanks!