shmuelzon/esp32-ble2mqtt #151: Only found Apple devices

sebyldino, 2022-02-22:
Hello,
I followed all step of the tutorial and the gateway works.
But i only see Apple devices (iPhone, iPad...) but not other devices.
I have a bluetooth tracker (Tile mate 2022) and i dont see it in the MQTT topics.

Is there a parameter must be modify for that will work?

Thank you

shmuelzon, 2022-02-23:
Hey,

This application has separate handling for connectable devices and those that are just broadcasters (which I assume the Tile is).
For the latter, we need code to handle each type of device since parsing it's information isn't standard. If you can tell me what it publishes and how to extract the information, then I can try to add support for it.

Good day

ozett, 2022-03-02:
isnt there a standard to identify broadcasters?
[i thought this identifies also broadcasters](https://github.com/wiredlab/esp32-ble-scan-mqtt/blob/master/esp32-ble-scan-mqtt.ino#L131)

[image]

i am interested to get RSSI from (broadcaster) tiles, 
https://github.com/ESPresense/ESPresense/issues/209#issuecomment-1056531083
so i follow here if a solution arises...

shmuelzon, 2022-03-02:
The ESP32 does receive the advertisement packets but it only sends on the MQTT bus those it knows how to parse. There's #12 which requested a generic broadcaster parser to just send the raw data on MQTT but that hasn't happened yet.

ozett, 2022-03-02:
i vote for https://github.com/shmuelzon/esp32-ble2mqtt/issues/12 and a generic broadcaster.
thats my usecase 👍

sebyldino, 2022-03-04:
> Hé,
> 
> Cette application a une gestion distincte pour les appareils connectables et ceux qui ne sont que des diffuseurs (ce que je suppose que Tile est). Pour ce dernier, nous avons besoin de code pour gérer chaque type de périphérique car l'analyse de ses informations n'est pas standard. Si vous pouvez me dire ce qu'il publie et comment extraire les informations, alors je peux essayer d'ajouter un support pour cela.
> 
> Bonne journée

Hello, 

Tile send this message in MQTT:

`{
"id":"E5:7B:12:24:GF:24",
"mac_type":1,
"rssi":-54,
"servicedata":"03007b4322934c6ab9ab9"
}`

I found a solution with Open MQTT Gateway for my project. (https://docs.openmqttgateway.com/)

Thanks