shmuelzon/esp32-ble2mqtt #157: Receiving stop after 705 packets

kostassourvinos, 2022-05-28:
*Please make sure you're using the ESP-IDF version specified in the [README](https://github.com/shmuelzon/esp32-ble2mqtt/blob/master/README.md) file, any other version is not supported.*

**Describe the bug**
A clear and concise description of what the bug is.
Hello, 
I have an BLE oximeter and when I  open it, it connected and send the data, but after of 705 receptions the program it stops to receive data from oximeter and i have to close and open the oximeter in order to automatically reconnect and receive another 705 packets and stop again.
When it doesn't  receive any more data, it understands the disconnection.
There is any limit ?? 
12:44:25.897 -> [0;32mI (12593) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAttribute/DatabaseHash = 33,146,75,205,106,211,53,232,57,124,148,167,58,113,166,62[0m
12:44:25.897 -> [0;32mI (12613) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAttribute/ClientSupportedFeatures = 0[0m
12:44:25.990 -> [0;32mI (12713) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAccess/DeviceName = BerryMed[0m
12:44:26.037 -> [0;32mI (12773) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAccess/Appearance = 3137[0m
12:44:26.084 -> [0;32mI (12833) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 207,61,73,127,127,143,60,73,127,127,143,60,73,127,127,143,60,73,127,127[0m
12:44:26.131 -> [0;32mI (12843) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/ManufacturerNameString = Berry[0m
12:44:26.177 -> [0;32mI (12873) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,59,72,127,127,143,58,72,127,127,143,58,72,127,127,143,57,72,127,127[0m
12:44:26.177 -> [0;32mI (12923) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,57,72,127,127,143,56,72,127,127,143,55,72,127,127,143,55,72,127,127[0m
12:44:26.223 -> [0;32mI (12933) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/ModelNumberString = BM1000C[0m
12:44:26.316 -> [0;32mI (12963) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,54,72,127,127,143,54,72,127,127,143,54,72,127,127,143,53,72,127,127[0m
12:44:26.316 -> [0;32mI (12983) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/HardwareRevisionString = V4.2[0m
12:44:26.316 -> [0;32mI (12993) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,53,72,127,127,143,53,71,127,127,143,52,71,127,127,143,52,71,127,127[0m
12:44:26.316 -> [0;32mI (13003) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/SoftwareRevisionString = V1.02.01.00[0m
12:44:26.316 -> [0;32mI (13023) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,52,71,127,127,143,52,71,127,127,143,52,71,127,127,143,51,71,127,127[0m
12:44:26.362 -> [0;32mI (13083) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,51,71,127,127,143,51,71,127,127,143,51,71,127,127,143,50,71,127,127[0m
..
..
44:54.523 -> [0;32mI (41223) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 134,47,7,74,97,134,46,7,74,97,134,45,6,74,97,134,44,6,71,97[0m

12:45:32.293 -> [0;32mI (79023) BLE2MQTT: Discovered iBeacon broadcaster[0m
12:45:32.293 -> [0;32mI (79033) BLE2MQTT: Discovered BLE device: dc:ed:7c:57:ae:7a (RSSI: -95), not connecting[0m
12:45:39.343 -> [0;32mI (86073) BLE2MQTT: Discovered iBeacon broadcaster[0m
12:45:44.724 -> [0;32mI (91433) BLE2MQTT: Discovered iBeacon broadcaster[0m
12:45:47.058 -> [0;32mI (93823) BLE2MQTT: Discovered iBeacon broadcaster[0m
12:45:52.870 -> [0;32mI (99593) BLE2MQTT: Discovered BLE device: 7c:d9:f4:1d:04:ec (RSSI: -92), not connecting[0m
12:46:00.179 -> [0;32mI (106943) BLE2MQTT: Discovered iBeacon broadcaster[0m
12:46:06.789 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
12:46:06.833 -> [0;33mW (113523) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x8[0m
12:46:06.833 -> [0;33mW (113523) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x8[0m
12:46:06.833 -> [0;32mI (113533) BLE: Connection closed, reason = 0x8[0m
12:46:06.833 -> [0;32mI (113533) BLE2MQTT: Disconnected from device: 00:a0:50:f2:87:95[0m
12:46:17.041 -> [0;32mI (123743) BLE2MQTT: Discovered BLE device: 00:a0:50:f2:87:95 (RSSI: -42), connecting[0m
12:46:18.247 -> [0;32mI (124973) BLE2MQTT: Connected to device: 00:a0:50:f2:87:95, scanning[0m
12:46:18.943 -> >>>> Unhandled characteristic type 19 <<<<
12:46:18.943 -> [0;32mI (125633) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAttribute/DatabaseHash = 33,146,75,205,106,211,53,232,57,124,148,167,58,113,166,62[0m
12:46:18.943 -> [0;32mI (125663) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAttribute/ClientSupportedFeatures = 0[0m
12:46:19.034 -> [0;32mI (125753) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAccess/DeviceName = BerryMed[0m
12:46:19.080 -> [0;32mI (125813) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/GenericAccess/Appearance = 3137[0m
12:46:19.128 -> [0;32mI (125883) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,41,102,127,127,143,41,102,127,127,143,41,102,127,127,143,41,102,127,127[0m
12:46:19.174 -> [0;32mI (125893) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/ManufacturerNameString = Berry[0m
12:46:19.220 -> [0;32mI (125923) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,41,102,127,127,143,41,102,127,127,143,40,102,127,127,143,40,102,127,127[0m
12:46:19.220 -> [0;32mI (125963) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,40,102,127,127,143,40,102,127,127,143,40,102,127,127,143,40,102,127,127[0m
12:46:19.266 -> [0;32mI (125973) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/ModelNumberString = BM1000C[0m
12:46:19.357 -> [0;32mI (126013) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,40,102,127,127,143,40,102,127,127,143,40,102,127,127,143,40,102,127,127[0m
12:46:19.357 -> [0;32mI (126023) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/HardwareRevisionString = V4.2[0m
12:46:19.357 -> [0;32mI (126043) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,40,102,127,127,143,39,101,127,127,143,39,101,127,127,143,39,101,127,127[0m
12:46:19.357 -> [0;32mI (126053) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/DeviceInformation/SoftwareRevisionString = V1.02.01.00[0m
12:46:19.357 -> [0;32mI (126073) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,39,101,127,127,143,39,101,127,127,143,39,101,127,127,143,39,101,127,127[0m
12:46:19.403 -> [0;32mI (126133) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,39,101,127,127,143,39,101,127,127,143,39,101,127,127,143,39,101,127,127[0m
12:46:19.449 -> [0;32mI (126173) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,39,101,127,127,143,39,101,127,127,143,39,101,127,127,143,39,101,127,127[0m
12:46:19.495 -> [0;32mI (126223) BLE2MQTT: Publishing: Kdn/Sensors/00:a0:50:f2:87:95/BerryMed/49535343-1e4d-4bd9-ba61-23c647249616 = 143,39,101,127,127,143,40,102,127,127,143,40,102,127,127,143,41,102,127,127[0

Please advice

shmuelzon, 2022-05-28:
Hey,

Unfortunately,

kostassourvinos, 2022-05-28:
> Hey,
> 
> Unfortunately, there's not a lot to go on here. There's no such limit in the application and I don't think there's a limit in the BLE libraries. I have devices publishing messages every minute and they stay connected and publishing for days. What happens if, for example, you connect your device to a mobile app, e.g. LightBlue, does it keep receiving messages past the first 705? In the meantime, I'll try to understand what the error is that caused the disconnection and we'll continue from there.
> 
> BTW - is it always 705 messages?

Yes, 705 !!!!!
And there are other devices connected, with no problem with them, no stopping, no reconnecting, they continue to send and i receive. 
The only thing is, the oximeter is sending many packets every second of 20 bytes each one.
Kostas

shmuelzon, 2022-05-29:
The disconnection reason `0x08` is, supposedly, a connection timeout.
Just so we can start ruling things out, can you try whitelisting only the oximeter  so no other devices are connected?
Also, what's the distance between the ESP and oximeter? Any reason to believe there are reception problems?

kostassourvinos, 2022-05-29:
> The disconnection reason `0x08` is, supposedly, a connection timeout. Just so we can start ruling things out, can you try whitelisting only the oximeter so no other devices are connected? Also, what's the distance between the ESP and oximeter? Any reason to believe there are reception problems?

The 0x08 happened because I close the  oximeter and after i opened again, only this way i can have data again, otherwise there is no data from oximeter. It stays connected, but the program doesn't receive anything
As for the distance, there are 50 cm away, and I tried with no other device connected and with one more (no problem with the second, was working)