shmuelzon/esp32-ble2mqtt #158: configuration via web interface

dakhnod, 2022-06-07:
It would be great if the gateway was configurable over a web interface, without any configuration files.
Here's what I have in mind:
If no wifi is configured, the ESP starts in AP mode and presents a website to any connected device prompting for the wifi credentials.
After that,  the website allows to scan BLE devices and connected to them.
After a successful connection and service discovery, the website allows to select which characteristics to subscribe to or unsubscribe from.
The website could even be used to whitelist/blacklist devices.

Obviously, all changes should somehow be persisted.

dakhnod, 2022-06-07:
ups, me stupid. sorry.

dakhnod, 2022-06-07:
After playing with the web interface for a bit, I would still say that it would be nice to have a little whitelist button next to each discovered device.
Also, I was a bit suprised to see that the esp tries to connect to every discovered device. Maybe the the default behaviour should be an empty whitelist instead of an empty blacklist?

shmuelzon, 2022-06-08:
Hey,

As you've probably already seen, we have an AP started when WI-Fi isn't configured and a basic web interface.
The web interface has access to the configuration file to load and modify it so implementing a whitelist/backlist from the UI should be possible with modifications to the frontend only. One would need to take into consideration what would happen if you have a whitelist configured, and someone would like to blacklist a device as the whitelist and blacklist can't work together.
A pull request for that would be most welcome.
I guess the same can be done for services and characteristics but those would need to be exposed by the backend as well.

As for the default, if it were to be an empty whitelist, it would mean that on first installation nothing would happen. I'm less fond of that option. Only after the users have installed the application and saw that everything works, then they can start tweaking it to their needs. 

Good day!