shmuelzon/esp32-ble2mqtt #160: Need help installing

stenzilone, 2022-06-28:
Hello, I'm not sure if this is the place to ask this, but I am very new to programming and linux, and I could use some help getting this to work with my current set up. I would like to connect my Chilisleep Ooler to Homekit. I already have homebridge set up and running great, and I am working on getting IR connections set up as soon as I get the IR blaster, but the Ooler uses bluetooth only unfortunately. I have done some research that this is the way to get that to work, but I am unsure what to do. I have homebridge running through Portainer and Docker which is installed on OMV on my server. Any help would be appreciated. Thanks!

stenzilone, 2022-06-30:
so I saw this https://github.com/shmuelzon/esp32-ble2mqtt/wiki/Chilisleep-OOLER and it refers to this project as a starting point. that's why I thought I would ask here. Also, I have a very limited knowledge of programming, so I have no idea what MQTT and ESP32 mean, so I'm not sure what to do. I think this is what I would need, so that I can connect to the bluetooth of the Ooler via a computer (probably my macbook pro since my PCs do not have bluetooth on them). I would love to learn about this kind of stuff because I have tried to do programming in the past (mainly javascript). Any help or a walkthrough would be nice.

shmuelzon, 2022-06-30:
OK, well, the ESP32 is a microcontroller (small CPU + basic peripherals) that supports both Wi-Fi and Bluetooth. You can get it in many shapes and sizes but [this](https://www.aliexpress.com/item/1005002611857804.html) is a very simple example of one.
This project runs on such a device, connects to Bluetooth Low Energy (BLE) devices and sends the information, over Wi-Fi, to your "regular" home network. It basically allows other computers on the network to talk with the Bluetooth device. As for how they can talk with the Bluetooth device, that's where [MQTT](https://mqtt.org/) comes in. It's a messaging protocol allowing different device to send, well, messages, to one another. Assuming you've got the point where and ESP32 device is set up, can communicate with your OOLER and send the information to your home network, you'd still need something else that supports MQTT and can allow you to control it from you mobile phone or whatever. The link you've posted above is the required configuration to allow [Home Assistant](https://www.home-assistant.io/), a home automation platform, to make use of the BLE device and control it from it's UI and automated functions.

That's pretty much the gist of it... If you still want to pursue this route, I suggest you purchase an ESP32 development board like the one linked above. Install the [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/) and/or required programs to load a pre-compiled image of this project onto said ESP32 device and continue from there.

stenzilone, 2022-07-01:
Ok cool. thanks. I'll have to look into it when I have money, but thanks for the help!

shmuelzon, 2022-07-02:
Sure thing. I'm closing this issue for now. Feel free to write here if you have additional questions about this project

closed 2022-07-02 by shmuelzon