shmuelzon/esp32-ble2mqtt #161: SSL not Working

JuGlass, 2022-06-29:
Hello all

**Question**
How i can setup the connection in SSL i already use the 8883 port and paste my path into the "server_cert" but every time it says the "handshake returned" does someone can help me ? i also try with the .pem format and it's not working too 
Thank you for helping me 

**My code for MQTT**
`    "server": {
      "host": "",
      "port": 8883,
      "ssl": true,
      "server_cert": "/certs/root.cer",
      "username": "",
      "password": "!",
      "client_id": "fsqkvfdsmvflbifdlofdqgmrqgnreqgùtrshgfhjtrsùptr"
    }, `

**Picture to help**

[image]

[image]

shmuelzon, 2022-06-30:
Hey,

I've never tried this myself (it came in as a pull request) but from what I can tell from the ESP-IDF documentation and how we load the configuration, the file must be in PEM format.
Just to be sure, are you sure you have the certificate loaded on the file system? Can you browse to the Web UI and verify you can see it there as well?

JuGlass, 2022-07-05:
Hey,
I followed your advice but it doesn't work...
First of all, to test my certificate I used MQTT Explorer to put my certificate and validate the encryption and use of my certificate and it's working well so I know my certificate is good. 
Just to be 100% sure I test with OpenSSL again my .pem is valid so it's not from my certificate (yay for me) 

As for the log, I enabled more logs with the "config menu". and so I have this :

and i found that `I (60915) WiFi: Connected
W (60955) wifi:<ba-add>idx:0 (ifx:0, 84:d4:7e:be:d2:45), tid:0, ssn:2, winSize:64
I (60985) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (61945) esp_netif_handlers: sta ip: 10.161.141.86, mask: 255.255.255.0, gw: 10.161.141.1
I (65095) BLE2MQTT: Connected to the network, connecting to MQTT
I (65095) MQTT: Connecting MQTT client
I (65145) mbedtls: ssl_tls.c:5868 => handshake
I (65145) mbedtls: ssl_cli.c:4406 client state: 0
I (65145) mbedtls: ssl_msg.c:2087 => flush output
I (65145) mbedtls: ssl_msg.c:2099 <= flush output
I (65155) mbedtls: ssl_cli.c:4406 client state: 1
I (65165) mbedtls: ssl_msg.c:2087 => flush output
I (65165) mbedtls: ssl_msg.c:2099 <= flush output
I (65175) mbedtls: ssl_cli.c:980 => write client hello
I (65185) mbedtls: ssl_msg.c:2525 => write handshake message
I (65185) mbedtls: ssl_msg.c:2684 => write record
I (65195) mbedtls: ssl_msg.c:2087 => flush output
I (65195) mbedtls: ssl_msg.c:2107 message length: 244, out_left: 244
I (65205) mbedtls: ssl_msg.c:2112 ssl->f_send() returned 244 (-0xffffff0c)
I (65205) mbedtls: ssl_msg.c:2140 <= flush output
I (65215) mbedtls: ssl_msg.c:2853 <= write record
I (65225) mbedtls: ssl_msg.c:2661 <= write handshake message
I (65225) mbedtls: ssl_cli.c:1448 <= write client hello
I (65235) mbedtls: ssl_cli.c:4406 client state: 2
I (65235) mbedtls: ssl_msg.c:2087 => flush output
I (65245) mbedtls: ssl_msg.c:2099 <= flush output
I (65255) mbedtls: ssl_cli.c:2050 => parse server hello
I (65255) mbedtls: ssl_msg.c:3884 => read record
I (65265) mbedtls: ssl_msg.c:1871 => fetch input
I (65265) mbedtls: ssl_msg.c:2028 in_left: 0, nb_want: 5
I (65275) mbedtls: ssl_msg.c:2053 in_left: 0, nb_want: 5
I (65275) mbedtls: ssl_msg.c:2054 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (65285) mbedtls: ssl_msg.c:2074 <= fetch input
I (65295) mbedtls: ssl_msg.c:1871 => fetch input
I (65295) mbedtls: ssl_msg.c:2028 in_left: 5, nb_want: 74
I (65305) mbedtls: ssl_msg.c:2053 in_left: 5, nb_want: 74
I (65315) mbedtls: ssl_msg.c:2054 ssl->f_recv(_timeout)() returned 69 (-0xffffffbb)
I (65315) mbedtls: ssl_msg.c:2074 <= fetch input
I (65325) mbedtls: ssl_msg.c:3958 <= read record
I (65335) mbedtls: ssl_cli.c:2363 server hello, total extension length: 25
I (65335) mbedtls: ssl_cli.c:2587 <= parse server hello
I (65345) mbedtls: ssl_cli.c:4406 client state: 3
I (65355) mbedtls: ssl_msg.c:2087 => flush output
I (65355) mbedtls: ssl_msg.c:2099 <= flush output
I (65365) mbedtls: ssl_tls.c:2855 => parse certificate
I (65365) mbedtls: ssl_msg.c:3884 => read record
I (65375) mbedtls: ssl_msg.c:1871 => fetch input
I (65375) mbedtls: ssl_msg.c:2028 in_left: 0, nb_want: 5
I (65385) mbedtls: ssl_msg.c:2053 in_left: 0, nb_want: 5
I (65395) mbedtls: ssl_msg.c:2054 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (65395) mbedtls: ssl_msg.c:2074 <= fetch input
I (65405) mbedtls: ssl_msg.c:1871 => fetch input
I (65405) mbedtls: ssl_msg.c:2028 in_left: 5, nb_want: 4887
I (65415) mbedtls: ssl_msg.c:2053 in_left: 5, nb_want: 4887
I (65425) mbedtls: ssl_msg.c:2054 ssl->f_recv(_timeout)() returned 1361 (-0xfffffaaf)
I (65825) mbedtls: ssl_msg.c:2053 in_left: 1366, nb_want: 4887
I (65825) mbedtls: ssl_msg.c:2054 ssl->f_recv(_timeout)() returned 1440 (-0xfffffa60)
I (66485) mbedtls: ssl_msg.c:2053 in_left: 2806, nb_want: 4887
I (66485) mbedtls: ssl_msg.c:2054 ssl->f_recv(_timeout)() returned 1216 (-0xfffffb40)
I (76495) mbedtls: ssl_msg.c:2053 in_left: 4022, nb_want: 4887
I (76495) mbedtls: ssl_tls.c:5879 <= handshake
W (76495) esp-tls: Failed to open new connection in specified timeout
E (76495) TRANSPORT_BASE: Failed to open a new connection
E (76515) MQTT_CLIENT: Error transport connect
I (76515) MQTT: MQTT client disconnected
I (76515) BLE2MQTT: Disconnected from MQTT, stopping BLE```

so I think that when the client and the server shake hands, the client says "it's good that we have the same certificate, goodbye". 
From my point of view, it comes from the client itself. 
Do you know where I can increment the verbose debugging to have all the flags where it can check one by one?

And yes, my config file is aligned with the file name, it's just because I'm researching this issue ahah 

And thanks for taking the time to answer me

shmuelzon, 2022-07-05:
Hey,

I'm seeing something else in the log. Mainly, the actual error message: `Failed to open new connection in specified timeout`. The encryption code times out when trying to do the SSL handshake.
In addition, take a look at the timestamp between these two lines:
```
I (66485) mbedtls: ssl_msg.c:2054 ssl->f_recv(_timeout)() returned 1216 (-0xfffffb40)
I (76495) mbedtls: ssl_msg.c:2053 in_left: 4022, nb_want: 4887
```
You'll notice that 10 seconds have passed which is, in fact, the default timeout of MQTT:
https://github.com/espressif/esp-mqtt/blob/a21c387d6280260894981c22494017c893d505b9/lib/include/mqtt_config.h#L43

I'm not sure what would cause it to hang for so long but if you want to try and increase it, you can add the following change:
```patch
diff --git a/main/mqtt.c b/main/mqtt.c
index 5efbfe5..f3f4536 100644
--- a/main/mqtt.c
+++ b/main/mqtt.c
@@ -297,6 +297,7 @@ int mqtt_connect(const char *host, uint16_t port, const char *client_id,
         .lwt_msg = lwt_msg,
         .lwt_qos = lwt_qos,
         .lwt_retain = lwt_retain,
+        .network_timeout_ms = 100000,
     };

     ESP_LOGI(TAG, "Connecting MQTT client");
```

For the sake of debugging, I'd also suggest moving the ESP32 closer to you access point to rule out networking issues.

JuGlass, 2022-07-25:
Hello 

It works now, maybe the waiting time was the problem. but in any case I now have a stable connection in MQTTS 
Thanks for your help 

by the way, I found a solution to read the "value" I saw that some question was asked about it. 
My device has a value of "30,44,0,3,1,0,34,8,48,48,46,48,48,50,54" 
and I write a python code that converts the decimal in hexadecimal and in ascii that here I hope that it can help the community of this project 

`micrometer = [30,44,0,3,1,0,34,8,48,48,48,46,48,48,50,54] #Value of the device
n=micrometer[10:] #I only take the values after the 10th value of the list for my own needs
mapped_numbers = list(map(lambda a: hex(a).replace("0x",""), n))#converts to hex and removes the 0x from the prefix values
mapped_numbers1 = list(map(lambda a: bytearray.fromhex(a).decode(), mapped_numbers))#converts to ASCII
res = ''.join(mapped_numbers1)#resituate the numbers without list 
print(res)`

the result is 0.0026 it's exactly the same value I have on my device screen

shmuelzon, 2022-07-25:
Hey,

Glad to hear it worked out for you!
Regarding the characteristic values, you don't really need to convert the bytes to hex and then to string. You could just have
```python
list(map(lambda a: chr(a), n))
```
Which will give you the characters. In any case, you can also use this application to convert it to string. I don't know what the first 10 bytes are, but you can tell it that the rest are a string with something like the following configuration:
```json
"characteristics": {
  "definitions": {
    "xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx": {
      "name": "SomeName",
      "types": [
        "uint8",
        "uint8",
        "uint8",
        "uint8",
        "uint8",
        "uint8",
        "uint8",
        "uint8",
        "uint8",
        "uint8",
        "utf8s"
      ]
    }
  }
}
```
The above will publish something like `30,44,0,3,1,0,34,8,48,48,0.0026`. Thinking about it now, it might be worthwhile adding a new "ignore" type to the characteristic definition so we can ignore the first 10 bytes and not publish them at all. It will publish only the `0.0026` section which is what you're interested in. Anyway, that's a different topic.

Can I close this issue or is there anything else?

JuGlass, 2022-07-26:
Yes, Thank you

closed 2022-07-26 by JuGlass