shmuelzon/esp32-ble2mqtt #162: Recommended ESP32 Board, case etc.

nodecentral, 2022-07-20:
**Question**
Can anyone recommend an appropriate ESP32 Board to do the work with ?

**Additional context**
Hi, I’m new to all this ESP type stuff, and can see SoC mentioned too, but keen to give it a go, an want to make sure I start off with the right parts/components..

shmuelzon, 2022-07-20:
All you need is an ESP32 module, preferably one with USB so you can easily connect it to your PC for installing the application and all. I have a few that are called ESP32 D1 Mini (not to be confused with ESP8266).
For example: https://www.aliexpress.com/item/32816065152.html or https://www.amazon.com/HiLetgo-Development-ESP8285-Wireless-Internet/dp/B07BK435ZW

No additional peripherals are needed as we're only using Wi-Fi and Bluetooth that are a part of the SoC (system on chip).

nodecentral, 2022-07-20:
Thanks @shmuelzon  - this seems the most popular on Amazon (uk) - https://www.amazon.co.uk/ESP-32S-Development-2-4GHz-Bluetooth-Antenna/dp/B071JR9WS9 - is this suitable ?

shmuelzon, 2022-07-20:
Yeah, both seem fine. They look pretty much the same to me so whatever's easier to get