shmuelzon/esp32-ble2mqtt #163: ../main/resolve.c:3:10: fatal error: mdns.h: No such file or directory

jonlt, 2022-07-21:
**Question**
I get this error:

    ../main/resolve.c:3:10: fatal error: mdns.h: No such file or directory

What am i missing to install?

**Additional context**
[build-log.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/9163176/build-log.txt)

```
HINT: Maybe you forgot to import tcpip_adapter_set_hostname, esp_read_mac, tcpip_adapter_set_hostname, ESP_NETIF_DEFAULT_ETH, esp_netif_new, gpio_pad_select_gpio, esp_eth_phy_new_lan8720, esp_netif_attach, esp_netif_init library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
ninja failed with exit code 1, output of the command is in the /home/jlt/dev/esp32-ble2mqtt/build/log/idf_py_stderr_output_65817 and /home/jlt/dev/esp32-ble2mqtt/build/log/idf_py_stdout_output_65817
```

shmuelzon, 2022-07-22:
Hey, please use the ESP-IDF version described in the README file and try again.

jonlt, 2022-08-04:
thank you! using v4.4.2 fixed the issue

closed 2022-08-04 by jonlt