shmuelzon/esp32-ble2mqtt #167: Issue trying to write characteristic to ThermoPro TP-25

rlarranaga, 2022-10-27:
**Question**
How do i properly write a characteristic to an attribute?

**Additional context**
I have a Thermopro TP-25 meat thermometer that i am trying to use in my home automation system.

I did some sniffing on the Bluetooth protocol between the thermometer and the provided app, and i can see that the Thermometer requires that a string be written to a characteristic to connect to it. After that, you can subscribe to the notifications in the thermometer.

The packet sent to the thermometer in order to connect is as follows (Screenshot taken from wireshark)

[image]

I am running esp32-ble-mqtt on an esp32-wroom-32. The device detects my Thermometer, as shown by the logs:

```
I (2479817) BLE2MQTT: Connected to device: ee:ca:79:d7:d4:27, scanning
I (2480437) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/GenericAccess/DeviceName = Thermopro
I (2480497) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/GenericAccess/Appearance = 0
I (2480537) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/GenericAccess/PeripheralPreferredConnectionParameters = 120,200,4,400
I (2480597) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/GenericAccess/CentralAddressResolution = 1
I (2480737) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/1086fff0-3343-4817-8bb2-b32206336ce8/1086fff1-3343-4817-8bb2-b32206336ce8 = 1,0,62,143,138,108,53,238,38,227,248,241,126,253,94,238,180,20,24,0
I (2480797) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/1086fff0-3343-4817-8bb2-b32206336ce8/1086fff2-3343-4817-8bb2-b32206336ce8 = 224,2,1,2,229,108,53,238,38,227,248,241,0,32,72,0,0,32,2,0
I (2480897) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/1086fff0-3343-4817-8bb2-b32206336ce8/1086fff2-3343-4817-8bb2-b32206336ce8 = 224,2,1,2,229,108,53,238,38,227,248,241,0,32,72,0,0,32,2,0
I (2480937) BLE2MQTT: Publishing: ee:ca:79:d7:d4:27/1086fff0-3343-4817-8bb2-b32206336ce8/1086fff1-3343-4817-8bb2-b32206336ce8 = 1,0,62,143,138,108,53,238,38,227,248,241,126,253,94,238,180,20,24,0
I (2483897) BLE2MQTT: Discovered BLE device: 42:79:48:2b:ad:6e (RSSI: -73), connecting
```

I also have a messaged published through mqtt-explorer that has the retain flag on, to write to this characteristic on connect:

[image]

However, i do not see the device staying connected to esp32-ble-mqtt. I also don't see anywhere in the logs (as shown above) where is this message published. So i am not sure if it is published, and if it is, it is not working.

What would be the best way to troubleshoot this issue?

Thanks

shmuelzon, 2022-10-27:
Hey,

Continuing over from our conversation [here](https://community.openmqttgateway.com/t/thermopro-tp25-and-switchbot-meter/2043), the set payload should be: `1,9,62,143,138,108,53,238,38,227,248,241` (the second byte should be `9` not `0`. I edited that post when I noticed the error but I guess it was after you've already read it.
Other than that, things look fine. I don't think there are log messages that should, by default, for write attempts but those are usually followed by an automatic read that, once done, should be visible in the logs and on the MQTT bus.

Hope that helps

rlarranaga, 2022-10-28:
That appears to have done the trick, thanks a lot! i can't believe i missed.

In the Logs, i see a number of the following errors:

`E (3440428) BLE: Open failed, status = 0x85`

Eventually, i also see a significant amount of these:

`E (10638927) BT_BTM: BTM_GetSecurityFlags false`

What are those attributed to?
Thanks again.

shmuelzon, 2022-10-28:
> That appears to have done the trick, thanks a lot! i can't believe i missed.

So now you're able to use the thermometer and get updates from it?

> In the Logs, i see a number of the following errors: `E (3440428) BLE: Open failed, status = 0x85`

That specific error is a `GATT Error` not sure exactly what causes it though, haven't seen those before.

> Eventually, i also see a significant amount of these: `E (10638927) BT_BTM: BTM_GetSecurityFlags false`

This is something that comes from the ESP32 Bluetooth libraries, it's printed whenever there aren't any security flags saved for the device that's queried but, again, no idea what exactly they mean.
I can't even tell if it's errors related to your thermometer or a different device it's trying to connect to. If you haven't already, I suggest whitelisting only the BLE device you're interested in as there are typically many of them around...

rlarranaga, 2022-11-01:
> So now you're able to use the thermometer and get updates from it?

I am able to receive the updates, yes. I still need to work on the protocol and formatting of the packets. 
The only problem that I see right now is that after a while, the Thermometer turns off, and the ESP32 appears to stop working, but I need to investigate this further before I pinpoint which of the devices is the culprit

> That specific error is a GATT Error not sure exactly what causes it though, haven't seen those before.

I have seen that error on some of the other issues opened here. I will try to investigate on what is the issue, but for now, I am whitelisting devices to the ones that I actually want to use (The thermometer, a couple of SwitchBot meters, and my smartwatch)

> This is something that comes from the ESP32 Bluetooth libraries, it's printed whenever there aren't any security flags saved for the device that's queried but, again, no idea what exactly they mean. [...]

Yes, I have a blacklist in place now, and will start troubleshooting from there

shmuelzon, 2022-11-01:
OK, great. Let me know if there's anything else I can help with..

rlarranaga, 2022-11-02:
I am happy to open a separate ticket as soon as i have enough information on why i see disconnections after a couple of hours.
Additionally, i have a couple of issues with some switchbot meter devices, but happy to open a separate issue also.
Thanks!

tinkeringtuck, 2023-10-07:
> I am happy to open a separate ticket as soon as i have enough information on why i see disconnections after a couple of hours.
> 
> Additionally, i have a couple of issues with some switchbot meter devices, but happy to open a separate issue also.
> 
> Thanks!

Hey @rlarranaga, sorry if this isn't the best place to ask but I'm very interested in learning what you figured out specifically for the TP25 Bluetooth integration. Is there somewhere I could find explore that in more detail and implement what you developed specifically for that device?

sorryusernameisalreadytaken, 2023-10-14:
> Hey @rlarranaga, sorry if this isn't the best place to ask but I'm very interested in learning what you figured out specifically for the TP25 Bluetooth integration. Is there somewhere I could find [...]

I really want to know the same