shmuelzon/esp32-ble2mqtt #168: Unable to build

Minims, 2022-10-31:
**Describe the bug**
Hello, I am unable to build the project

Tried on Mac M1 and on AMD64 ubuntu 22.04

```
[1162/1175] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/ble.c.objninja: build stopped: subcommand failed.
HINT: esp_mac.h header file is not included by esp_system.h anymore. It shall then be manually included with #include "esp_mac.h"
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
HINT: Function esp_eth_mac_new_esp32() has been refactored to accept device specific configuration and MAC specific configuration.
Please refer to the Ethernet section of Networking migration guide for more details.
HINT: Function esp_eth_phy_new_lan8720() has been removed, please use esp_eth_phy_new_lan87xx() instead.
Please refer to the Networking migration guide, section PHY Address Auto-detect, for more details.
ninja failed with exit code 1, output of the command is in the /Users/alex/.minims/usr/src/github/esp32-ble2mqtt/build/log/idf_py_stderr_output_17920 and /Users/alex/.minims/usr/src/github/esp32-ble2mqtt/build/log/idf_py_stdout_output_17920

```

**To Reproduce**
Follow build process on master branch 

**Expected behavior**
Build succes

**Configuration and logs**
If applicable, add your configuration file (without passwords) and logs (preferably of DEBUG level) to help understand your problem.
[idf_py_stdout_output_17920.zip](https://github.com/shmuelzon/esp32-ble2mqtt/files/9903243/idf_py_stdout_output_17920.zip)

**Additional context**
Add any other context about the problem here.

shmuelzon, 2022-10-31:
Hey, the `master` branch isn't, currently, supported. Please use the tag mentioned in the README file.

Minims, 2022-11-01:
OK it is the idf version 4.4

shmuelzon, 2022-11-01:
Yep, that's the one. Hope you were able to build it..

MrJeems, 2023-01-27:
I too ran into this issue.  Being new to both git and compiling, I dabbled on and off for a year before catching the v4.4 requirement.

Since we are tied to the specific branch rather than the most recent esp-idf, I suggest adding the git clone line under item 1 of the Compiling section, just before item 2.

Something like: 
  Be sure to clone the v4.4 branch of esp-idf.git
  `git clone --branch release/v4.4 --recursive https://github.com/espressif/esp-idf.git`

shmuelzon, 2023-01-28:
Hey @MrJeems, thanks for the input. The problem is that ESP-IDF also requires a few steps for it to install dependencies and things like that which are better left for their documentation and, to anyone new to it, would be missing from ours.
I do provide pre-built images whenever a release is made, that should be simpler for anyone who's just starting out.