shmuelzon/esp32-ble2mqtt #169: Switchbot meter broadcasters

rlarranaga, 2022-11-07:
**Question**
How do i get the Switchbot meter device to get discovered properly by esp32-ble-mqtt?

**Additional context**
I have 3 switchbot meter devices. These devices are broadcasters that report temperature and Humidity.
They get discovered by esp32-ble-mqtt:

```
 (7188) BLE2MQTT: Discovered BLE device: df:06:40:1a:4d:fe (RSSI: -79), connecting
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
W (11278) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
I (22048) BLE2MQTT: Discovered BLE device: 5a:0a:3b:3a:c4:ca (RSSI: -89), not connecting
I (22248) BLE2MQTT: Connected to device: df:06:40:1a:4d:fe, scanning
I (22908) BLE2MQTT: Publishing: df:06:40:1a:4d:fe/GenericAccess/DeviceName = WoSensorTH
I (22958) BLE2MQTT: Publishing: df:06:40:1a:4d:fe/GenericAccess/Appearance = 768
I (23008) BLE2MQTT: Publishing: df:06:40:1a:4d:fe/GenericAccess/PeripheralPreferredConnectionParameters = 0,0,0,0
I (23058) BLE2MQTT: Publishing: df:06:40:1a:4d:fe/GenericAccess/CentralAddressResolution = 0
```

However, they are being discovered with no services.

If i scan and connect to the devices through nrfconnect, i see the following:

[image]
[image]

Shouldn't service `cba20d00-224d-11e6-9fb8-0002a5d5c51b` and its characteristics be automatically discovered?

If there is any technical information that needs to be provided for this device to be detected properly, i would be happy to get some captures. Additionally, I believe there is information about these devices in the theengs github project, specifically here:

https://github.com/theengs/decoder/blob/development/src/devices/SBMT_json.h

Thanks!

shmuelzon, 2022-11-08:
Hey, all the services and characteristics should be detected and published when connecting to the device and I agree the list above seems a bit short.
Just to be sure, are you sure the missing ones are "regular" GATT services and not something that's broadcasted without a connection?
Have you whitelisted or blacklisted any services and/or characteristics in your configuration file? Can you post it here (after removing any sensitive information)?
Perhaps we discover the services but, for some reason, don't get the values for them when trying to read them?
Can you enable debug logs and try again? Maybe we'll see something of interest there.
Thanks

rlarranaga, 2022-11-08:
Hello @shmuelzon , answers below:

> Just to be sure, are you sure the missing ones are "regular" GATT services and not something that's broadcasted without a connection?

They look like regular GATT services to me, but i am not sure. How can i check?

> Have you whitelisted or blacklisted any services and/or characteristics in your configuration file? Can you post it here (after removing any sensitive information)?

No, only devices, see full config below:

```
{
  "network": {
    "wifi": {
      "ssid": "XXX",
      "password": "XXX"
    }
  },
  "mqtt": {
    "server": {
      "host": "XXX",
      "username": "XXX",
      "password": "XXX",
      "port": 1883
    },
    "publish": {
      "retain": true
    }
  },
  "ble": {
    "whitelist": ["xx:xx:xx:xx:xx:xx","xx:xx:xx:xx:xx:xx","xx:xx:xx:xx:xx:xx","df:06:40:1a:4d:fe","xx:xx:xx:xx:xx:xx"]
  },
  "log": {
    "host": "XXX",
    "port": 5000
  }
}
```

> Can you enable debug logs and try again? Maybe we'll see something of interest there.

I have enabled debug logs, but i encountered I run into a stack overflow issue when enabled, and the device resets. In any case, here an extract of the logs, from the moment one of the devices is detected to the moment the device overflows.

```
I (36280) BLE2MQTT: Discovered BLE device: df:06:40:1a:4d:fe (RSSI: -80), connecting
D (36290) BLE: Enqueue: type: 0, device: df:06:40:1a:4d:fe, char: 00000000-0000-0000-0000-000000000000, len: 0, val: 0x0
D (36600) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (36800) BLE: Queue timer expired
D (36800) BLE: Dequeue: type: 0, device: df:06:40:1a:4d:fe, char: 00000000-0000-0000-0000-000000000000, len: 0, val: 0x0
D (36810) BLE: Received GAP event 18 (ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT)
D (42350) BLE: Received GATTC event 40 (ESP_GATTC_CONNECT_EVT), gattc_if 1
D (42350) BLE: GATTC event 40 wasn't handled
D (42360) BLE: Received GATTC event 2 (ESP_GATTC_OPEN_EVT), gattc_if 1
D (42370) BLE: Received GAP event 7 (ESP_GAP_BLE_SCAN_START_COMPLETE_EVT)
D (43210) BLE: Received GATTC event 46 (Invalid GATTC event), gattc_if 1
D (43220) BLE: GATTC event 46 wasn't handled
D (43260) BLE: Received GATTC event 18 (ESP_GATTC_CFG_MTU_EVT), gattc_if 1
D (43260) BLE2MQTT: Queuing event BLE_DEVICE_CONNECTED (df:06:40:1a:4d:fe)
I (43260) BLE2MQTT: Connected to device: df:06:40:1a:4d:fe, scanning
D (43280) MQTT: Subscribing to df:06:40:1a:4d:fe/Connected
D (43280) MQTT_CLIENT: mqtt_enqueue id: 63993, type=8 successful
D (43280) OUTBOX: ENQUEUE msgid=63993, msg_type=8, len=34, size=34
D (43290) MQTT_CLIENT: Sent subscribe topic=df:06:40:1a:4d:fe/Connected, id: 63993, type=8 successful
D (43300) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (43310) BLE: GATTC event 7 wasn't handled
D (43310) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (43320) BLE: GATTC event 7 wasn't handled
D (43320) BLE: Received GATTC event 7 (ESP_GATTC_SEARCH_RES_EVT), gattc_if 1
D (43330) BLE: GATTC event 7 wasn't handled
D (43330) BLE: Received GATTC event 6 (ESP_GATTC_SEARCH_CMPL_EVT), gattc_if 1
[... 96 lines cut ...]
D (44080) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (44090) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
D (44100) MQTT_CLIENT: mqtt_message_receive: read_len=3
D (44100) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (44110) MQTT_CLIENT: msg_type=9, msg_id=12615
D (44110) MQTT_CLIENT: pending_id=12615, pending_msg_count = 1
D (44120) OUTBOX: DELETED msgid=12615, msg_type=8, remain size=0
D (44120) MQTT_CLIENT: deliver_suback, message_length_read=5, message_length=5
D (44380) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (44380) BLE: Received GAP event 3 (ESP_GAP_BLE_SCAN_RESULT_EVT)
D (44450) BLE: Queue timer expired
D (44450) BLE: Dequeue: type: 4, device: df:06:40:1a:4d:fe, char: 00002a05-0000-1000-8000-00805f9b34fb, len: 2, val: 0x3fff07a4
D (44460) BLE: Dequeue: type: 1, device: df:06:40:1a:4d:fe, char: 00002a00-0000-1000-8000-00805f9b34fb, len: 0, val: 0x0
***ERROR*** A stack overflow in task Tmr Svc has been detected.
Backtrace:0x40081bfa:0x3ffcf7500x4009051d:0x3ffcf770 0x40093b06:0x3ffcf790 0x4009200d:0x3ffcf810 0x40090618:0x3ffcf840 0x400905ca:0x3f40ebc8  |<-CORRUPTED
0x40081bfa: panic_abort at /home/ricardol/esp/esp-idf/components/esp_system/panic.c:402
0x4009051d: esp_system_abort at /home/ricardol/esp/esp-idf/components/esp_system/esp_system.c:128
0x40093b06: vApplicationStackOverflowHook at /home/ricardol/esp/esp-idf/components/freertos/port/xtensa/port.c:394
0x4009200d: vTaskSwitchContext at /home/ricardol/esp/esp-idf/components/freertos/tasks.c:3505
0x40090618: _frxt_dispatch at /home/ricardol/esp/esp-idf/components/freertos/port/xtensa/portasm.S:436
0x400905ca: _frxt_int_exit at /home/ricardol/esp/esp-idf/components/freertos/port/xtensa/portasm.S:231

```

Thanks

rlarranaga, 2022-11-08:
I actually managed to overcome the stack overflow issue by disabling remote logging.
I can see the characteristics being detected by esp32-ble-mqtt below (Device is `e4:65:06:ad:a1:ab`)
I do not see the characteristics in mqtt explorer though....

```
entry 0x40080694
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 15:32:09
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (98) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (106) boot:  5 fs_0             Unknown data     01 82 00310000 00040000
I (114) boot:  6 fs_1             Unknown data     01 82 00350000 00040000
I (121) boot: End of partition table
I (125) boot_comm: chip revision: 1, min. application chip revision: 0
I (133) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=49bbch (302012) map
I (250) esp_image: segment 1: paddr=00059be4 vaddr=3ffbdb60 size=06434h ( 25652) load
I (261) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=fb448h (1029192) map
I (633) esp_image: segment 3: paddr=0015b470 vaddr=3ffc3f94 size=01554h (  5460) load
I (636) esp_image: segment 4: paddr=0015c9cc vaddr=40080000 size=1c66ch (116332) load
I (687) esp_image: segment 5: paddr=00179040 vaddr=50000000 size=00010h (    16) load
I (702) boot: Loaded app from partition at offset 0x10000
I (702) boot: Disabling RNG early entropy source...
I (713) cpu_start: Pro cpu up.
D (714) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (714) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (722) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (729) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (736) cpu_start: Starting app cpu, entry point is 0x4008128c
0x4008128c: call_start_cpu1 at /home/ricardol/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
D (750) clk: RTC_SLOW_CLK calibration value: 3284838
I (763) cpu_start: Pro cpu start user code
I (763) cpu_start: cpu freq: 160000000
I (763) cpu_start: Application information:
I (767) cpu_start: Project name:     ble2mqtt
I (772) cpu_start: App version:      v0.16.0-8-g6ce66ab
I (778) cpu_start: Compile time:     Nov  8 2022 15:32:11
I (784) cpu_start: ELF file SHA256:  64004d190add06cc...
I (790) cpu_start: ESP-IDF:          v4.4.2
D (795) memory_layout: Checking 11 reserved memory ranges:
D (801) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (807) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (814) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (820) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (827) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (833) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffced88
D (839) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (846) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (852) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (859) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/ricardol/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1736

D (865) memory_layout: Reserved memory range 0x40080000 - 0x4009c66c
0x40080000: _WindowOverflow4 at /home/ricardol/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1736

D (872) memory_layout: Building list of available memory regions:
D (878) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
D (884) memory_layout: Available memory region 0x3ffb6388 - 0x3ffb8000
[... 1225 lines cut ...]
D (179400) OUTBOX: DELETED msgid=62359, msg_type=10, remain size=66
D (179410) MQTT_CLIENT: UnSubscribe successful
D (179410) MQTT_CLIENT: unsubscribe, topic"e4:65:06:ad:a1:ab/cba20d00-224d-11e6-9fb8-0002a5d5c51b/cba20002-224d-11e6-9fb8-0002a5d5c51b/Set", id: 30562
D (179430) MQTT_CLIENT: mqtt_enqueue id: 30562, type=10 successful
D (179430) OUTBOX: ENQUEUE msgid=30562, msg_type=10, len=101, size=167
D (179440) MQTT_CLIENT: Sent Unsubscribe topic=e4:65:06:ad:a1:ab/cba20d00-224d-11e6-9fb8-0002a5d5c51b/cba20002-224d-11e6-9fb8-0002a5d5c51b/Set, id: 30562, successful
D (179460) MQTT_CLIENT: mqtt_message_receive: first byte: 0xb0
D (179460) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (179470) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (179480) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (179480) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (179490) MQTT_CLIENT: msg_type=11, msg_id=56438
D (179490) MQTT_CLIENT: pending_id=30562, pending_msg_count = 2
D (179500) OUTBOX: DELETED msgid=56438, msg_type=10, remain size=101
D (179500) MQTT_CLIENT: UnSubscribe successful
D (179510) MQTT_CLIENT: mqtt_message_receive: first byte: 0xb0
D (179510) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (179520) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (179530) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (179530) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (179540) MQTT_CLIENT: msg_type=11, msg_id=30562
D (179550) MQTT_CLIENT: pending_id=30562, pending_msg_count = 1
D (179550) OUTBOX: DELETED msgid=30562, msg_type=10, remain size=0
D (179560) MQTT_CLIENT: UnSubscribe successful
```

shmuelzon, 2022-11-09:
Hey, in the logs (and actually also in your screenshots after looking again), the 4 topics you're seeing on the MQTT bus are the only 4 that are readable. So this application reads them and publishes the values. The two characteristics specific to the Switchbot are `cba20002-224d-11e6-9fb8-0002a5d5c51b` which is write only, so there's nothing to read here, and `cba20003-224d-11e6-9fb8-0002a5d5c51b` which is notify only so we can't request to read it but we do register for notifications, at which point, we need the Switchbot to notify us when something changes.

Perhaps you need to write something to the `cba20002-224d-11e6-9fb8-0002a5d5c51b` characteristic before it will publish anything. Looking at [this](https://github.com/OpenWonderLabs/python-host/blob/master/switchbot.py), you might need to write `0x570F31` first. So, after your device is connected, try publishing `87,15,49` to `e4:65:06:ad:a1:ab/cba20d00-224d-11e6-9fb8-0002a5d5c51b/cba20002-224d-11e6-9fb8-0002a5d5c51b/Set` and see if that makes any difference.

rlarranaga, 2022-11-09:
will do, thanks for the help!

rlarranaga, 2022-11-11:
Hello @shmuelzon ,
This appears to be working, thanks a lot for your help!

I have one more question before closing the issue. When BLE devices are discovered, there is usually an entry in BLE2MQTT-XXXX with the device mac, RSSI and distance. Once devices are connected, this information appears to be lost. It does not show up anymore in BLE2MQTT-XXXX , and it is not under the device values.

Is there any way to still receive this information?

Thanks again

shmuelzon, 2022-11-13:
Hey, that information is usually broadcasted by the BLE device. Some device aren't connectable at all, like the Eddystone beacon, iBeacon or other sensors and they only periodically publish information to whoever's in range. It appears that your device, while it isn't connected, publishes that information and stops while it is connected, so I don't think there's anything I can do to have that information show up. Unless your device also publishes the sensor data and then you can avoid connecting to altogether.

rlarranaga, 2022-11-13:
Hello @shmuelzon ,
If that is the case, and the device stops advertising signal strength when connected then why do i still receive the following messages in the logs:

2022-11-13 11:05:12.445442 (BLE2MQTT-B790.lan): I (9788345) BLE2MQTT: Discovered BLE device: ee:ca:79:d7:d4:27 (RSSI: -54), connecting
2022-11-13 11:06:10.408474 (BLE2MQTT-B790.lan): I (9846335) BLE2MQTT: Discovered BLE device: e4:65:06:ad:a1:ab (RSSI: -57), connecting
2022-11-13 11:06:32.110368 (BLE2MQTT-B790.lan): I (9868005) BLE2MQTT: Discovered BLE device: df:06:40:1a:4d:fe (RSSI: -42), connecting
2022-11-13 11:08:26.597390 (BLE2MQTT-B790.lan): I (9982495) BLE2MQTT: Discovered BLE device: e4:65:06:ad:a1:ab (RSSI: -62), connecting

I would have expected that since the device RSSI is still showing up in the logs, then it would still be available over mqtt.

Thanks

shmuelzon, 2022-11-13:
Since it's a different MAC address every time, I'm guessing it's just different BLE devices in your area and not the one you're interested in

rlarranaga, 2022-11-13:
No, all these 3 devices are the switchbot meter devices i am getting data from. They are whitelisted in my comfiguration, and i am definetely getting the temperature, humidity and battery values. But i do not get RSSI and distance.
Thanks

shmuelzon, 2022-11-13:
Those log messages are only printed when a device is discovered for the first time after which we either connect or ignore them. In your logs I can see we connect to them so you shouldn't see those messages until we disconnect from them, for whatever reason.
The MQTT topic for distance is only published for iBeacons and Eddystone beacons (which I guess the sensor mimics) and I don't believe those will be published when the sensor is connected.

shmuelzon, 2022-11-13:
If you have a BLE capture or something like that, we can try to verify the above...

rlarranaga, 2022-11-14:
Hey @shmuelzon , thanks for the reply,

Yes, the behavior you describe aligns with what i am experiencing. These devices are broadcasters, so esp32-ble-mqtt connects to them for a short period, to receive new data.

It would be great if the rssi (and other values from the broadcast message) would not get ignored after discovery, and could either be kept under BLE2MQQT-XXXX or even better, under the device topic (Maybe a flag that enables or disables this behavior in the configuration?). 

Signal level values are great for troubleshooting, even when the devices do as installed in a fixed location. Sometimes things in the environment change that make the signal worse, and being able to log the signal information can greatly help troubleshooting.

Thanks Again!

shmuelzon, 2022-11-17:
Hey,

> It would be great if the rssi (and other values from the broadcast message) would not get ignored after discovery

I don't think they're ignored, they're simply not broadcasted anymore since they're used for discovery. Once the peripheral was discovered and connected to, it has no reason to continue publishing itself.

> Signal level values are great for troubleshooting, even when the devices do as installed in a fixed location.

I'm not sure I have that information from the ESP-IDF APIs, I'll need to look into it.

rlarranaga, 2022-11-18:
> I don't think they're ignored, they're simply not broadcaster anymore since they're used for discovery. Once the peripheral was discovered and connected to, it has no reason to continue publishing itself.

I am trying to understand if i got the flow wrong. Once i get a topic for a device that is not part of the BLE2MQTT topic, i consider that device discovered, and connected to (At least once).

Now, if after that happens, i still see `Discovered BLE device: e4:65:06:ad:a1:ab (RSSI: -62), connecting` messages (which i do), i would think that the rssi is still being broadcasted  My point is, since i still get those messages in the logs, it should be possible to just publish them in the same connection topic the discovered device already has.

Thanks