shmuelzon/esp32-ble2mqtt #171: /get and /set issues

whatdaybob, 2023-01-10:
I have followed all the guides and have successfully set up an esp32 to connect and read characteristics but now am having 2 issues. I get intermittent disconnects on the BLE and I also can do a get or set request. Below is my configuration.

``` json
{
  "network": {
    "wifi": {
      "ssid": "SSID_NAME",
      "password": "SSID_PASSWORD"
    }
  },
  "mqtt": {
    "server": {
      "host": "MQTT_IP_ADDRESS",
      "port": 1883,
      "ssl": false,
      "username": "MQTT_USERNAME",
      "password": "MQTT_PASSWORD"
    },
    "publish": {
      "retain": true
    },
    "topics": {
      "prefix": "ESP32-BLE/",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },

  "ble": {
    "whitelist": ["10:D0:7A:03:E7:83"],
    "services": {
      "definitions": {
        "F5D0BF71-9670-4B25-97F1-861F5EBF0000": {
          "name": "Features"
        },
        "5842062B-3CF8-4D1B-8DFA-5D5120460000": {
          "name": "Network"
        }
      },
      "blacklist": ["8A6B286F-4198-4765-8BFD-17838F2E0000"]
    },
    "characteristics": {
      "definitions": {
        "F5D0BF71-9670-4B25-97F1-861F5EBF0500": {
          "name": "OP_SWITCH_MOODLIGHT",
          "types": ["utf8s"]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0400": {
          "name": "OP_BRIGHTNESS",
          "types": ["utf8s"]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0300": {
          "name": "OP_PLUGGED_INPUTS",
          "types": ["utf8s"]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0200": {
          "name": "OP_SWITCH_INPUT",
          "types": ["utf8s"]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0100": {
          "name": "OP_SWITCH_MODE",
          "types": ["utf8s"]
        },
        "5842062b-3cf8-4d1b-8dfa-5d5120460030": {
          "name": "WIFI_STATUS",
          "types": ["utf8s"]
        }
      },
      "blacklist": [
        "5842062b-3cf8-4d1b-8dfa-5d5120460020",
        "5842062b-3cf8-4d1b-8dfa-5d5120460010",
        "8a6b286f-4198-4765-8bfd-17838f2e0100",
        "8a6b286f-4198-4765-8bfd-17838f2e0200",
        "8a6b286f-4198-4765-8bfd-17838f2e0300"
      ]
    }
  }
}

```
I try setting both `ESP32-BLE/10:d0:7a:03:e7:83/Features/OP_SWITCH_INPUT` and `ESP32-BLE/10:d0:7a:03:e7:83/Features/OP_SWITCH_INPUT/Set` but neither do anything. I can use GATT on a raspberry pi to change the input but nothing happens when I use the esp2mqtt software.

I am unsure how to do debug logging as well, is this detailed anywhere?

shmuelzon, 2023-01-10:
Hey,

Do you see messages being published on your MQTT bus at all? Do you see the values read from the BLE device sent on the `ESP32-BLE/10:d0:7a:03:e7:83/Features/OP_SWITCH_INPUT` topic?
Debug logs will be very helpful here, you can enable them by executing `idf.py menuconfig` and go to "Component config" -> "Log output" and change "Default log verbosity" to "Debug"

whatdaybob, 2023-01-10:
Yeah, the updates happen pretty regularly once connected, sweeps through and updates them all in order approximately 4 times then disconnects for about 40 seconds. Update notifications also work well and in real-time. I will get the debug logs to you as soon as I can.

whatdaybob, 2023-01-10:
Log file available [here](https://pastebin.com/raw/kEYf53Es).

Couldn't trim it down as I didn't know what is useful or not.

In short it booted. Sat and did nothing except standard MQTT topic (the one with Uptime).

It then connected and created the read characteristics, then disconnected, it did this a few times. I then added /Set MQTT topics into the mix. Nothing happened.

shmuelzon, 2023-01-10:
Regarding the disconnections, I see the following messages: `I (179034) BLE: Connection closed, reason = 0x13`. `0x13` is `Connection terminate by peer user`. In any case, this project doesn't initiate disconnections (except when upgrading) so it's either really the peripheral or something in the ESP32 BLE implementation.

As for setting values, I don't see any MQTT messages arriving. There should be a log message like `MQTT: Received: ESP32-BLE/10:d0:7a:03:e7:83/....`. Are you sure you've published that message while the device is still connected?