shmuelzon/esp32-ble2mqtt #173: Help needed: Vogel MotionMount not showing

MunkusDK, 2023-01-27:
Hope someone can help as im going nuts :)

I am trying to have my [Vogel MotionMount](https://brandportal.vogels.com/m/32a2dbd23eb24166/original/8734070_NEXT7355_Leaflet_Document_EN.pdf) connected to HomeAssistant. I have tried with normal HA bluetooth intergration and also added a [ESP32 bluetooth proxy ](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiC3P3Ju-f8AhWWVfEDHap8AjsQFnoECA8QAQ&url=https%3A%2F%2Fesphome.github.io%2Fbluetooth-proxies%2F&usg=AOvVaw0Lm93fR9Vj9pB2YaAiiTtH) but it is not listing anywhere in the debug logs or popping up in HA.
I have now tried flashing this to my ESP32 WEMOS LOLIN32 Lite V1 but also no luck.
The device is shown on all my Windows PCs, Android phone and also from Ubuntu where I can control it using Gattool.

Using the nRF connect app I can also send and receive the characteristics needed for controlling it (Have added screenshots from app)
[image]
[image]

Config:
`{
  "network": {
    "wifi": {
      "ssid": "xxxx",
      "password": "xxxx"
    }
  },
  "mqtt": {
    "server": {
      "host": "192.168.1.187",
      "port": 1883,
      "ssl": false,
      "username": "xxxx",
      "password": "xxxx"
    },
    "publish": {
      "retain": true
    },
    "topics": {
      "prefix": "ESP32-BLE/",
      "get_suffix": "/Get",
      "set_suffix": "/Set"
    }
  },
  "ble": {
    "blacklist": [
      "54:13:79:57:fe:40"
    ],
    "services": {
      "definitions": {
        "e6fe65d-ed78-11e4-895e-00026fd5c52c": {
          "name": "Position"
        },
        "5842062B-3CF8-4D1B-8DFA-5D5120460000": {
          "name": "Network"
        }
      },
      "blacklist": [
        "8A6B286F-4198-4765-8BFD-17838F2E0000"
      ]
    },
    "characteristics": {
      "definitions": {
        "c005fa01-0651-4800-b000-000000000000": {
          "name": "Sideways",
          "types": [
            "utf8s"
          ]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0400": {
          "name": "OP_BRIGHTNESS",
          "types": [
            "utf8s"
          ]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0300": {
          "name": "OP_PLUGGED_INPUTS",
          "types": [
            "utf8s"
          ]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0200": {
          "name": "OP_SWITCH_INPUT",
          "types": [
            "utf8s"
          ]
        },
        "F5D0BF71-9670-4B25-97F1-861F5EBF0100": {
          "name": "OP_SWITCH_MODE",
          "types": [
            "utf8s"
          ]
        },
        "5842062b-3cf8-4d1b-8dfa-5d5120460030": {
          "name": "WIFI_STATUS",
          "types": [
            "utf8s"
          ]
        }
      },
      "blacklist": [
        "5842062b-3cf8-4d1b-8dfa-5d5120460020",
        "5842062b-3cf8-4d1b-8dfa-5d5120460010",
        "8a6b286f-4198-4765-8bfd-17838f2e0100",
        "8a6b286f-4198-4765-8bfd-17838f2e0200",
        "8a6b286f-4198-4765-8bfd-17838f2e0300"
      ]
    }
  }
}`

shmuelzon, 2023-01-28:
Hey,

The first thing that stands out in your configuration is that you've blacklisted one device (`54:13:79:57:fe:40`) which means the ESP will try to connect to anything else it finds. I suggest changing that to whitelist the device you really want,

Anything more than that would require more information and/or logs.
* Networking - Do you see the ESP32 connected to your home network? Was it assigned an IP address? Can you ping it?
* MQTT - Do you know if it's able to connect to the MQTT broker? It should publish some topics about it self, such as `ESP32-BLE/BLE2MQTT-XXXX/Status`, version, uptime, etc. Do you see any of those?
* BLE - This would require looking at the logs. Do you see it scanning for BLE devices? Does it detect the device you want? Does it connect to it? Does it complete service discovery and publish anything related to the device?

MunkusDK, 2023-01-28:
> Hey,
>
> The first thing that stands out in your configuration is that you've blacklisted one device (`54:13:79:57:fe:40`) which means the ESP will try to connect to anything else it finds. I suggest [...]

Hey 
Do you have a small guide on how to retrieve the logs to a file? I can only see then when in the terminal (Still newbie in the ESP world :))

In regards to you other questions - Yes everything seems to be working as expected. Webinterface showing many other BLE devices and the one I have tried to blacklist is some iBeacon that seems to spam a lot so was just to filter that out. So it is only the Vogel not showing up. I also dont see it in the logs in the terminal so it seems its not detecting it at all (its less than 1 meter away form the ESP)
MQTT is also working as expected

shmuelzon, 2023-01-29:
Hey,

> Do you have a small guide on how to retrieve the logs to a file? I can only see then when in the terminal (Still newbie in the ESP world :))

Copy and pasting from the terminal is the simplest option :)

Can you please try to whitelist the device you want instead of blacklisting the iBeacon? There are usually a lot of Bluetooth devices that won't all accept connection attempts and, while we try to connect, we don't scan for other devices. If you whitelist only what you want, there's a better chance for it to work.
After you do that, you should see logs like `Discovered BLE device: XX:XX:XX:XX:XX:XX` and hopefully, you'll see yours.

BTW - Some devices randomize their MAC addresses as a security feature and somehow, through pairing, both BLE devices identify themselves for the next time they connect. I've never really dealt with that in the past so if your device does that, I'm not sure what to do. Anyway, let's see what happens first with the whitelist and take it from there.

MunkusDK, 2023-01-30:
Hi again :)
> Hey,
> 
> > Do you have a small guide on how to retrieve the logs to a file? I can only see then when in the terminal (Still newbie in the ESP world :))
> 
> Copy and pasting from the terminal is the simplest option :)
> 

Have attached the log output copied from terminal

> Can you please try to whitelist the device you want instead of blacklisting the iBeacon? There are usually a lot of Bluetooth devices that won't all accept connection attempts and, while we try to [...]

Unfortunately it still is not showing up

> BTW - Some devices randomize their MAC addresses as a security feature and somehow, through pairing, both BLE devices identify themselves for the next time they connect. I've never really dealt with [...]

I Whitelisted the MAC now (34:15:13:D0:50:1A) - seems like every device it finds now, it says not connecting in log so assume that works as it should?

[Debug_log_output.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/10534308/Debug_log_output.txt)

shmuelzon, 2023-01-30:
That is a weird one... The MAC address of the device seems to belong to a TI chipset which I have a few at home some work (though they're pre BLE v5.0).
In the logs I see 22 devices so it does detect something and it seems to have the same MAC address when detected by your phone and Linux box so it's probably not a random address...
BTW - have you tried browsing to the ESP's web interface? It should show the detected devices and, if available, their names (which the MotionMount seems to publish).
If there's nothing there, I'm afraid I don't have any ideas at the moment. We can try updating ESP-IDF if has something to do with the BLE library.

MunkusDK, 2023-01-30:
Yeah also not in the webinterface... So weird. Well im up for testing every idea you might have. As said, was the same issue when I flashes it with ESPhome bluetooth proxy for homeassistant so might be same lib also used in there or?

shmuelzon, 2023-02-07:
The fact that you're not seeing your device on an ESP32, regardless of which application it runs, makes me think of some sort of interoperability issue. Maybe it's transmitting on a channel the ESP32 doesn't support? Or the timing is off somewhere?
Can you identify which channel it's transmitting on and perhaps we can take it to ESP's forums to ask about it?