shmuelzon/esp32-ble2mqtt #174: Issuing commands to Opal Ice Maker

MrJeems, 2023-02-07:
**Question**
Up front, I'm still rather new at HA, MQTT, and esp32-ble2mqtt, so there are multiple points of failure that can be considered.

My setup is using esp32-ble2mqtt with the [GE Opal Ice Maker](https://github.com/shmuelzon/esp32-ble2mqtt/wiki/Opal-(GE)-Ice-Maker).  I created an HA switch to issue MQTT packets turning the Ice Maker On or Off.  The changes of state were observed in MQTT Explorer, but there was no physical change to the status of my Ice Maker.  What all is required to make Home Assistant successfully command the Ice Maker to turn off/on using this setup?

Again, this could be an issue with my HA switch, a missing config item on my esp32, or missing parameters in my MQTT packet.  Any assistance is appreciated to further educate me in these fields.

shmuelzon, 2023-02-07:
Hey,

Have you verified that the ESP32 was able to connect to your network and to the MQTT broker?
In addition, the configuration you've linked to above includes a whitelist entry for the specific MAC address of @chrislawson device. Did you change that to the address of yours?
If you have and it still doesn't work, please provide some logs from the ESP32 so I can try to understand what's going on.

Thanks

MrJeems, 2023-02-12:
Hey, thanks for responding.  I was able to dedicate some time this weekend to troubleshoot, working backward from the Opal toward my HA instance.  I ended up getting everything working and the issue was entirely in my attempts creating MQTT switches in Home Assistant.  I published my updated scripts for the Opal in the Wiki.

closed 2023-02-12 by MrJeems

shmuelzon, 2023-02-12:
Hey,

I'm glad to hear you got it working and thanks for updating the Wiki with your configuration.
Also, thank you for the sponsorship, it's very appreciated!