shmuelzon/esp32-ble2mqtt #175: Connecting to a custom device is causing stack overflow crash

ithinkido, 2023-02-14:
**Describe the bug**
I would like to connect to the BLE service on a[ Pinecil v.2 ](https://wiki.pine64.org/wiki/Pinecil)soldering iron. This device has no active broadcasts, and all data has to be polled at regular intervals. As the device is not battery powerd, this is not an issue for power consumption. 
I am not sure, but I suspect that this may be causing an issue with ble2mptt.  After creating a base config file the ESP32 starts a constant stack overflow boot loop once it detects the Pinecil.  Looking at the logs , I think it may be that the ESP is expecting some sort of active data reply , but not receiving it. 

**To Reproduce**
I can imagine that this may be quite tough to reproduce without the device.  I am using the pre-compiled v1.6 bin and starting from a fresh install with  ```esptool.py  -b 921600 --chip esp32 write_flash --erase-all  0 ble2mqtt-full.bin```

My config file is reduced to a base file to simplify testing  and looks like this   

```json
{
  "network": {
    "wifi": {
      "ssid": "xxxxxxxxxxx",
      "password": "xxxxxxxxxx"
    }
  },
  "mqtt": {
    "server": {
      "host": "192.168.1.200",
      "port": 1883,
      "username": "xxx",
      "password": "xxxxx"
    },
    "publish": {
      "qos": 0,
      "retain": true
    }
  },
  "ble": {
    "whitelist": [
      "xx:D7:FD:D7:83:xx" ]
   }
 }

```
**Logs**
the log for a complete reboot loop cycle :
[log.txt](https://github.com/shmuelzon/esp32-ble2mqtt/files/10733006/log.txt)  

and and  extracted form,  when the issue happens : 

```.[0;32mI (5516) BLE2MQTT: Connected to MQTT, scanning for BLE devices.[0m
.[0;32mI (6026) BLE2MQTT: Discovered BLE device: xx:d7:fd:d7:83:xx (RSSI: -70), connecting.[0m
.[0;32mI (7256) BLE2MQTT: Discovered BLE device: xxxxxxxxxxx(RSSI: -96), not connecting.[0m
.[0;31mE (8406) BT_APPL: char not added, no resources.[0m
.[0;31mE (8436) BT_APPL: char not added, no resources.[0m
.[0;31mE (8436) BT_APPL: char not added, no resources.[0m

***ERROR*** A stack overflow in task BTU_TASK has been detected.

Backtrace:0x4008f80a:0x3ffe47d0 0x4008ff35:0x3ffe47f0 0x400931b6:0x3ffe4810 0x40091ad8:0x3ffe4890 0x4009002c:0x3ffe48b0 0x4008ffde:0x3ffe4364 |<-CORRUPTED

ELF file SHA256: c6cc4a6af92e3f56
```

**Additional context**
Using the same setup, there is no stack overflow issue , as long as the Pinecil is not detected. 

I have successfully connected to and sent control messages to the Pinecil using BLE-client on ESPhome, so I am quite sure that the communication does work.

shmuelzon, 2023-02-15:
Hey @ithinkido,

From what I can tell, that error message (`char not added, no resources`) is printed when we connect to a device which has more characteristics than the configured maximum (default is 40). From the scan log of the Pinecil, is looks like there are, at least, 59.
Luckily, ESP-IDF has allowed, since v4.3, to change this value. Are you able to compile this project and try it? If so, the configuration name is `BT_GATTC_MAX_CACHE_CHAR`. If not, I'll try, over the weekend, to prepare a new image for you to test.

The second part that troubles me is that stack overflow in the log. I don't know if it's related to the maximum characteristics thing or if it's a different issue but, if it still appears after changing the above configuration, we can also try to increase the stack size of the BT library with the `BT_BTU_TASK_STACK_SIZE` configuration option (the default is 4096).

Let me know...

ithinkido, 2023-02-15:
Thanks for the reply. 
Changing the `BT_GATTC_MAX_CACHE_CHAR` from 40 to 80 has stopped the stack overflow crash.  So far - it all looks great.

shmuelzon, 2023-02-15:
Nice! Thanks for the update!
Please leave this ticket open for now. I'll increase the value by default here as well as connecting to characteristics is basically what this is all about...

ithinkido, 2023-02-15:
Thank you for the great work on this project. I am most impressed by the automatic  detection and setup of the services as well as the assigning of the correct data type automatically. 
Coming form ESPhome , I had a config file close on 1800 lines for this device, as each service needed to be defined with data type and, and in most cases, the data then had to be cast  and parsed  to a std::vector<uint8_t> array for sending ,of course, the same in  reverse for reading data. 
Here , all I need is - ```whitelist-mac_address``` and everything just works. What a pleasure !

shmuelzon, 2023-02-17:
Thank you for the kind works.
This project has a very limited scope and, as such, it has the privilege of going into finer details to make things as automatic/simple as possible and (I hope) there's a niche for it.
I'm glad to hear you found it useful!

PR #177 (2023-02-17): Increase maximum number of allowed characteristics
Fixes #175
commit ee28e076 (2023-02-17): Increase maximum number of allowed characteristics (#177)

Fixes #175
closed 2023-02-17 by shmuelzon