shmuelzon/esp32-ble2mqtt #181: Maintain BLE connection on WiFi disconnect

dakhnod, 2023-02-21:
I noticed that the Bridge, when losing wifi connection, disconnects all the BLE connections.
Only to reinitiate the connections after connecting to WiFi.

Since often times an active BLE connection is more power-efficient for the Peripheral than Advertising, I would see it as beneficial to handle BLE connections and WiFi independantly. BLE connections should be kept alive even if the WiFi connection drops out.

Also, GATT writes via MQTT to a temporarily disconnected BLE device should be queued until the BLE device is reconnected, and notifications/indications during disconnected WiFi should also be queued and send after WiFi is re-established.

I would love to discuss these Ideas and help where I can.
Thank you for your work so far.

shmuelzon, 2023-02-22:
I don't agree with what you're suggesting here.

It may be fine if there's only one BLE2MQTT instance in your setup but, in my mind, as they're pretty cheap, you would scatter them with overlapping ranges where if one disconnects for any reason, a different BLE2MQTT instance may pick up the slack. For that to work, the one that disconnected from Wi-Fi (or event just from MQTT) would also need to disconnect from the BLE peripheral so the latter will start advertising again.
This is also why I have the other "book-keeping" MQTT topics like if the device is connected or not (with the active BLE2MQTT monitoring that) and also the "owner" topic telling you which instance is actually connected to the peripheral.

As for saving published MQTT topics until you actually connect to the peripheral, that's also a problem. First, you need to be aware of the peripheral and its services in order to do that, so it has to connect at least once. Second, with the above in mind, you may run into synchronization issues. For example, let's say I have a light switch. I publish a topic to turn in on. The BLE2MQTT that's actually connected to the switch turns it on and the idle one just saves the fact that it should be turned on. You go on about your day and manually toggle the switch off (not via MQTT). Now if the BLE connection with the first BLE2MQTT drops and the second takes over, it still remembers that it should turn the light on, despite that being not relevant anymore.
If you really want whatever is published on the MQTT topic to happen the next time the BLE2MQTT connects to the peripheral, simply publish it with the retained flag. When the BLE2MQTT connects to the peripheral and starts subscribing to the relevant topics, it will immediately get the new value from the MQTT broker and apply it. The syncing scenario from above could still happen, though, but I'd still prefer the "powerful" broker to maintain a state rather than the BLE2MQTT.

Does that make sense?

dakhnod, 2023-02-23:
I am glad to hear your opinion on this, and I see your point about having backup gateways if one loses WiFi connection.

Yet, in my scenario I have multiple Bridges, and the peripherals are not share between them, since I want to control which central connects to which peripheral. If the peripherals are not moving, I find it preferable to have that kind predictability.

If the peripheral is moving we have a different story, then I see an application for your principle.

In a scenario with stationary peripherals I find disconnecting and reconnecting to a peripheral on WiFi loss (which only lasts a few seconds, usually) an unnecessary waste of power on the peripherals side.

So, since I also understand your point, maybe the behaviour on WiFi disconnect should be adjustable?

dakhnod, 2023-02-24:
> BTW - out of curiosity, how significant is this really? How often does Wi-Fi drop and how much battery life-time are you loosing each time?
> 
> IIRC, the peripherals I've coded were set to advertise themselves every second and, when connected, the connection interval was 0.25s with up to 3 sleep cycles (don't remember the real term for it now) but, the point is, it would transmit once a second whether it's connected or not

you're forgetting the packets needed for setting up the BLE connection.

> I didn't mean that the peripheral may me mobile, but that multiple gateways may be in reach of it. I'm not sure why it's important to you to know which gateway connects to which peripheral as that's abstracted away on the MQTT bus (unless you've added a different MQTT prefix on each gateway).
> 
> Regardless, if you're up to the task, I'd be happy to accept a pull request to allow this behavior (leaving the current one as default). But, take into consideration, that due to all sorts of weird things that happened with ESP32s, we also reconnect to Wi-Fi if we fail connecting to MQTT and also try to reboot if Wi-Fi disconnects too many times so there are other things which may interfere with what you want to do (to the point where the ESP32 may never reconnect without manually rebooting it).

Okay, I see your point. I will take a look at the Code.